velvet-os/imagebuilder #1: eMMC Images?

nmaas87, 2020-04-11:
Thank you very much for providing the Odroid U3 Image - the Debian Buster version works perfectly for me. I had the problem that - upon login - two different HDMI displays did just cutout (black screen). I could resolve this by removing the 1024x768 option in the /boot/extlinux/extlinux.conf .

One question: Is it possible to use the image on an eMMC card or building it for that usage? I burned the image with etcher on the eMMC card - it works if I am trying to boot it from the uSD slot (basically as an uSD Card) - but not as eMMC, probably it needs some change in the bootloader or initial config...

Thanks a lot :)!

hexdump0815, 2020-04-30:
thanks for the nice feedback and suggestion - good to hear that this stuff is useful for someone else too :)

i think the bootloader layout is a little different for emmc - it should be written in the doc folder of the u-boot sources somewhere. i'm not planning to add emmc support as i do not even have emmc for my u3.

my idea to provide all this was to fill the gap for some hardware for which there is no more recent linux available (like armbian etc.) to make it still easy to get started with that hardware. for more complex setups one will have to dive deeper into the setup ...

nmaas87, 2020-05-03:
Thank you for your awesome work, I will then just use it with an Micro SD Card and try to mount the eMMC as a second drive. I hope you don't mind, but I decided to point to your work, as I did some stuff with the Odroid U3 about 3 years ago and get sometimes questions for newer kernels and images - which I don't have the time to work at anymore - so that those people can more easily find your work :) https://www.nico-maas.de/?p=1491 

All the best and thanks!

hexdump0815, 2020-05-04:
@nmaas87 - thanks for further distributing the link to this repo - it is good if people who want to run an actual mainline kernel still on their u3 can easily find it

best wishes - hexdump

poVoq, 2020-07-13:
Same issue with the Odroid-X2 and my emmc card. Bummer that you do not plan to support booting from the emmc.

Does anyone have a suggestion how to modify it to work with the emmc slot?

hexdump0815, 2020-07-13:
i just got an u3+ with emmc module, so lets see what will come out of it :) .... but it might take some time until i'll get to it ...

poVoq, 2020-07-13:
Cool. Can you reopen this then so that you will remember it?

nmaas87, 2020-07-13:
Sorry I was so bad and reopened it, I hope thats ok :)

hexdump0815, 2020-07-14:
sure - no problem :)

t-b-x, 2020-07-15:
Hello all,
will appreciate an image for U3 eMMC as well. I tried some inplace-upgrade to Ubuntu 18.04 LTS, but this was kind of buggy (ro filesystem, no ssh) so i decided to restore my backup. Maybe an new image will work better than my poor try ;)

Thanks^^

hexdump0815, 2020-07-18:
so - i was successful with writing a u-boot to emmc so that the u3+ boots from it - i have described the steps here: https://github.com/hexdump0815/u-boot-misc/blob/master/readme.exy#L35-L49 ... here is a precompiled u-boot for that - please test it beforehand if it works well for you on a sd card - just write the odroid u3 image to it and afterwards the u-boot as described here: https://github.com/hexdump0815/u-boot-misc/blob/master/readme.exy#L32-L33

good luck and let me know if it works for you too ...

nmaas87, 2020-07-27:
Sorry I might be low on caffeine - but cannot really wrap my head around it. I cannot find a precompiled u-boot file linked, but I am guessing following: We shall just prepare the eMMC module as normal (i.e. recover the boot partition as shown here: https://forum.odroid.com/viewtopic.php?t=969 in case we damaged that) - then make a backup of that boot partition as shown here https://github.com/hexdump0815/u-boot-misc/blob/master/readme.exy#L35-L49 - flash the image of the eMMC and then reflash the backup boot partiton so that its there again? Is that a good understanding? Thank you :)

hexdump0815, 2020-07-28:
@nmaas87 - the u-boot boot block binary is here: https://github.com/hexdump0815/u-boot-misc/releases/download/200718-01/exy-boot.dd.gz (to be uncompressed with gunzip before use) ... you can just write my ubuntu or debian image to the main emmc area (something like /dev/mmcblk0 or so) and write the boot block as described in my readme to the boot area of the emmc device (/dev/mmcblk0boot0 for instance) ... for this you need a linux machine with a built in sd card reader and the odroid sd card adapter for your emmc - only then you'll get those devices as given in the example above - with a usb card reader you'll most probably not see the boot0 part of the emmc i guess ... good luck

nmaas87, 2020-07-29:
Worked for me 👍 
My eMMC did not work anymore, so I just wipped it completly and used https://forum.odroid.com/viewtopic.php?f=53&t=969 to install the bootloader again.

First I made a backup of the bootloader
sudo dd if=/dev/mmcblk0boot0 of=old-boot-block.dd iflag=dsync oflag=dsync status=progress

Then became root, removed the ro protection and flashed the new bootloader, afterwards included ro mode again
sudo su
echo 0 > /sys/block/mmcblk0boot0/force_ro
dd if=exy-boot.dd of=/dev/mmcblk0boot0 bs=512 skip=1 iflag=dsync oflag=dsync status=progress
echo 1 > /sys/block/mmcblk0boot0/force_ro

Afterwards I flashed your lastest odroid_u3-armv7l-debian.img.gz  with etcher. Thanks a lot! :)

hexdump0815, 2020-07-29:
happy to hear that ... so i think we can close this issue now?

nmaas87, 2020-07-29:
I guess so :)! Closing it!

closed 2020-07-29 by nmaas87