velvet-os/imagebuilder #4: U-Boot reboots after "usb start" on x96 mini

probonopd, 2020-04-28:
Hi @hexdump0815, thanks for pointing me to https://github.com/hexdump0815/imagebuilder/blob/master/boot/boot-amlogic_gx-aarch64/u-boot.bin on IRC.

I can successfully chainload this U-Boot build, but trying to use `usb start` leads to a reboot on the x96 mini:

```
gxl_p281_v1#fatload mmc 0 0x01000000 u-boot.ext
(...)
gxl_p281_v1#go 0x01000000
## Starting application at 0x01000000 ...
U-Boot 2019.01 (Mar 26 2019 - 22:16:31 +0100) libretech-cc
DRAM:  1 GiB
MMC:   mmc@72000: 0, mmc@74000: 1
(...)
Hit any key to stop autoboot:  2 
(...)
=> usb start
starting USB...
USB0:   Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.00
scanning bus 0 for devices... XHCI timeout on event type 33... cannot recover.
BUG at drivers/usb/host/xhci-ring.c:473/xhci_wait_for_event()!
BUG!
resetting ...
bl31 reboot reason: 0xd
bl31 reboot reason: 0x0
system cmd  1.
GXL:BL1:9ac50e:bb16dc;FEAT:ADFC318C:0;POC:3;RCY:0;EMMC:0;READ:0;0.0;CHK:0;
```

Do you have any idea what may be causing this or how to debug it?

Thank you very much.

probonopd, 2020-04-28:
Different result when using the other USB port:

```
=> usb start
starting USB...
USB0:   Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.00
scanning bus 0 for devices... Device not responding to set address.

      USB device not accepting new address (error=80000000)
1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
```

hexdump0815, 2020-04-29:
i guess this is expecting too much from a chainloaded u-boot ... for such
low level stuff to work you'll have to adapt the mainline u-boot to be
built with a dts matching your hardware exactly ... you may try to build
the different gxl u-boots and see if any of their u-boot.bin will work for
you

good luck and best wishes - hexdump

probonopd, 2020-04-29:
Thanks @hexdump0815  

> you'll have to adapt the mainline u-boot to be
built with a dts matching your hardware exactly

Do you know where I can learn about how exactly to do that?

hexdump0815, 2020-04-30:
i think there is no single ressource to learn that from - its a long learning and a lot of trial and error and sometimes its impossible because one does not really know how a certain hardware exactly works ...

hexdump0815, 2020-04-30:
btw. did you try to run the "usb start" on the legacy u-boot before the chainloading?

probonopd, 2020-05-01:
No, I didn't do that. Will see whether doing this will make a difference.

probonopd, 2020-05-01:
Indeed @hexdump0815 it __works__ when I run `usb start` before _and_ after chainloading. Does this mean that some (which?) aspects of the stock U-Boot need to be carried over/compiled into the new one?

probonopd, 2020-05-01:
`run usb_boot` will not boot an [openSUSE aarch64 Live ISO](http://download.opensuse.org/ports/aarch64/tumbleweed/iso/) dd'ed to the USB mass storage device though:

```
gxl_p281_v1#usb start   
(Re)start USB...
USB0:   USB3.0 XHCI init start
Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.00
scanning bus 0 for devices... Device not responding to set address.
Setup ERROR: address device command for slot 1.
retry new usb device
2 USB Device(s) found
       scanning usb for storage devices... init_part() 278: PART_TYPE_DOS
1 Storage Device(s) found
gxl_p281_v1#fatload mmc 0 0x01000000 u-boot.ext
card in
init_part() 278: PART_TYPE_DOS
[mmc_init] mmc init success
reading u-boot.ext
527980 bytes read in 34 ms (14.8 MiB/s)
gxl_p281_v1#go 0x01000000
## Starting application at 0x01000000 ...

U-Boot 2019.01 (Mar 26 2019 - 22:16:31 +0100) libretech-cc

DRAM:  1 GiB
MMC:   mmc@72000: 0, mmc@74000: 1
In:    serial@4c0
Out:   serial@4c0
Err:   serial@4c0
[BL31]: tee size: 0
[BL31]: tee size: 0
Net:   
Warning: ethernet@c9410000 (eth0) using random MAC address - 8a:6b:0d:xx:xx:xx
eth0: ethernet@c9410000
Hit any key to stop autoboot:  0 
=> 
=> run usb_boot
starting USB...
USB0:   Register 2000140 NbrPorts 2
Starting the controller
USB XHCI 1.00
scanning bus 0 for devices... 2 USB Device(s) found
       scanning usb for storage devices... 1 Storage Device(s) found

IDE device 0: Vendor: Kingston Rev:  Prod: DataTraveler 3.0
            Type: Removable Hard Disk
            Capacity: 7377.6 MB = 7.2 GB (15109516 x 512)
** Bad device specification usb -bootable **
** Unrecognized filesystem type **

=> fatls usb 0 
** Unrecognized filesystem type **
```

It seems that this bootloader cannot handle ISO9660 which is needed to boot openSUSE aarch64 Live ISOs?

hexdump0815, 2020-05-02:
maybe rebuild the chainloaded u-boot with iso fs enabled - i trhink mine was a libretech potato u-boot without any changes built from mainline u-boot

probonopd, 2020-05-02:
Thanks, it looks like https://build.opensuse.org/package/binaries/hardware:boot/u-boot:libretech-cc/openSUSE_Factory_ARM can be used (although it has the same USB mass storage issue).

How did you find out that `0x01000000` must be used instead of `${loadaddr}`?

hexdump0815, 2020-05-02:
if i remember correctly it was either from the amlogic irc or trail and error