velvet-os/imagebuilder #5: question: Advanced build script

RKaczmarek, 2020-05-31:
Hi,

i like this project. I'm using it to build an image for my odroid u3.

I'm looking forward to extend the abilities of the build scripts to optimize the build process with:

- bootstrap cache
- creating ubuntu 20.04 instead of 18.04
  - ubuntu server or desktop
- more options controlling the build process
- using newest kernel for odroid u3 with my own [kernel-odroid-u3](https://github.com/RKaczmarek/kernel-odroid-u3)
- and much more (got a lot of ideas)

I'd like to make a merge request if you like my changes. It's still work in progress, but you can have a look at [my forked imagebuilder](https://github.com/RKaczmarek/imagebuilder)

Best regards

hexdump0815, 2020-06-03:
hi @RKaczmarek - sorry for the long delay of the response, but time is a bit short right now. i'm very happy to see others actually making use of those scripts and its even nicer if they get improved or extended this way (either here or in a forked repo). i'll need a bit of time to have a closer look at what you did and merge requests are welcome, but i would prefer to keep the scripts more on the simple side and i have to make sure they still work together with my sonaremin build (parallel repo) ...

best wishes - hexdump

hexdump0815, 2020-08-09:
just fyi: i have started a dev branch with some improvements - see https://github.com/hexdump0815/imagebuilder/issues/8 ... maybe i'll pick up some of your ideas as well ...