velvet-os/imagebuilder #7: Update Ubuntu to 20.04 LTS + kernel

BKPepe, 2020-07-26:
Hi,

do you have any plans to update Ubuntu to 20.04 LTS? I am using your images just fine on Odroid U2, however, I am using kernel 5.4.5. Is there any way how to update it to version 5.4.53 with u-boot without configuring everything from scratch?

Anyway, thank you for your work! It's incredible. You should consider using [Github Sponsors](https://github.com/sponsors), so anyone including me can send you a few bucks.

hexdump0815, 2020-07-28:
hi,

i have focal on my list, but not really at high prio as bionic is worked very well still for me and focal brings its own problems with it like forced snap packages for some things etc. ...

thanks for the sponsor suggestion (i did not know about it) but it looks not trivial to get there and i also like to keep my open source activities very relaxed, as otherwise they might start to feel too much like work :)

the u3 was my first "real" arm sbc (compared to a banana pi m1 and raspberry pi 1 which both were to slow for anything serious) and i kept mainline running on it, so i decided it might make sense to provide it for others in an easy to consume format to potentially benefit from it too ... it is always nice to see and hear that others are really using it, making this old but still very good hardware still a useful device instead of ending up in the trash ...

hexdump0815, 2020-08-09:
just fyi: i have started a dev branch with some improvements - see https://github.com/hexdump0815/imagebuilder/issues/8

librehat, 2020-10-15:
I also wonder if there is a way to update *only* the kernel without flashing the whole image?

Btw, I'm also using U3 and thank you for your efforts in bringing mainline Linux kernel to it. It's been a little server at home

hexdump0815, 2020-10-25:
yes - kernel and system are quite independent - you can just compile a new kernel and install it is unpack on of the kernels i build from time to time in the root dir and add another boot entry in the extlinux.conf file for it ... make sure your /boot partition is large enough and keep the old kernel around until you tested the new one to be working well. you can find my kernel builds here: https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases - for the odroid the stb-exy ones are the right ones

good luck and best wishes - hexdump