velvet-os/imagebuilder #8: kind of a changelog of the dev branch

hexdump0815, 2020-08-09:
i'll try to use this issue to track some of the things i'm thinking about, experimenting with or working on in the dev branch for some kind of a next generation of the imagebuilder scripts

so far we have there:
* switch ubuntu from bionic to focal and start to make the ubuntu/debian version more generic (wip)
* switch from swap partition to a swap file - this is more flexible adjustable later
* cancelled: prepare for non snap chromium install - i think even armbian dropped it and firefox is more resource friendly anyway, whoever needs chromium may try the snap or do the debian repo trick
* fix a problem with the lightdm greeter on some systems (ubuntu focal) - slick-greeter seems to work
* disable apport
* disable automatic and unattended upgrades - i hated it when the system was blocked by the unattended upgrades directly after boot
* add zswap support - now that i have a script (/root/zswap-stats.sh) to show how much it really saves on disk writes i think this might stay
* make the created username flexible

hexdump0815, 2020-08-09:
btw. i also experimented with running an armv7l userland on aarch64 to save memory (armv7l have a something like 30% +/- lower memory footprint than aarch64 both on memory usage and filesize of binaries on disk) but finally gave up on it for the following reasons:

* i was not able to get firefox working stable with lots of tabs in such a setup - it always ended up with crashing tabs as soon as many of them were opened - i did not see this problem on native armv7l or running aarch64 natively on the same system
* compiling stuff and the kernel is not easy with such a setup as uname will report aarch64 which can confuse some configure scripts and for the kernel one needs an aarch64 chroot to compile in

it looks like the best approach to run for instance a firefox with about 40-50 tabs open on a 2gb aarch64 system is using zswap - in my tests so far it only resulted in minimal swapping (50-100mb)

tosiara, 2020-09-07:
I have flashed this test image, and so fat using as a server without gui (lightdm disabled)
Idle instance without any apps running shows 0.20 load - is that normal?

[image]

hexdump0815, 2020-09-07:
maybe this is top itself or some kernel daemons - looking at the top output for a while in H mode might give some clue ... also is the load usually averaged over some time, so maybe waiting a bit more might lower it even more ... maybe ubuntu is still doing some auto update checks on bootup too - i think i disabled quite a bit of that already but there might still some settings be left in /etc/apt/ ...

tosiara, 2020-09-07:
I made this screenshot in 1 hour 23 minutes after boot, so it was idling for quite long time. Never seen such high load on a clean system doing nothing

Will continue checking