velvet-os/imagebuilder #9: Server-image

Schnabelriechen, 2020-09-06:
Any chance there will be a "server-image" without the the desktop-enviroment? Just barebone ssh and stuff?

hexdump0815, 2020-09-06:
it is rather unlikeliy to happen soon as building (and especially testing) all the images already takes a lot of time, so adding even more will not really help :) ... the images should always start an openssh server, so you can just boot them and afterwards login and delete whatever you do not need ... or alternatively fork the repo and adjust the list of installed packages

best wishes - hexdump

Letme, 2022-11-13:
I have headless old odroid u3 and no keyboard/display at hand. SSH is not enabled by default. I am trying to find something Linuxy on location so that I could maybe check the settings within the image. Or should I maybe wait more than 10 minutes after boot for the server to appear on network (eth).

hexdump0815, 2022-11-14:
@Letme - sshd should be enabled by default ... there is only one special thing about the odroid u3 etc and eth: that eth mac address is hardcoded at the kernel cmdline and should be changed for all but the first of them in case someone is using multipe, but otherwise it should be reachable by ssh via eth - are you sure it boots properly at all (led blinking changes)?

Letme, 2022-11-15:
I don't know if it booted or not, but led indeed was not blinking after someone went and check it after 10 minutes. I don't have any MAC limitations on local network and was looking at ARP table, where it should for sure be visible.

I ended up with [meveric image](https://forum.odroid.com/viewtopic.php?f=79&t=45448) on the end, sorry.

hexdump0815, 2022-11-15:
@Letme - no problem, fine with me :)