velvet-os/imagebuilder #12: Include lima module in the kernel

librehat, 2020-11-04:
I'm trying to play with upstream `lima` driver and just recomplied `mesa` packages. Then I realised that `lima` module is not there from the kernel (Odroid U3).

I'll try to compile the kernel myself in the meantime.

It'd be nice to enable `lima` module in the next release :D

hexdump0815, 2020-11-20:
hello,

as lima still has some bugs in case of more complex opengl applications i default still to the legacy blob for now and as far as i remember both (legacy driver and lima) do not cooperate well, so i decided to leave out lima for now. the easiest is to simply build your own kernel leaving out the legacy mali related patches and enabling lima as you did. i plan to build with the legacy driver the last time when 5.10 is out as that will be an lts kernel and afterwards to forget about it and only build lima from 5.11 on which is hopefully bug free by then :)

best wishes - hexdump

librehat, 2020-11-25:
Compiling `upstream` kernel turns out to be easier than I expected, I simply grabbed the latest kernel (I stick with `5.4.x` for now) and `make menuconfig` enabled `lima`. Now the module and the new kernel are booted without any issues! However, for my use case, I realized it doesn't work yet (I wanted to use `VirtualGL` to have accelerated remote VNC session, however, it seems that `VirtualGL` doesn't support OpenGL ES)

Thanks again for your hard work!

librehat, 2020-12-02:
For whoever is interested in doing something similar, I've put up my steps into a [blog post here](https://www.librehat.com/building-linux-kernel-on-odroid-u3/) on building kernels natively on Odroid-U3 (the base system image is the amazing image provided by @hexdump0815)

Closing this issue now.

closed 2020-12-02 by librehat