velvet-os/imagebuilder #15: ipv6 disabled for what reason?

bw1faeh0, 2021-02-26:
I'm using the `odroid_u3-armv7l-ubuntu-focal-dev` image from the current release (200823-01) and I'm wondering why ipv6 is disabled in  `/boot/extlinux/extlinux.conf` (`ipv6.disable=0`).

Is it ok to enable it in a ipv6 environment or do I have to face problems?

hexdump0815, 2021-03-06:
hi @bw1faeh0 - there should be no problem enabling it - its just my personal preference to keep it disabled for now ...

p0358, 2023-03-09:
@hexdump0815 Sorry for ping, but is there a chance it could be flipped back to enabled by default? I spent a few hours figuring out why my UDP socket in a server would not start. Turns out it uses AF_INET6 family for it and would fail for this exact reason, and it also took me a moment to think that support in kernel could be disabled ;_;

I really think it doesn't make sense to have it this way, not even because someone might use/need IPv6, but because a lot of apps have a **hardcoded** bind to AF_INET6 socket family. Even apps that are mostly used for IPv4, because if you want your socket to work with both, you gotta bind to v6's family (then v4 clients can connect and will have IP of `[::ffff:1.2.3.4]` format.

That means that even if you only care about IPv4, you will run into breakage with these apps and they'll be unusable.

Also how do I even turn it back on? This or any sysctl's don't seem to work
```
# echo 0 > /sys/module/ipv6/parameters/disable
bash: /sys/module/ipv6/parameters/disable: Permission denied
```

hexdump0815, 2023-03-09:
@p0358 - maybe i should think about it, time is moving on :) ... to get rid of it, you'll have to change it on the kernel cmdline, so /boot/extlinux/extlinux.conf in the u-boot case - just drop the ipv6.disable=1 there ...

p0358, 2023-03-09:
@hexdump0815 The issue is that I don't have `/boot/extlinux/extlinux.conf` because I'm using the chromebook builds that have separate kernel image partition
```console
# cat /boot/extlinux/extlinux.conf
cat: /boot/extlinux/extlinux.conf: No such file or directory
# tree /boot
/boot
|-- Image-6.1.11-stb-mt7+
|-- System.map-6.1.11-stb-mt7+
|-- config-6.1.11-stb-mt7+
|-- dtb-6.1.11-stb-mt7+
|   |-- mt8173-elm-hana-rev7.dtb
|   |-- mt8173-elm-hana.dtb
|   |-- mt8173-elm.dtb
|   `-- mt8173-evb.dtb
|-- initrd.img-6.1.11-stb-mt7+
|-- lost+found
`-- vmlinux.kpart-6.1.11-stb-mt7+

3 directories, 9 files
```

I might try to do a hex-edit of the parition and change it to 0 there maybe xd

hexdump0815, 2023-03-09:
@p0358 - oh, then its not that easy - this should give you an idea of how you can adjust it based on the regular kernel and how you can test it: https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-with-luks-full-disk-encryption.txt#L247-L279 ... this is then even with initrd, but you can also omit it (which is the default for the kernels in the images) - this is the corresponding part from the kernel build: https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/blob/master/readme.mt7#L48-L57 with this being the cmdline file: https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/blob/master/misc.cbm/misc/cmdline

good luck and best wishes - hexdump

p0358, 2023-03-10:
Hm, I thought I'd go for easy way and just edit it like this
```
# ensure it's there
cat /dev/mmcblk0p1 | fgrep -a -o 'ipv6.disable=1'

dd if=/dev/mmcblk0p1 of=/root/mmcblk0p1_backup bs=4096
bbe -e 's/ipv6.disable=1/ipv6.disable=0/g' /root/mmcblk0p1_backup > /root/mmcblk0p1

# ensure nothing else was changed
diff <(od -An -tx1 -w1 -v mmcblk0p1_backup) <(od -An -tx1 -w1 -v mmcblk0p1)

# write back
dd if=/root/mmcblk0p1 of=/dev/mmcblk0p1 bs=4096
```

but it doesn't boot anymore xD (shows "Please insert a recovery USB stick or SD card")

does it do some integrity check or something on the partition contents?

p0358, 2023-03-10:
Okay, thanks for your directions and links I was able to actually rebuild the image properly. Pasting my final steps here for future reference, maybe would help someone:
```bash
cat /proc/cmdline > /boot/cmdline
# <now modify cmdline with a text editor manually here, also remove leading "cros_secure">

cp -v vmlinux.kpart-${kver} vmlinux.kpart-${kver}.old
cp Image-${kver} Image
lz4 -f Image Image.lz4
cp initrd.img-${kver} initrd.img.xz
dd if=/dev/zero of=bootloader.bin bs=512 count=1
ls dtb-${kver}/*.dtb | xargs printf " -b %s" | xargs mkimage -D "-I dts -O dtb -p 2048" -f auto -A arm64 -O linux -T kernel -C lz4 -a 0 -d Image.lz4 kernel.itb
vbutil_kernel --pack vmlinux.kpart --keyblock /usr/share/vboot/devkeys/kernel.keyblock --signprivate /usr/share/vboot/devkeys/kernel_data_key.vbprivk --version 1 --config cmdline --bootloader bootloader.bin --vmlinuz kernel.itb --arch arm
cp -v vmlinux.kpart /boot/vmlinux.kpart-${kver}
rm -f Image Image.lz4 initrd.img.xz cmdline bootloader.bin kernel.itb vmlinux.kpart

# single boot only testing:
dd if=/boot/vmlinux.kpart-${kver} of=/dev/mmcblk0p2 bs=4096
cgpt add -i 2 -S 0 -T 1 -P 15 /dev/mmcblk0

# worked, then for permanent:
dd if=/boot/vmlinux.kpart-${kver} of=/dev/mmcblk0p1 bs=4096
```

But my plea for changing the defaults to save others the headache is still up to date xd

I think if someone has issues with IPv6 stack, they can easily disable it on lower level down the chain with sysctl's, or even add the kernel param if they don't have Chromebook partitions. That could be much easier than other way around.

Thanks for help in re-enabling it again!