velvet-os/imagebuilder #16: unable to boot 210321-01 veyron on asus c100 flip

cracket, 2021-05-06:
Not sure if I have installed you image correctly, I have extracted via xz -d and   flashed via   dd if=.img of=/dev/sdc  to sd card. 

However asus c100 flip is not booting from it. I'm using developer mode and can boot different version of linux with ctrl-U, but from your image I see black screen and no error.

Did anyone succeeded booting veyron image on c100?

hexdump0815, 2021-05-07:
i think it might need its own u-boot (which the images boot instead of a kernel) - i'll try to build one for the flip in the next days and then you maybe can try again?

cracket, 2021-05-07:
are you sure replacing u-boot is needed to boot from SD? I've already booted different distros from SD card without any issue, which contained cgpt partitions with kernel and rootfs only.
I would't like to replace or wipe anything on built-in MMC memory as it contains the only working original chromeos kernel capable to support bluetooth and other devices, and in case I messed something up - I don't know how to bring back MMC if it will not boot.

hexdump0815, 2021-05-07:
no need to worry :) - the u-boot i mean is on the sd card and it is there on the kernel partition instead of the kernel - this way we get a nice boot selector and the u-boot then loads then the kernel we want from the /boot partition on the sd card ... so in the end the emmc will not be touched at all

cracket, 2021-05-07:
ok, I found it in the sources:
https://github.com/hexdump0815/imagebuilder/blob/master/boot/boot-chromebook_veyron-armv7l/extlinux/extlinux.conf
the image is for 3 different machines, but the default one is veyron-jaq and no minnie at all. 

You are using DTB from mainline kernel? 
Just curious because I've compared DTS from mainline and shipped with chromeos, and there are small differences which I don't exactly understand if they are relevant, as I'm newbie to DTS

hexdump0815, 2021-05-08:
i did not include the minnie (flip) dtb yet as i do not have a device to test it, but the dtb for minnie should be installed in /boot/dtb-kernel-version i think (if not please let me know) - can you please create another entry for it in your /boot/extlinux/extlinux.conf ... and then gunzip this file: https://github.com/hexdump0815/u-boot-chainloading-for-arm-chromebooks/releases/download/v2021.01-cbr/uboot.kpart.cbr-minnie.gz and dd it to the first partition of your sd card ... in case this will boot we are done, otherwise i'll have to build minnie versions for older u-boots too, as for some others the latest version from 2021 did not work anymore ...

regarding the dtb files: chromeos and mainline kernel are completely different and thus the dtb files are not interchangable

hexdump0815, 2021-05-09:
i have created a new veyron image here: https://github.com/hexdump0815/imagebuilder/releases/tag/210509-01 ... it contains some changes for the flip - maybe you can give it a try ... you still have to write the proper u-boot (see above) to the first partition and maybe change the default boot target in extlinux.conf on the third partition to minnie ... good luck :)

cracket, 2021-05-10:
So - good news: veyron minnie image is working on c100. I have followed your steps and ubuntu has started including xfce. 
What didn't work (just briefly because I didn't have much time to test) - bluetooth, wifi and battery was not recognized. 
What else worked - sd and touch screen. I had to choose system image during boot by pressing the number (in my case it was 4), so looks like didn't have to change much except extlinux.conf contents. Feel free to add entry for veyron minnie into your repo. 

Regarding dtb - how do you know if chromeos & mainline are completly different? I have compared  dtb ripped out from my chromebook (kernel 4.x) to mainline one (5.x) there were maybe 4 or 5 differences, the rest is almost the same. I'm not an dts expert, so it should be reviewed by someone who could tell if the differences are relevant or cosmetic, but still look pretty similar.

hexdump0815, 2021-05-10:
that sounds good ... i'm surprised that battery and wifi did not work as i extra added those in the new image - which kernel version do you have running there? it should be 5.10.35 - at least this is the version with the additions. did you have the menu entry in u-boot already there for minnie? (as that should also already be part of the new image)

regarding the dtb differences: if it is the case as you describe then i assume google is porting back a lot of stuff from mainline into their 4.x kernel tree so that the differences are that small.

cracket, 2021-05-11:
Looks like the filename for nvram was wrong, I have renamed 
brcmfmac4354-sdio.google,veyron-speedy-rev9.txt        to
brcmfmac4354-sdio.txt
and it started working. But kernel version is 5.10.25-stb-cbr+, all entries in extlinux.conf  are related to 5.10.25 and menu entry was for minnie.
Your dts is correct, as in /proc/device-tree/compatible  shows veyron-minnie list. 
Anyway thank you for your misleading comment - I've packed mainline kernel 5.10.34 with chromiumOS (4.19.113) dtb and this combination has made my bluetooth working. Without your note I would have never tried "interchangeable dtb"
I'll try to test more flip hardware as there are more people interested in working c100 devices and I'll probably compare your kernel settings with mainline&chromium to see the difference.

cracket, 2021-05-12:
kernel installed fine, just haven't used .kpart file - I assume with u-boot on first partition it's not needed (second partition is filled with zeros, so kernel is in /boot only). 
Indeed - this kernel gives battery, but no bluetooth again.  During   boot - kernel freezed for about 30 seconds to check BTRFS rootpart - not sure if this is normal, as I have no experience with BTRFS.
I will try using it now, if you want me to check anything special - just let me know

hexdump0815, 2021-05-13:
that sounds good ... the kpart kernel is just there in case someone does not want to use the u-boot option ... i'm using btrfs as root fs in those images as this gives a lot of extra space due to zstd compression without any negative side effects - i guess the freeze is when it checks all the emmc partitions for btrfs maybe

it might be interesting to see if swapping the dtb with your chromeos one will give you bluetoothere as well - i guess it should work again

hexdump0815, 2021-05-16:
i think the 30 sec hang on boot is not the btrfs scanning - it is the initramfs searching for the resume device - running a "update-initramfs -c -k 5.10.35-stb-cbr+" should solve that (or adding "noresume" to the kernel cmdline for the next vmlinux-kpart kernel build)

cracket, 2021-05-20:
swapping dtb does not provide bluetooth, probably you need to add couple of extra config items in kernel config. Hanging is not a problem anymore. 
However my concern poor performance during video play (worse than with PrawnOS image).  Did you set up any hardware acceleration for 3288?

hexdump0815, 2021-05-28:
i just checked and looks like prawnos has some additional userspace in place for hantro hw accel ... i'll for sure add the CONFIG_HANTRO kernel option to future kernels, userland might have to wait until it arrives in newer ubuntu/debian versions - but maybe i find the time to look at it even before that, but its lower prio for me

cracket, 2021-05-30:
indeed - to have a good hardware support system needs userland libraries configured to use hw features, it's needed not only for video acceleration, but even for crypto hw acceleration. 
But coming back to kernel options for imagebuilder - as I understand to build system image you are building kernel from linux-mainline-and-mali-generic-stable-kernel  repo with config.cbr-5.10.35-stb-cbr+  config - which should support both tinkerboard and veyron?
If so - I have compared my config with this one, and looks like to support bluetooth from bcm4354 following option is missing:
CONFIG_SERIAL_DEV_CTRL_TTYPORT
there is a discussion about bt and veyron chromebooks explaining it:
https://unix.stackexchange.com/questions/498473/why-is-bluetooth-not-working-on-my-chromebook-arch-linux-arm
and chromiumos device tree. Not sure if this can cause any problems to other boards like tinkerboard. 

I will be checking hantro support for 5.x - however it's still very limited. I'll let you know if I find out something usable.

hexdump0815, 2021-05-30:
cool - thanks a lot for this info - i have added it so that it will not be forgotten for the next version: https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/commit/e41f6826a7012c3b465a6e3117c276f41673f9e3

cbr is for veyron chromebooks only (as they seem to require quite a few special options beyond defconfig), so that should be fine - tinkerboard etc. are supposed to be handled by the generic armv7l kernel av7

hexdump0815, 2023-09-20:
@vfonov - could you maybe try a different sd card? - the minnie u-boot should in theory work for your device and the fact that it seems to work from time to time might maybe be related to problems with the sd card ... bluetooth is sometimes complicated on veyron chromebooks

vfonov, 2023-09-21:
ok, will try

vfonov, 2023-09-23:
I tried on a different SD card, and got same result. I didn't try too many times, `uboot.kpart.cbr-minnie` - I got a black screen after u-boot prompt, `vmlinux.kpart-6.1.11-stb-cbr+`  - works fine.

hexdump0815, 2023-09-30:
no real idea then what it could be - you might try the other u-boot.kpart files from the /boot/extra dir - maybe one of them is working better?

vfonov, 2023-10-08:
So, i tried all the bootloaders in the `extra` , the `uboot.kpart.cbr-speedy` was showing boot screen and started booting the kernel, but black screen after.

hexdump0815, 2023-10-21:
did you try all possible options in the menu with the working boot block then?

vfonov, 2023-10-21:
> did you try all possible options in the menu with the working boot block then?

no, was trying options only for minnie

michelberger, 2023-12-01:
Continuing from [#130](https://github.com/hexdump0815/imagebuilder/issues/130#issuecomment-1830454965): I was able to recompile kernel 6.1.51.stb-cbr+ with the lts branch of https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/. Copied the new vmlinux.kpart-6.1.51-stb-cbr+ and dd'd it to `mmcblk1p1` (microsd) and the kernel now reads veyron's emmc partitions.

dd'd it over to `mmcblk0p1` (emmc) and the image now boots 100% of the time on the Asus C201P speedy!

Still curious why u-boot doesn't play nicely with veyron, but I suppose it's not strictly necessary to have it in place.

ounsatn, 2024-09-12:
Hello, got an asus flip C100P / Minnie, tried different way to have a linux an i read with attention the differents posts. 
Just to make a little summary to make it works here is what i did : 
- grab this linux version : 
[230218-05(https://github.com/hexdump0815/imagebuilder/releases/tag/230218-05 )
under chromeOS inside linux console in root you can do : 

- dd img to your sd ( dd if=xxx.img of=/dev/mmcblk0 ) 
- mount /dev/mmcblk0p3 /tmp  
- dd if=/tmp/vmlinux.kpart-6.1.11-stb-cbr+ of=/dev/mmcblk0p1 

reboot & press Ctrl U to boot linux. 
tell me if it's wrong, but it worked like this ;) 

thanks for the works done to have a full linux instead Chrome.

fleminra, 2026-02-03:
I have it running on the eMMC of the Asus Flip, a.k.a. C100PA.

Turns out the first 8 MB of the eMMC are not writable. `dd` will appear to work, but you'll have kernel errors if you look (`dmesg`).

So the key is just to push everything out, past the 8-MB mark. Here's my working partition layout:

```
# cgpt show /dev/mmcblk0
WARNING: Primary GPT header is being ignored
       start        size    part  contents
           0           1          PMBR (Boot GUID: EE734B9B-2796-FA46-AECF-A2749CB1F2D1)
           1           1 IGNORED  Pri GPT header
    30777311          32          Sec GPT table
       16384       65536       1  Label: "KernelA"
                                  Type: ChromeOS kernel
                                  UUID: 6AC92D87-D45C-4D46-B4E2-2A3B19E5A019
                                  Attr: priority=10 tries=2 successful=1 
       81920       65536       2  Label: "KernelB"
                                  Type: ChromeOS kernel
                                  UUID: 7233F26D-50E1-7341-8B17-AA3BEDEF1682
                                  Attr: priority=0 tries=0 successful=0 
      147456     1048576       3  Label: "Boot"
                                  Type: Linux data
                                  UUID: A68D5AE3-401F-564B-86E7-20BBE6D3D627
                                  Attr: legacy_boot=1 
     1196032    29581278       4  Label: "Root"
                                  Type: Linux data
                                  UUID: F326F44A-F916-3C40-BA26-ED45D7458D4F
    30777343           1          Sec GPT header
                                  Sig: [EFI PART]
                                  Rev: 0x00010000
                                  Size: 92 (blocks)
                                  Header CRC: 0xced320ba 
                                  My LBA: 30777343
                                  Alternate LBA: 1
                                  First LBA: 34
                                  Last LBA: 30777310
                                  Disk UUID: 881338EE-7857-4F45-8265-8FD9578382BB
                                  Entries LBA: 30777311
                                  Number of entries: 128
                                  Size of entry: 128
                                  Entries CRC: 0xc1480911
```

For comparison with the downloaded image:
```
> cgpt show chromebook_veyron-armv7l-trixie-250907.img
       start        size    part  contents
           0           1          PMBR
           1           1          Pri GPT header
           2          32 INVALID  Pri GPT table
     7970783          32          Sec GPT table
        8192       65536       1  Label: "KernelA"
                                  Type: ChromeOS kernel
                                  UUID: C27F5D64-5454-3B4C-A32C-E50D015685DD
                                  Attr: priority=10 tries=2 successful=1 
       73728       65536       2  Label: "KernelB"
                                  Type: ChromeOS kernel
                                  UUID: A5D23B8D-0CF7-D842-B8B4-CCB5B34AC13B
                                  Attr: priority=5 tries=2 successful=0 
      139264     1048576       3  Label: ""
                                  Type: Linux data
                                  UUID: 8751B509-AEE4-4C60-ADEC-40FC4CC6723E
                                  Attr: legacy_boot=1 
     1187840     6780928       4  Label: ""
                                  Type: Linux data
                                  UUID: 8B9C8F46-87F9-4DF1-9FD6-14F5DEA601C0
     7970815           1          Sec GPT header
WARNING: One of the GPT headers/entries is invalid
```

Steps to achieve this:
```
# create fresh GPT (ignore the "Primary GPT header is being ignored" warnings)
cgpt create /dev/mmcblk0

cgpt add -i 1 -t kernel -b   16384 -s    65536 -l KernelA -P 10 -T 2 -S 1 /dev/mmcblk0
cgpt add -i 2 -t kernel -b   81920 -s    65536 -l KernelB -P  0 -T 0 -S 0 /dev/mmcblk0
cgpt add -i 3 -t data   -b  147456 -s  1048576 -l Boot    -B  1           /dev/mmcblk0
cgpt add -i 4 -t data   -b 1196032 -s 29581278 -l Root                    /dev/mmcblk0

# refresh kernel's partition table knowledge
sudo partx -a /dev/mmcblk0

# copy from the running system (the SD card) to the eMMC
dd if=/dev/mmcblk1p1 of=/dev/mmcblk0p1 bs=4M status=progress
dd if=/dev/mmcblk1p2 of=/dev/mmcblk0p2 bs=4M status=progress
dd if=/dev/mmcblk1p3 of=/dev/mmcblk0p3 bs=4M status=progress

mkfs.brtfs /dev/mmcblk0p4
mkdir /mnt/mmcblk0p4
mount /dev/mmcblk0p4 /mnt/mmcblk0p4
rsync -aAXv --exclude={"/dev/*","/proc/*","/sys/*","/tmp/*","/run/*","/mnt/*"} / /mnt/mmcblk0p4/

btrfs filesystem label /dev/mmcblk0p4 rootpart
```

Then reboot, *remove the SD card*, and press Ctrl-D to boot this from the eMMC. If you don't remove the SD card, the boot process might catch filesystems from it instead of the eMMC, since they have the same labels.

Thanks hexdump0815 for the amazing work!

thenameisluk, 2026-02-03:
please refer to [this](https://velvet-os.github.io/chromebooks/installation/basic-installation.html) instead of whatever else you were reading