velvet-os/imagebuilder #18: Suggestion: Ubuntu @/@home btrfs subvolume layout

stsichler, 2021-06-09:
Hi Hexdump,
first: **thank you** for your work on that images!!! I'm very happy with it using your [Ubuntu 20.04 image](https://github.com/hexdump0815/imagebuilder/releases/tag/210508-01) for my Odroid-U3.
I have a small suggestion: 
Now, as you switched to btrfs as rootfs, what do you think about using the Ubuntu-typical @/@home btrfs subvolume layout to allow the usage of timeshift right out-of-the-box? This could be very handy when things are suddently messed up...
As I pointed out [here](https://github.com/teejee2008/timeshift/issues/241#issuecomment-857750118), due to timeshift's internals, this would also require to name the root partition explicitly by UUID in the fstab.
Also a change in extlinux.conf would be necessary, namely selecting the correct boot subvolume using kernel option `rootflags=subvol=@`, because timeshift seems to also require that the default subvolume of rootfs stays /.

hexdump0815, 2021-06-10:
thanks for the positive feedback and thanks for that idea. i'll have a look at it (i'm not that famliar with timeshift and just have a rought idea what it does) and will see how well it would fit in and which advantages and disadvantages it might have ... what is the main usage scenario: to timeshift backup the home dir or the complete system?

stsichler, 2021-06-10:
The purpose of Timeshift is to create and restore snapshots of the _System_ without affecting user data in home directories (its's similar to what is called "system restore points" in Windows).
This is done by using btrfs subvolume snapshots of the '@' subvolume, where the whole system files reside in, whereas the '@home' subvolume containing all home directories always stays untouched.

There are also hooks into package managers for automatic creation of snapsots before package install/remove actions, which is also very handy, besides of regular schedule-based snapshots and manual snapshots. 
See [timeshift-autosnap](https://gitlab.com/gobonja/timeshift-autosnap) and [timeshift-autosnap-apt](https://github.com/wmutschl/timeshift-autosnap-apt).
 
Timeshift is very similar to [Snapper](http://snapper.io/) from OpenSUSE. Also for Snapper, there is a hook into package management for automatic snapshot creation on apt transactions, see [apt-btrfs-snapper](https://github.com/sweptr/apt-btrfs-snapper).