velvet-os/imagebuilder #20: Display static and no boot on sku0 [Kukui Chromebook Duet]

ClockworkElijah54, 2021-07-25:
Hello, I tried to boot focal on my Chromebook Duet and the display never turned on besides some static. The reason I mentioned sku0 is because of the Cadmium discord, which instructed me to run a command that revealed a difference in display panels, the cause of the issue. It was resolved using a kernel patch.

hexdump0815, 2021-07-25:
hi, i have read it in the cadmium context. the kernel used here does not have the sku0 dtb in yet. i can try to add it and prepare you a new kernel tar.gz in a few days. do you have another machine at hand which you can use to add such a new kernel to the written image?

best wishes - hexdump

ClockworkElijah54, 2021-07-25:
Yes, I do. Thank you for asking, and message me back when you get it ready :)

ClockworkElijah54, 2021-07-28:
Hello, I'm just curious if you've made any progress on this issue? Thank you

hexdump0815, 2021-07-28:
i'll not get to it before the weekend - i'll ping you here when i have something ready - so stay tuned

ClockworkElijah54, 2021-07-28:
Ah ok, thank you

hexdump0815, 2021-07-29:
what does "cat /proc/device-tree/model" say exactly in chromeos for your duet - "MediaTek krane rev3/rev4 sku0 board" or "MediaTek krane sku0 board"?

ClockworkElijah54, 2021-07-29:
In crosh or root shell? I tried crosh and it said "Unknown command: cat"

hexdump0815, 2021-07-30:
in the root shell

hexdump0815, 2021-07-31:
can you please give this one a try? - https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/download/5.13.2-stb-mt8%2B/5.13.2-stb-mt8+.tar.gz

just do the following (assuming sdcard to be your sdcard device with the kukui image on it) on some linux machine:
```
mount /dev/sdcardp4 /mnt
mount /dev/sdcardp3 /mnt/boot
cd /mnt
tar xzf  5.13.2-stb-mt8+.tar.gz
dd if=boot/vmlinux,kpart-5.13.2-stb-mt8+ of=/dev/sdcardp1
umount /mnt/boot /mnt
```
and then boot your duet with it and let me know if it will work

good luck and best wishes - hexdump

ClockworkElijah54, 2021-07-31:
Ah ok, thank you :D I missed your last message, sorry

ClockworkElijah54, 2021-07-31:
It worked, it booted fine with no issue, but the screen rotation is still wrong, just so you know

hexdump0815, 2021-07-31:
on krane screen rotation of xorg is corrected on first boot, so should be correct on second one, but maybe your model string in the dtb is so different that the script does not catch it :)

ClockworkElijah54, 2021-07-31:
Ah ok, no problem, can you close this or is there more to this that needs testing?

hexdump0815, 2021-07-31:
if it works for you now i think we can close it ... btw. just updated the images and they now contain that kernel with more dtb files by default :)

best wishes and thanks for testing - hexdump

ClockworkElijah54, 2021-07-31:
Cool, thanks

closed 2021-07-31 by ClockworkElijah54