velvet-os/imagebuilder #21: wake up from deep sleep doesn't work on the snow builds dated July 25 2021.

stiltgit, 2021-07-26:
wake up from deep sleep does not work on the snow builds dated July 25 2021, s2idle works fine, though.

hexdump0815, 2021-07-27:
interesting - i tested it and for me it worked (booted from sd card and rev4 snow chromebook). did you boot from sd card or usb (if that even works)? do you have a rev4 or rev5 snow chromebook?

best wishes - hexdump

stiltgit, 2021-07-27:
I booted from sdcard -usb boot doesn't work- and it's definitively a rev4 snow chromebook, and apart from not waking up from deep sleep, your bullseye image is working like a charm. 

perhaps the issue is related to the built in 3g modem board? (XE303C12-H01)

As a workaround I set sleep mode back to s2idle.

hexdump0815, 2021-07-27:
i think mine is h01 too. i'll retest it during the next days. btw. there is a small typo in the /etc/rc.local of the snow images - it does not harm, but maybe you want to fix it locally: https://github.com/hexdump0815/imagebuilder/commit/8fb853375570cf34a56dd55a0523edf869a1d9a2

stiltgit, 2021-07-27:
thx, I modified the rc.local file already in order to set s2sleep there.

what was interesting, the uboot versions starting with the June images are more picky related to sdcard timings at boot time.

I had the March image booting from a sdcard flawlessly, however starting with the June builds that specific sdcard is not initialized any more and the further boot process is stuck.

top of the boot screen:

```
Net: No ethernet found.
Hit any key to stop autoboot: 0 
mmc_init: -5, time 13

With a different sdcard, no boot problems with the June and July images. However, maybe similar sdcard initialization problems could be the reason for wake up problems from deep sleep?

hexdump0815, 2021-07-27:
interesting again: i changed u-boot versions in april, as with the old one the rev5 snow of a friend die not boot at all and with the new (actually older) one it worked - looks like with yours its the other way around - can you maybe try to use the v2018.11-cbe one referenced here: https://github.com/hexdump0815/imagebuilder/commit/305146a0fedcaa6bd2a494b68abdf28ed8066b21#diff-03196274b2b0329a66fe27567e1184239cf6b6e8d5a76e48933cfd1260cbc026 ... maybe i should add it as an alternative u-boot in the /boot/extra dir in case it works better for you

hexdump0815, 2021-07-30:
here is the direct link to the alternative u-boot, which worked for another user of the images already: https://github.com/hexdump0815/u-boot-chainloading-for-arm-chromebooks/releases/download/v2018.11-cbe/uboot.kpart.cbe.gz ... just download it, gunzip it and dd it to the first partition on the sd card ... it would be interesting to see if suspend/resume works better with that for you ... i retested suspend/resume on my rev4 h01 and on the rev5 of a friend and it works fine on both btw.

stiltgit, 2021-07-31:
thank you for the alternative u-boot for the 2021/07/25 bullseye image, it works again with all of my sdcards.

... but no success with resume from deep sleep - syslog reports deep sleep
I also tried service lightdm stop and entering suspend from console via systemctl suspend, even entered single user/mainenance mode via systemctl emergency before entering suspend, with no success to resume from deep sleep

no abnormal entries in syslog/kernlog during entering deep sleep, power button led flashes as expected after entering deep sleep, when trying to wake up, the power button led will cease flashing and turns on, but no further progress occurs.

One more note about rc.local, there's a missing colon or true in the last condional statement, as all commands there are commented out

```
Jul 31 06:21:10 changeme NetworkManager[2822]: <info>  [1627712470.8366] manager: sleep: sleep requested (sleeping: no  enabled: yes)
Jul 31 06:21:10 changeme ModemManager[2920]: <info>  [sleep-monitor] system is about to suspend
Jul 31 06:21:10 changeme NetworkManager[2822]: <info>  [1627712470.8368] device (mlan0): state change: disconnected -> unmanaged (reason 'sleeping', sys-iface-state: 'managed')
Jul 31 06:21:10 changeme NetworkManager[2822]: <info>  [1627712470.8434] device (mlan0): set-hw-addr: reset MAC address to 5C:F8:A1:71:C3:93 (unmanage)
Jul 31 06:21:10 changeme NetworkManager[2822]: <info>  [1627712470.8677] device (p2p-dev-mlan0): state change: disconnected -> unmanaged (reason 'sleeping', sys-iface-state: 'managed')
Jul 31 06:21:10 changeme NetworkManager[2822]: <info>  [1627712470.8695] manager: NetworkManager state is now ASLEEP
Jul 31 06:21:10 changeme systemd[1]: Reached target Sleep.
Jul 31 06:21:10 changeme systemd[1]: Starting Suspend...
Jul 31 06:21:10 changeme bluetoothd[3250]: Endpoint unregistered: sender=:1.66 path=/MediaEndpoint/A2DPSink/sbc
Jul 31 06:21:10 changeme bluetoothd[3250]: Endpoint unregistered: sender=:1.66 path=/MediaEndpoint/A2DPSource/sbc
Jul 31 06:21:10 changeme wpa_supplicant[2840]: nl80211: deinit ifname=mlan0 disabled_11b_rates=0
Jul 31 06:21:12 changeme systemd-sleep[3711]: Suspending system...
Jul 31 06:21:12 changeme kernel: [ 1003.816797] PM: suspend entry (deep)
```

hexdump0815, 2021-07-31:
oh - thanks about that hint regarding this other rc.local error - i should have been doing some more testing :)

your suspend problems seem to be related to your machine somehow - so far it works for mine, the rev5 of a friend and i got another report that it works ... you might try to switch to the legacy kernel and see if it works there (its in /boot/extra - just untar it in / and wirte the kernel image to the first partition) ... there seem to be quite a few variations of snow chromebooks regarding hardware and firmware - i had one which i was not able to get running stable at all, not even with the legacy kernel, but in chromeos it was stable - no idea what the problem was there

please let me know if you find out more about your suspend problem and worst case you'll have to use s2idle

hexdump0815, 2021-07-31:
three more ideas:
- give the hack i had to use on the nyan chromebook a try: https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_nyan/extra-files/lib/systemd/system-sleep/mrvl-and-edp-reload
- explicitely set the brightness back to some value on resume in that file
- login to wifi before suspending, suspend, resume and see if you can still login or the session comes back - this worked quite reliable for me in such cases where the system was still alive, but just did not reinitialize the display again properly after resume

jenneron, 2024-03-03:
@stiltgit, hi

> perhaps the issue is related to the built in 3g modem board? (XE303C12-H01)

do you have a 3g variant of snow? would you be able to test a few things? we have some chances of getting it to work

stiltgit, 2024-03-03:
> @stiltgit
> do you have a 3g variant of snow? would you be able to test a few things? we have some chances of getting it to work

Yes, I've a 3g variant of snow, respectively two of them (snow marcus f-e 9919 and snow rila f-e 5099 

While I could test a few things, I have to state that 3g support of mobile operators has been dropped over here meanwhile. 2g support stays active for a while, it usually hardly delivers usable data throughput, though.

As far as the original issue is concerned (deep sleep not working on one of my two snows), it has been solved by postmarketOS and its kernel (mentioned over there: https://github.com/hexdump0815/imagebuilder/issues/78)

However, one issue with postmarketOS / that kernel is that the battery does not charge while the device is in suspend state (I can reproduce that for both of my snows)

jenneron, 2024-03-04:
> As far as the original issue is concerned (deep sleep not working on one of my two snows), it has been solved by postmarketOS and its kernel

huh, I don't think we did anything to fix it. it can regress again. if you want to make it more reliable you would probably have to participate in kernel testing (and maybe some debugging in case of anything)

> However, one issue with postmarketOS / that kernel is that the battery does not charge while the device is in suspend state (I can reproduce that for both of my snows)

interesting. could you open an issue at https://gitlab.com/postmarketOS/pmaports/-/issues?

anyway, i want to try getting modem to work. could you please take `dmesg | grep -i usb` from pmOS and upload it here? If modem doesn't get initialized I will try to fix it

stiltgit, 2024-03-04:
There you are:

[dmesg-usb-snow.log](https://github.com/hexdump0815/imagebuilder/files/14478826/dmesg-usb-snow.log)

jenneron, 2024-03-04:
@stiltgit, thanks

try this kernel build:

[linux-postmarketos-exynos5-6.1.56_p20240304094918-r0.apk.zip](https://github.com/hexdump0815/imagebuilder/files/14479752/linux-postmarketos-exynos5-6.1.56_p20240304094918-r0.apk.zip)

```
cd ~/Downloads
unzip linux-postmarketos-exynos5-6.1.56_p20240304094918-r0.apk.zip
apk add --allow-untrusted ./linux-postmarketos-exynos5-6.1.56_p20240304094918-r0.apk
```

(sending as .zip archive because github doesn't allow to send .apk files)

stiltgit, 2024-03-04:
3g modem detected

```text 
[dmesg-usb-snow-2.txt](https://github.com/hexdump0815/imagebuilder/files/14480305/dmesg-usb-snow-2.txt)
~ $ lsusb 
Bus 001 Device 001: ID 1d6b:0002 Linux 6.8.0-rc2-next-20240201-postmarketos-exynos5 ehci_hcd EHCI Host Controller
Bus 001 Device 002: ID 0424:3503  
Bus 001 Device 003: ID 2232:1037 Generic WebCam SC-03FFM12339N
Bus 001 Device 004: ID 1410:a023 Novatel Wireless, Inc. Novatel Wireless Gobi3000
Bus 002 Device 001: ID 1d6b:0001 Linux 6.8.0-rc2-next-20240201-postmarketos-exynos5 ohci_hcd USB Host Controller
Bus 003 Device 001: ID 1d6b:0002 Linux 6.8.0-rc2-next-20240201-postmarketos-exynos5 xhci-hcd xHCI Host Controller
Bus 004 Device 001: ID 1d6b:0003 Linux 6.8.0-rc2-next-20240201-postmarketos-exynos5 xhci-hcd xHCI Host Controller
~ $
```

jenneron, 2024-03-04:
@stiltgit if it is okay for you can you give me a sign off for the patch? the format is `Tested-by: Name Surname <email>`

jenneron, 2024-03-04:
also, try this:

```
sudo apk add modemmanager
sudo rc-service modemmanager start
mmcli --list-modems
```

stiltgit, 2024-03-04:
Alpine's nmcli does not recognize --list-modems (nor -L or -M)

There are no /dev/ttyUSBx devices and I can't see any modules related to the 3g modem in lsmod output.

reboot does not bring any other lines in dmesg output about the modem.
(sudo rc-service modemmanager start applied manually after reboot)
modemmanager-gui does not find a modem, trying to create a mobile broadband connection through nm-coonection-editor does not list a selectable device.

[lsmod-2.txt](https://github.com/hexdump0815/imagebuilder/files/14480832/lsmod-2.txt)

jenneron, 2024-03-04:
> Alpine's nmcli does not recognize --list-modems (nor -L or -M)

you need `mmcli` with 2 lower case `M`

stiltgit, 2024-03-04:
oops, but that delivers "No modems were found"

double-checked on both of my snow devices

jenneron, 2024-03-04:
can you upload dmesg?

jenneron, 2024-03-04:
There is an arch wiki page about these modems. https://wiki.archlinux.org/title/Gobi_Broadband_Modems#gobi_loader

I think we would need to grab some firmware files from chrome os for this to work, and we would need this `gobi_loader`. I tried to build it locally and it seems okay

```
sudo apk add alpine-sdk git
git clone https://github.com/kicer/gobi_loader
cd gobi_loader
make
gobi_loader /dev/ttyUSB1 <path to firmware>
```

but we need to find firmware files firstly

once everything is done i could package gobi_loader and firmware to pmOS, so it will be working on snow by default

jenneron, 2024-03-04:
meanwhile the kernel patch works and makes modem detected, it's cool

stiltgit, 2024-03-04:
After visiting your cited page about gobi loader for Arch linux, I managed to get the ttyUSB devices created:

```text
~ $ modprobe qcserial
dmesg output:
[   64.535092] usbcore: registered new interface driver usbserial_generic
[   64.535148] usbserial: USB Serial support registered for generic
[   64.539636] usbcore: registered new interface driver qcserial
[   64.539698] usbserial: USB Serial support registered for Qualcomm USB modem

~ $ lsusb -d 1410:
Bus 001 Device 004: ID 1410:a023 Novatel Wireless, Inc. Novatel Wireless Gobi3000
~ $ echo "1410 a023" | sudo tee /sys/bus/usb-serial/drivers/qcserial/new_id 
dmesg output:
[  139.871909] qcserial 1-2.1:1.1: Qualcomm USB modem converter detected
[  139.872205] usb 1-2.1: Qualcomm USB modem converter now attached to ttyUSB0
[  139.880211] qcserial 1-2.1:1.2: Qualcomm USB modem converter detected
[  139.880498] usb 1-2.1: Qualcomm USB modem converter now attached to ttyUSB1
[  139.887230] qcserial 1-2.1:1.3: Qualcomm USB modem converter detected
[  139.889471] usb 1-2.1: Qualcomm USB modem converter now attached to ttyUSB2

~ $ echo "1410 a024" | sudo tee /sys/bus/usb-serial/drivers/qcserial/new_id # needed after flashing the firmware later

```

[dmesg-usb-snow-3.txt](https://github.com/hexdump0815/imagebuilder/files/14482149/dmesg-usb-snow-3.txt)

jenneron, 2024-03-04:
nah, this is not how it is supposed to work. i'm trying to fix it in kernel, it is just that building kernel on my current machine takes a lot of time

jenneron, 2024-03-04:
@stiltgit try this

[linux-postmarketos-exynos5-6.1.56_p20240304142612-r0.apk.zip](https://github.com/hexdump0815/imagebuilder/files/14482887/linux-postmarketos-exynos5-6.1.56_p20240304142612-r0.apk.zip)

qcserial should find modem automatically

please upload these:
- dmesg
- lsmod
- mmcli --list-modems

also, please tell me if qcserial gets loaded automatically

jenneron, 2024-03-04:
If it doesn't work i will try to patch modemmanager to add it

jenneron, 2024-03-04:
> the only manual step necessary was 'sudo rc-service modemmanager start'

if you install gnome desktop installation of pmOS it should come with modemmanager pre-installed and you will be able to access modem in GUI

> I was able to create a "mobile broadband connection" via networkmanager nm-connection-editor and this connection transfers data at the modest speed I expected (edge 2.5g as 3g coverage has been abandoned over here).

nice

> after disabling mobile broadband connections in networkmanager, and enabling it again, re-registering in the mobile network seems to be stuck

it might be worth reporting to modemmanager

Thank you for the testing. We will ship these changes in the next kernel upgrade in pmOS

jenneron, 2024-03-04:
btw if you install gnome it will be also worth testing GPS in gnome-maps. this modem may be capable of GPS through modemmanager