velvet-os/imagebuilder #22: unable to boot XE303C12-A01SE (SNOW)

Linux123123, 2021-07-30:
I am not sure if the problem is model/revision, but every time I try to boot the 210725-02 or 210613-04 image (haven't tried older ones) I come straight to some kind of command line? I think it is u-boot? well then it throws a lot of errors about no Ethernet and no boot file. 
[image]

Very sorry for low quality ;(

Then when it finally stops it is a command line where you type without seeing what you type.....

So if I am correct (and correct me if I am wrong) the u-boot can't find the kernel?

hexdump0815, 2021-07-30:
hi @Linux123123 - the problem here is that u-boot does not see the mmc/sd card and thus cannot boot the kernel. i changed the u-boot version around may as the old (actually newer one: v2018.11) did not work on a rev5 snow of a friend and the new (actually older u-boot v2017.09) worked, but it looks like this gives problems for some rev4 systems now. i have just built a new image, which has the other u-boot as an alternative in the "extra" folder of the boot partition, so that its easier to try it - i plan to upload the nwer version later today or tomorrow and will let you know then ... i think the chances are good that the alternative u-boot will work for you snow ... in case you are impatient you might try to gunzip this one and write it via dd to the first partition of your sd card: https://github.com/hexdump0815/u-boot-chainloading-for-arm-chromebooks/releases/download/v2018.11-cbe/uboot.kpart.cbe.gz

Linux123123, 2021-07-30:
Yes it indeed works perfectly with the newer `u-boot` version. Thanks for the quick reply.

Are there any plans to release minimal images? without any DE

closed 2021-07-30 by Linux123123

hexdump0815, 2021-07-30:
that sounds good - thank you for testing ... btw. does suspend/resume work for you?

i do not plan to release any other images - even the existing ones are nearly too much work for all of the different systems ... i see those images as a starting point from where everyone can go where they want to go to ... i think with a few apt-get remove it should be possible to make the setup more minimal relatively quickly

Linux123123, 2021-07-30:
> that sounds good - thank you for testing ... btw. does suspend/resume work for you?

Yes it works. 👍🏿 
 
 Fair enough, I see your point. Either way if someone really wants something changed, they can just build images themselves :)

Well that being said I am waiting for better building instructions (don't feel rushed). Also you could setup a CI to auto build for you at some point.

hexdump0815, 2021-07-31:
happy to hear that it works now ... yes there are so many things possible to improve the framework and the images, but all is limited by time as this is just some spare time project ... so things will progress, but it will take quite a bit of time ...