velvet-os/imagebuilder #25: My T9 tv box does not boot with your rk33xx images

jjcampo, 2021-10-31:
Hi @Hexdump

I have a leelbox mxq max, with has a T9 board and works with your dtb booting armbian, manjaro, openwrt, etc. I have tried your latest debían and ubuntu images and they boot but cant get to kernel  and CPUs are not found. I am using the T9 dtb file included.

Thanks for your work.

jjcampo, 2021-11-02:
I have been able to boot with rk3328-h96max.dtb , so despite my board is a T9 with rtl8723 wifi  the included dtb doesnt work for me.

hexdump0815, 2021-11-06:
that sounds good - the name of a tv box often does not say too much about what is actually in it ... it might be worth to have a look at this other very well maintained rk3328 image: https://forum.armbian.com/topic/17597-csc-armbian-for-rk3318rk3328-tv-box-boards/

jjcampo, 2021-12-08:
I have changed the uboot on your rk3328 image, with a roc-3328-pc uboot, and now all dtbs work without any problem. With rk3328-box. dtb everything works sound, hdmi, ethernet and wifi!. Also dpkg errors with kernel have gone away.
Only the cpu frequency is wrong with an 1.3Ghz maximum instead of 1.5Ghz.

hexdump0815, 2021-12-26:
interesting and good find with the roc-3328-pc u-boot - which one did you use exactly? - thanks for sharing this ... regarding the frequency: i think those rk3328 tv boxes are fully reliable up to about 1.3ghz anyway - at least if you put full load on them and running with full load might also result in thermal throttling after a while ... some of them might run up to 1.5ghz, but in my experience they are mostly only stable up to light to medium load levels

jjcampo, 2021-12-27:
This u-boot is from balbes150 thread "Single Armbian image for RK + AML + AW (aarch64 ARMv8)" in the armbian forum. In this repo there are some uboot images.

hexdump0815, 2022-01-03:
ok - thanks for the info - i'll have to have a look at it when i find some time ...