velvet-os/imagebuilder #28: No CIFS support in 5.10.25-stb-av7+ Kernel?

thinkbig1979, 2021-12-29:
Hi there, I used the Allminner H3 image with kernel 5.10.25-stb-av7+ succesfully on my Beelink X2 TV box and I've been very happy with it so far. However today I tried to mount a SMB share using CIFS and got an error that CIFS was not supported by the system.
All CIFS packages are installed. I double checked that.

Further research led me to the modprobe command, which revealed that no CIFS modules were found on my system and other commands also seemed to indicate that it simply is not part of this kernel. 

Can that be correct or am I overlooking something?
I'd really like to be able to mount CIFS shares, so any pointers would be welcome. I'm happy to switch out kernels if that will help, but I don' t know to which one I can switch. 

Thanks in advance.

hexdump0815, 2021-12-30:
hello,

looks like cifs support is not enabled in the kernel. i might consider enabling it in the future but for now the only option might be to build your own kernel following my notes from https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/master/readme.av7 and enable CONFIG_CIFS etc. - a good oprtunity to get into the kernel build business :)

thinkbig1979, 2022-01-18:
> hello,
>
> looks like cifs support is not enabled in the kernel. i might consider enabling it in the future but for now the only option might be to build your own kernel following my notes from [...]

Unfortunately this is not a task I can take on at the moment, so I looked for another solution. I ultimately found an Armbian image for another H3 based device that had a pretty recent kernel with CIFS support, and which worked perfectly well for my device. I ended up using that image.

Thanks again for making these tools available.

hexdump0815, 2022-01-22:
if armbian is working well for your board its definitely a good choice