velvet-os/imagebuilder #29: Howto write uboot in Acer C810 2Gb

jjcampo, 2022-01-03:
Hi @hexdump0815 

I understand have to write a file in the extra directory in order to get this 2Gb chromebook booting from sd, but it's not clear to me exactly how. I used dd to write the file to sda1 partition without any argument. The chromebook flashes 5 times and then gets to a quiet black screen and then I type 1 and enter but nothing is happening.

The image is 211411-1 debian.

hexdump0815, 2022-01-03:
i think it is not trivial to enther the number at the right time without seeing anything - best is to simply edit the default in the extlinux/extlinux.conf file on the first partition to the one you need for your system.

good luck and happy new year!

jjcampo, 2022-01-04:
I have edited extlinux and remove HD and blaze and now my chromebook boots till desktop. Many thanks hexdump for your excellent work and support!

hexdump0815, 2022-01-04:
@jjcampo i'm happy to hear that you have it working now and thanks a lot for the thanks :)