velvet-os/imagebuilder #31: Notes on image 210803-01 on odroid-u3

antonialoytorrens, 2022-02-02:
First of all, as usual, thank you very much @hexdump0815 for all the effort you are putting into your [custom stable mali kernel](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel) and this imagebuilder :)

However, I haven't seen any newer Odroid U3 image releases for a while, so I'm putting mine below.
I haven't installed anything, just updated the base Userland of your image from Debian 10 (buster) to Debian 11 (bullseye) and the kernel from your [exy+ custom stable mali kernel](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases/tag/5.15.12-stb-exy%2B).
Maybe the image I provide here is a bit larger than usual due to cached files and repositories (?)

Furthermore, I have seen that `/etc/apt/sources.list` points the repositories to `stable` instead of the actual release. That could be a problem if the image is a little old or from a previous stable release. I replaced it for `buster` at first, and then updated to `bullseye`, as said before.

In terms of usability, not much have changed after upgrading. It's an old device after all (~8 years) and cannot expect brilliant performance. I use this device quite often and I use it as a spare mini computer for weekly (non-programming) tasks.

However, when inspecting the kernel logs, some errors appeared due to sensors I think, because they are not identified in kernel. If I recall correctly, all of the previous kernel releases had this error (dmesg below).

But other than that, I can work normally without any extra issue. It's all working out of the box even with the upgraded 5.15.x kernel and Debian 11 userland, very pleasant experience.

Image here:
[odroid_u3-armv7l-debian11.img.xz](https://drive.google.com/file/d/1CG7CU9cTkKFAtyimtrlYrW3N6gUqB7yB/view?usp=sharing)

Dmesg here:
[dmesg-kernel-5.15.12.txt](https://github.com/hexdump0815/imagebuilder/files/7983009/dmesg-kernel-5.15.12.txt)

<br/>
All the best, great work as always!

hexdump0815, 2022-02-06:
hi antony,

thanks a lot for your efforts ... there is actually a debian bullseye image
for the odroid:
https://github.com/hexdump0815/imagebuilder/releases/tag/210803-01 ... it
is from short before the official release i think, but it will update well
with a dist-upgrade - i just installed my odroid x2 based on it today :)
... i also built a newer kernel recently:
https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases/tag/5.15.12-stb-exy%2B
... i just need to find some time to spin new images maybe, but so far the
last images are still quite useable i think

a lot of thanks once more and best wishes - hexdump

antonialoytorrens, 2022-02-07:
I noticed the 5.15.12 kernel, and my image has this exact version in it, however I didn't know anything about the bullseye image, so thanks for that!

You have a point, aside from a newer kernel version, your latest images still works as expected. Thanks to you ;)

Best,
Antoni Aloy Torrens

jon-uk, 2022-04-28:
@hexdump0815 / @antonialoytorrens  - need to build linux-headers for a 5.10.x or a 5.15.x kernel, as I get an error when running **sudo apt-get install linux-headers-$(uname -r)** 

E: Unable to locate package linux-headers-5.10.45-stb-exy
E: Couldn't find any package by glob 'linux-headers-5.10.45-stb-exy'
E: Couldn't find any package by regex 'linux-headers-5.10.45-stb-exy'

Is it readily-available as a download from some repository please?
Or perhaps a guide or pointers to links, on how to build headers for odroid-u3?

Thanks,
J

hexdump0815, 2022-05-03:
@jon-uk - i think best would be to build your own kernel on the system following the notes from https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/master/readme.exy - then hopefully the kernel headers will be properly in place ... i'll add a note regarding the kernel headers to the todo as well for later

closed 2025-07-31 by antonialoytorrens