velvet-os/imagebuilder #34: PowerVR Vulkan driver

EMLommers, 2022-03-07:
There was a new release of open source vulkan driver for PowerVR.
see: https://lists.freedesktop.org/archives/mesa-dev/2022-March/225699.html

Can you add this in the new version for the 8173 Oak version?

hexdump0815, 2022-03-08:
hi,

i'm aware of this and will include it as soon as its somewhat useable, but the current state is very basic and not really useable for anything at all right now - i guess it will be about half a year to a year until it has reached a state where it might start to be basically useful ... if that project does not disappear in the meantime (which sadly happens to such projects from time to time as well, but i hope that this one will make progress)

best wishes - hexdump

EMLommers, 2022-03-08:
Hi,

Is a bit basic and bit working  better than nothing at all? If it improves performance it has at lease some use. Performance without GPU sucks big time, won't you agree?

rgrds ,EML

hexdump0815, 2022-03-12:
i did not look closer at it, but the way i read the announcement it looks like so far it can only draw some triangles in a kms test app - so still quite far away from being able to drive xorg or a wayland session which both need a lot of properly working functionality

EMLommers, 2022-03-16:
Thanks! Let's wait and see.....

a-priestley, 2022-09-13:
Just wanted to chime in and add that I am also waiting eagerly for this capability. But I'm not well-versed in how to gauge the progression of kernel-level developments. 
Judging by the volume of activity [here](https://gitlab.freedesktop.org/frankbinns), it seems like things are progressing very nicely, but I have no idea how to make even an educated guess as to when it will be ready.

hexdump0815, 2022-09-13:
@a-priestley - i think its still quite a bit of way to go until this starts to be useable - i definitely have this effort on my radar and will try to integrate it into the mt8173 tree as soon as it gets useable - my current assumption is that this might be early or mid next year ...

EMLommers, 2023-07-30:
In your opinion is their gitlab (frankbinn) povervr driver (review v4)  good enough to embed in your next build?
Together maybe with beaglebone build of Mesa driver for pvr?

hexdump0815, 2023-07-31:
as far as i understand there is still a bit of way to go until the driver starts to be useable - there is only a very minimal set of vulkan extensions implemented and a lot is missing to get zink running on top of it ... also it requires quite a few extra patches on top - so for now i'll wait until it starts to be useable before thinking about including it into my oak kernel builds ... see also: https://oftc.irclog.whitequark.org/linux-cros-arm/2023-06-25#32258477

EMLommers, 2023-08-21:
What I read is TI is referring users to use
https://git.ti.com/cgit/graphics/ti-img-rogue-driver/tree/?h=linuxws/kirkstone/k6.1/23.1.6404501

It has also the config for Mt8173.
Why would TI point users to use this driver for Rogue GPU?

EMLommers, 2023-08-22:
Sure.
It is not a TI driver alone, it is an img rogue driver,  (made by or for imgtec). I have checked the code and using same rogue firmware as chromeos, no new build firmware needed.
The 5.10 kernel driver is only for TI AM62x and Mt8173.
The 6.1 kernel has more TI devices.

I was planning to build the driver for 5.10 on OAK,  have copied the firmware files from CrOS already and they match to this driver.
The driver version is 1.18, which is 1 version over latest OAK CrOS recovery image, which is 1.17

After closely checking the code, I have 100% confidence, this is a working driver. I will use your script for 5.10, OAK, include driver and replace kernel, driver (incl. Mesa Img).

The driver is some kind of Middleware between kernel, fb llvm driver and GPU.  It is a better driver than VisionFive2 img rogue driver for risc. 
Latest version was 1.17 as well, so assuming it is same driver as well, but 1 version up.

a-priestley, 2024-02-05:
Some news.. the powervr kernel driver for this device is in 6.8, of which there are now release candidates available (I believe release is slated for March). AFAIK this does not necessarily mean vulkan is in full working condition, but I think this is a big update still.

hexdump0815, 2024-02-05:
@a-priestley - i'm following the progress news there as well - as far as i know the main target for the development is some ti sbc and no longer the elm chromebook, so i expect it to not fully supported with what is in v6.8 and mesa now but hopefully soon as at least support for oak is at least still considered - see: https://lists.freedesktop.org/archives/dri-devel/2023-June/410995.html and https://gitlab.freedesktop.org/frankbinns/mesa/-/commit/7de7761c82d2b90692bd96e9340109ee1b89acbe

i currently do not have time to play around with the current code, but in case someone ellse is actually giving it a try, please add some info here

closed 2025-11-15 by hexdump0815