velvet-os/imagebuilder #35: Optional script to update newer kernel

EMLommers, 2022-03-09:
Hi,
Is there an optional script to update/build a newer kernel release and  write the first partition to run this kernel?
From the ISO's you build for ARM and Intel, an kernel build script and installer. (I like to test new Vulkan PVR driver)
Apollolake has same issue as cannonlake, only RW from chromebox,but like uboot/depthcharge.

Thnks,
Ernst

hexdump0815, 2022-03-12:
i have no extra scripts for installing newer kernels - the idea is to give a basically running system and enough room (i.e. two kernel partitions to play around with when using the proper depthcharge priority valies) for people to work from there

aneeshlingala, 2022-06-01:
I think you can use dd to image the two kernel partitons, and get 2 images on the live usb image and flash those two images using dd. You can use this command:` dd if=KERNELPAR1 of=/home/linux/Downloads/KernelOne.img bs=1M status=progress `make sure to replace KERNELPAR1 with your 1st kernel partiton, on the live USB/SD  and d`d if=KERNELPAR2 of=/home/linux/Downloads/KernelTwo.img bs=1M` and replace KERNELPAR2 with your second kernel partition on the live USB/SD