velvet-os/imagebuilder #36: systems/chromebook_kukui

BenjaminWegener, 2022-03-31:
Wanted to give some feedback...
After allowing dev_boot_usb, dev_boot_altfw and usb controller, the july kukui jmage boots on my FENNEL14

Have to try out what works and what not, giving reply soon

BenjaminWegener, 2022-03-31:
Burnt with balena etcher on some stuck fiy

BenjaminWegener, 2022-04-01:
so far some things i discovered:

after first reboot xfce gets tearing, cant see anything but pixelcrap
--> now only one thing helps: installing lxde an booting into it. tested gnome too, cant open firefox for example

BenjaminWegener, 2022-04-02:
now i updated ubuntu to 22.04 via update-manager
cant install firefox(uses snapd, error with squashfs, likely due to missing loopback support in kernel)
have no sound, havent dived into alsa ucm
using falkon as browser, giving me 30 fps on webgl aquarium, nearly the half of the 60 on cromeos

hexdump0815, 2022-04-03:
lets keep this issue open to give others a chance to find it too

first, thanks a lot for the feedback ... the 30fps for the aquarium sound reasonable, as mainline uses the open source panfrost gpu driver and chromeos the legacy mali driver and panfrost is not yet on par performace wise (and still has a few more bugs maybe as well)

the problems after first reboot sound strange to me - at least as long as nothing has been done inbetween (update etc.) ... if things were updated inbetween it might be that the env vars pointing to the nwwr self compiled mesa do not work anymore on newer ubuntu versions just like they do no longer work on debian bullseye - in this case the debian approach might be required (apt-mark hold for mesa and the xserver and directly linking the mesa dri dirs - i did not yet find a better way than this)

ucm files might require some adjustements per machine and the kernel config options for snap support are already on my todo list - i did not yet look at ubuntu 22.04 at all - i plan to include it into the imagebuilder framework once it is fully out

best wishes - hexdump

BenjaminWegener, 2022-04-03:
Aight, sounds reasonable. I updated to your 5.15. kernel,got only black screen,so reverted back to 5.13.

BenjaminWegener, 2022-04-08:
had some time, so was digging into the new kernel, 5.15. which is giving me black screen. i pressed power after 1 mnute an it went to shutdown, so i thought it would be possible that it boots normal. so here is the dmesg with black screen:
-----------------------------------------
[    0.000000] kernel: Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] kernel: Linux version 5.15.22-stb-mt8+ (root@aarch64) (gcc (Debian 10.2.1-6) 10.2.1 20210110, GNU ld (GNU Binutils for Debian) 2.35.2) #5 SMP PREEMPT Tue Feb 22 21:52:33 CET 2022
[    0.000000] kernel: Machine model: Google fennel14 sku0 board
[    0.000000] kernel: efi: UEFI not found.
[    0.000000] kernel: Reserved memory: created DMA memory pool at 0x0000000050000000, size 41 MiB
[    0.000000] kernel: OF: reserved mem: initialized node scp_mem_region, compatible id shared-dma-pool
[    0.000000] kernel: NUMA: No NUMA configuration found
[    0.000000] kernel: NUMA: Faking a node at [mem 0x0000000040000000-0x000000013fffffff]
[    0.000000] kernel: NUMA: NODE_DATA [mem 0x13f7c0e00-0x13f7c2fff]
[    0.000000] kernel: Zone ranges:
[    0.000000] kernel:   DMA      [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] kernel:   DMA32    empty
[    0.000000] kernel:   Normal   [mem 0x0000000100000000-0x000000013fffffff]
[    0.000000] kernel: Movable zone start for each node
[    0.000000] kernel: Early memory node ranges
[    0.000000] kernel:   node   0: [mem 0x0000000040000000-0x000000004fffffff]
[    0.000000] kernel:   node   0: [mem 0x0000000050000000-0x00000000528fffff]
[    0.000000] kernel:   node   0: [mem 0x0000000052900000-0x00000000545fffff]
[    0.000000] kernel:   node   0: [mem 0x0000000054700000-0x00000000ffdfffff]
[    0.000000] kernel:   node   0: [mem 0x0000000100000000-0x000000013fffffff]
[    0.000000] kernel: Initmem setup node 0 [mem 0x0000000040000000-0x000000013fffffff]
[    0.000000] kernel: psci: probing for conduit method from DT.
[    0.000000] kernel: psci: PSCIv1.1 detected in firmware.
[    0.000000] kernel: psci: Using standard PSCI v0.2 function IDs
[    0.000000] kernel: psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] kernel: psci: SMC Calling Convention v1.1
[    0.000000] kernel: percpu: Embedded 19 pages/cpu s39960 r8192 d29672 u77824
[    0.000000] kernel: pcpu-alloc: s39960 r8192 d29672 u77824 alloc=19*4096
[    0.000000] kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 [0] 6 [0] 7 
[    0.000000] kernel: Detected VIPT I-cache on CPU0
[    0.000000] kernel: CPU features: detected: GIC system register CPU interface
[    0.000000] kernel: CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] kernel: CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] kernel: CPU features: detected: ARM erratum 845719
[    0.000000] kernel: Built 1 zonelists, mobility grouping on.  Total pages: 1031424
[    0.000000] kernel: Policy zone: Normal
[    0.000000] kernel: Kernel command line: cros_secure console=tty1 root=PARTUUID=9c3ea286-c06d-7943-8f99-26cb5bb57712/PARTNROFF=3 rootwait ro fsck.fix=yes fsck.repair=yes net.ifnames=0 ipv6.disable=1 fw_devlink=off noresume 
[    0.000000] kernel: Unknown kernel command line parameters "cros_secure", will be passed to user space.
[    0.000000] kernel: Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] kernel: Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] kernel: mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] kernel: software IO TLB: mapped [mem 0x00000000fbe00000-0x00000000ffe00000] (64MB)
[    0.000000] kernel: Memory: 3977948K/4191232K available (13696K kernel code, 1506K rwdata, 5388K rodata, 3008K init, 437K bss, 213284K reserved, 0K cma-reserved)
[    0.000000] kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=8, Nodes=1
[    0.000000] kernel: rcu: Preemptible hierarchical RCU implementation.
[    0.000000] kernel: rcu:         RCU event tracing is enabled.
[    0.000000] kernel: rcu:         RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=8.
[    0.000000] kernel:         Trampoline variant of Tasks RCU enabled.
[    0.000000] kernel: rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] kernel: rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=8
[    0.000000] kernel: NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] kernel: GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] kernel: GICv3: 640 SPIs implemented
[    0.000000] kernel: GICv3: 0 Extended SPIs implemented
[    0.000000] kernel: GICv3: Distributor has no Range Selector support
[    0.000000] kernel: Root IRQ handler: gic_handle_irq
[    0.000000] kernel: GICv3: 16 PPIs implemented
[    0.000000] kernel: GICv3: CPU0: found redistributor 0 region 0:0x000000000c100000
[    0.000000] kernel: GICv3: GIC: PPI partition interrupt-partition-0[0] { /cpus/cpu@0[0] /cpus/cpu@1[1] /cpus/cpu@2[2] /cpus/cpu@3[3] }
[    0.000000] kernel: GICv3: GIC: PPI partition interrupt-partition-1[1] { /cpus/cpu@100[4] /cpus/cpu@101[5] /cpus/cpu@102[6] /cpus/cpu@103[7] }
[    0.000000] kernel: random: get_random_bytes called from start_kernel+0x478/0x65c with crng_init=0
[    0.000000] kernel: arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] kernel: clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[... cut ...]
[   12.123488] kernel: anx7625 4-0058: [drm:anx7625_bridge_get_edid] *ERROR* WARNING! edid check fail!
[   12.123521] kernel: anx7625 4-0058: [drm:anx7625_bridge_get_edid] *ERROR* Fail to read EDID: -22
[   12.159159] kernel: EDID block is all zeroes
[   12.159169] kernel: anx7625 4-0058: [drm:anx7625_bridge_get_edid] *ERROR* WARNING! edid check fail!
[   12.159188] kernel: anx7625 4-0058: [drm:anx7625_bridge_get_edid] *ERROR* Fail to read EDID: -22
[   14.379285] kernel: fuse: init (API version 7.34)
-----------------------------

closed 2022-05-17 by BenjaminWegener