velvet-os/imagebuilder #41: ℹ️ trogdor: coachz: Better Graphical Acceleration

Geofferey, 2022-04-20:
Been playing with the coachz trogdor convertible and your builds... I do have to say I am impressed and want to help 😁 

When trying to run graphical applications I noticed quite a bit of crashing and some reboots when running compositors mixed with other GL applications (e.g. Google Earth in Firefox). Executing `glxinfo -B` device reports 10MB of available video memory which is not right for linux distros. In my experience the reported video memory is usually equal to onboard RAM when using freedreno/zink.

Anyways I compiled the latest developmental Mesa libs (haven't tried on stable, yet) and graphical acceleration is working almost perfectly now (- chromium). In fact I don't even think I needed the GLES4 lib. More apps run now and meets requirements needed to run Blender, wow! 

Here are the flags I used to build Mesa on trogdor using meson. Took me a bit since I'm used to doing it on Android and I'm not sure if gallium zink is necessary but I left it.
     
     meson build --prefix /usr -D platforms=x11,wayland -D gallium-drivers=freedreno,zink,swrast,virgl -D vulkan-drivers=freedreno -D freedreno-kgsl=false -D dri3=enabled -D egl=enabled -D gles1=enabled -D gles2=enabled -D glvnd=true -D glx=dri -D libunwind=disabled -D osmesa=true -D shared-glapi=enabled -D microsoft-clc=disabled -D valgrind=disabled

 I may update this with more detailed instructions in future but I hope this helps someone.

hexdump0815, 2022-04-24:
@Geofferey - thanks a lot for reporting this and writing your approach down, it is very welcome - i have building and adding a newer mesa on my todo list, just currently taking a little break :)

best wishes and once more thanks a lot - hexdump

Geofferey, 2022-04-24:
@hexdump0815 no worries. Anything I can do to help. Keep an eye out on this 👀. I have plans to add instructions for cleaning up the dependencies afterwards and maybe a simple script to let it rip quick. Most `*-dev` packages can go but few must stay. As it stands `apt build-dep *` always makes an awful mess of things. I'm not sure if the extras like script would help you, but it definitely helps me learn and retain. Binaries can prolly just be packaged, not sure tho. 

SIDE NOTE/REQUEST:

One more thing, if you have telegram discord or something it would be great to speak with you live. I know you're taking break but I believe a little one on one help from you could go a long way... Specifically in regards to building the kernel natively on device. I have A LOT of time on hands to play and I also have experience patching kernels, etc. I'm coming from building Android here and the build environment makes me lazy 😜 Meaning all the manual labor involved in building the kernel is handled for me, I just find patches and use menuconfig. Honestly I really think I just need some pointers. 

P.S. 

I have used the Debian documentation to successfully compile kernel on x86 is past as well. However I'm not sure if this can be applied to Chromebooks specifically. Feel like I need an extra step to package it in a format Chromebook understands. 

https://www.debian.org/doc//manuals/debian-handbook/sect.kernel-compilation.pl.html

hexdump0815, 2022-04-24:
@Geofferey - the kernel build can be done following my notes which are always linked in the readme for the corresponding system - for trogdor it is for instance https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/master/readme.cbq ... as with all my "docs" its just a simple dump of the information, which i usually use for copy and paste :) ... i'm not sure if kernel packages are a good idea as some of the kernel builds are quite complex and i prefer to encourage people to build their own kernels from time to time ...

hexdump0815, 2022-04-25:
@Geofferey - just a few more thoughts (numbered for easier reference):

1.  you also asked about telegram/discord above: i prefer to stay with github issues and maybe pull request reviews for now as i do not have the bandwidth to deal with a lot of topics in realtime anyway - also this might make this little project feel to much like work after a while, which i would like to avoid
2. maybe just a quick try to summarize my goal for the imagebuilder framework: it should provide useable images for the supported systems which should just work - they do not have to be perfect but be a good starting point for people to work from and i prefer the whole setup to be rather simple than complex
3. in general i would like to keep automation for the image build at a healthy level - from other contexts i know that it can easily end up in over engineering and do not give as much back as it seems and it keeps people away from understanding what happens behind the scenes
4. as a result of the above i would prefer to keep the kernel and mesa build non packaged - this so far worked out very well and gave me a surprising flexibility dealing with around 40 partially completely different systems in this framework without having to fight with dependency issues etc. all the time - i would like to ship a working kernel and mesa in the images and instructions on how to build and install your own if needed - especially on arm systems this is essential, as things are moving quite fast in this area and being able to update at any time might improve functionality (like you saw with mesa)
5. regarding your mesa build: are you aware of https://github.com/hexdump0815/mesa-etc-build - this is where i have my special mesa build for the images - the doc is as bad as usual, but maybe we can use this as a base to make it cleaner - in the end i would prefer to have a drop in mesa installation in /opt/mesa which contains everything specially built without changing anything on the base system - also see https://github.com/hexdump0815/imagebuilder/blob/main/scripts/create-fs.sh#L229-L252 and https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_kukui/postinstall-chroot.sh (example)

best wishes - hexdump

hexdump0815, 2022-04-30:
btw. you said, that you are using my image on a coachz trogdor - so may i mark it is tested an working?

Geofferey, 2022-04-30:
@hexdump0815 Yes, it works for most part minus some quirks. I have actually compiled a list of what works and what doesn't. I will post an issue here coming soon.

hexdump0815, 2022-05-01:
@Geofferey thanks for the coachz issue - it is very good to track the actual state there for now ... i have added coachz as tested now: https://github.com/hexdump0815/imagebuilder/commit/e9f2b40bd23dd85cf10b7243162a8b92e4cc3591

hexdump0815, 2022-05-15:
@Geofferey - fyi: this is my basic plan to build the drop in /opt/mesa in the future so that it will bring its own libdrm and wayland-protocols with it and this way should allow to always build the latest mesa version even if some libs on the build/host system are not new enough - it is essentially what you described above just adjusted to install into /opt/mesa (/opt/mesanew for my testing right now) - https://github.com/hexdump0815/mesa-etc-build/commit/1d3821e2e172cb4fc211a9e961f9ea38e16fde17