velvet-os/imagebuilder #42: How can I run a simulation/build being on an architecture different from the one desired in the build

sc20ka, 2022-04-20:
https://github.com/hexdump0815/imagebuilder/blob/0c315a22d598e4f0d161c08b542441224d74343a/scripts/args-and-arch-check-functions.sh#L41

How can I run a simulation/build
being on an architecture different from the one desired in the build

hexdump0815, 2022-04-24:
@sc20ka - currently the framework is only supposed run natively and opening this up would make it quite a bit more complicated in some areas and i'm not sure if i really want to do this ... i'm providing images for all supported systems, so one option might be to simply use one of those images as a build system to build own images initially if no other native system is available

best wishes - hexdump

sc20ka, 2022-04-30:
i will qemu to emulate in win_64 platform arm64 using this [guide](https://gist.github.com/billti/d904fd6124bf6f10ba2c1e3736f0f0f7) and ubuntu 64 cloud server img from [here ](https://cloud-images.ubuntu.com/releases/focal/release/) [ubuntu-20.04-server-cloudimg-arm64.img](https://cloud-images.ubuntu.com/releases/focal/release/ubuntu-20.04-server-cloudimg-arm64.img)

closed 2022-05-06 by sc20ka