velvet-os/imagebuilder #45: Screen corruption on Lenovo ideapad 3 14 inch after reboot.

antonioaneiros, 2022-05-02:
Tried latest images on lenovo idepad 3 14 inch fennel14. Both ubuntu and debian show up fine on first login for user linux  but after reboot they show screen corruption after lightdm login without having made any changes to the system. Initial lightdm greeter always shows up fine though.

antonioaneiros, 2022-05-02:
Update: if I go to the console, create a new user, reboot and login with the user I just created, everything is OK on first login again but when I restart the system the same problem appears. Also, sometimes the mouse cursor is locked on lighdtm greeter screen.

hexdump0815, 2022-05-02:
@antonioaneiros - thanks a lot for reporting this - i think someone else reported something similar for the fennel 14 at https://github.com/hexdump0815/imagebuilder/issues/36 - i have no real clue yet where this problem comes from ... i think its time for a newer kernel and mesa - maybe i'll wait for v5.18 and create new images based on that one ...

antonioaneiros, 2022-05-03:
Hi, after reading comments on the systems/chromebook_kukui thread I realize I am experiencing the same problems and that the solution  proposed in that thread (using lxde) actually works fine. I also realize I forgot to thank you developers for your huge effort ;-)

hexdump0815, 2022-05-03:
@antonioaneiros - ok then at least you seem to have a somewhat working system for now - i hope that we will find proper fixes for the problem during the next weeks or months, maybe by using newer kernels or mesa versions. the goal is definitely to make it work properly on fennel 14 as well.

BenjaminWegener, 2022-05-08:
Updated my 5.13.19 custom kernel with the panfrost files from 5.17.4, got the xubuntu and even the wayland DE working without pixelcrap, touchpad gets disabled frequently during system running, probably due to not applied patches from cadmium. Working on it...

hexdump0815, 2022-05-09:
@BenjaminWegener - does the toucpad get disabled during use or from bootup? i saw a sometimes non working touchpad on kukui jacuzzi juniper - there it was simply not working at all after bootup from time to time and a reboot usually fixed the problem. which cadmium patches are you referring to?

btw. i'm currently trying to get newer kernels working for kukui-jacuzzi systems, but so far the display remains black for v5.16-v5.18-rc - i tried different things, but will most probably have to bisect it step by step ... it seems to be related to the mediatek drm and the anx7625 drm bridge driver as for kukui-krane (which uses a different bridge setup and chip) the newer kernels are working fine ...

BenjaminWegener, 2022-05-10:
@hexdump0815 
- the touchpad works for 5 seconds, after that the mousecursor remains still, the mouse itself gets invisible and is able to click somewhere else. If i switch to console back and forth, the cursor gets connected to the mouse again for 5 seconds. That is with kernel 5.13.19 without any cadmium patches from 5.13.2 but with the panfrist dir from 5.17.4.
- but thats not what i want, so the black screen remauns top priority. What did you try so far?

hexdump0815, 2022-05-10:
@BenjaminWegener - i tried the latest version of v5.15, 5.16, 5.17 and 5.18-rc and so far only v5.15 based on https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/blob/master/readme.mt8 gives a working display (but has a problem with display power management- but maybe older versions have this too) ... v5.16 works if i take driver/drm/mediatek from the above mentioned v5.15 tree ... i'm currently discussing this on the #linux-mediatek irc channel and there are some suggestions about what might cause this - will look into it soon and maybe do some git bisect on the weekend

BenjaminWegener, 2022-05-10:
@hexdump0815 thanks for your reply, gives me something to work with. Do you mean blank screen after screensaver turns on? I disabled it everytime i flashed the image.
I was thinking maybe the drivers/firmware/google folder gives blank screen on boot?

BenjaminWegener, 2022-05-11:
5.15 is still giving black screen, will try with drm/mediatek from 5.13

hexdump0815, 2022-05-11:
@BenjaminWegener - can you please try with adding the following to the kernel cmdline: "video=eDP-1:1920x1080@60 drm.debug=0xe fw_devlink=off" with the v5.15 kernel? - i hope i got the video more right - maybe better boot once with the working v5.13.2 and check the mode with xrandr

hexdump0815, 2022-05-11:
@BenjaminWegener - good news: i got the display working on my kukui-jacuzzi-kappa with v5.18-rc6 and the following patches (thaks to wens from #linux-mediatek): https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/tree/master/misc.cbm/patches/5.18.0 ... maybe you can give it a try if it works for you on your fennel14 as well - the kernel cmdline options from above seem to be no longer required, but keeping the drm debug one might be good for debugging

good luck and best wishes - hexdump

BenjaminWegener, 2022-05-12:
@hexdump0815 good work, thank you! Used 5.18rc6 from librekernel, screen is visible!
Got kernel panic because of missing btrfs driver, will convert /root to ext4. After that ill share my script for anyone else.

hexdump0815, 2022-05-12:
@BenjaminWegener - that sounds good ... little update: i have now also added patches (from someone from the #panfrost irc channel), which seem to make gpu frequency scaling work as well ... things are progressing well :)

BenjaminWegener, 2022-05-12:
@hexdump0815 thanks, are the panfrost patches also in the 5.18 folder?
In order to make the focal image from last year with kernel 5.13.2 boot the new 5.18 kernel, i needed to edit the docker-options.cfg, especially the line `CONFIG_BTRFS_FS=y`.
reporting about issues and performance soon. Thanks again!

hexdump0815, 2022-05-12:
@BenjaminWegener - yes all patches are in the v5.18 folder now

hexdump0815, 2022-05-13:
@BenjaminWegener - another little update: i found a hack to make dpms display resume work - its most probably not the proper solution, but it seems to work :) ... https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/commit/5f2bc76fcbf1bceb952327d2186248c9ffcf2d30 (its in the 5.18.0 patches dir now as well)

BenjaminWegener, 2022-05-14:
@hexdump0815 thank you, gets better every day. im out for a few days, but i have 2 problems, i want to fix next week.
Panfrost Driver doesnt support opengles 3.1, but should
No soundcard detected
I have faith, it will work by next weekend.

hexdump0815, 2022-05-15:
@BenjaminWegener - regarding gles 3.1 for panfrost (which i think the hardware does not support completely) you may give the env variables PAN_MESA_DEBUG=gl3 or MESA_GL_VERSION_OVERRIDE=3.1 a try (see: https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/5584) and depending on if you use 3.1 features supported by the hardware or not it might work or not ...

regarding audio support the best would be to open a new ticket about it and discuss how far it works or does not work yet in there - or even better: it would be nice if you could create a ticket similar to https://github.com/hexdump0815/imagebuilder/issues/44 or https://github.com/hexdump0815/imagebuilder/issues/47 for fennel14 with the current state and then we can discuss machine specific topics inside there and they are easy to find for others as well

BenjaminWegener, 2022-05-16:
@hexdump0815 i had a hunch, maybe the config was the problem - maybe the line `CONFIG_SND_SOC_MT8183_DA7219_MAX98357A=y` should be enabled, because the fennel14 sku0 dts imports specific the "mt8183-kukui-audio-da7219-rt1015p.dtsi".
I have to test it.

BenjaminWegener, 2022-05-17:
@hexdump0815  https://github.com/hexdump0815/imagebuilder/issues/49
this can be closed i think

hexdump0815, 2022-05-17:
thanks for creating the fennel14 issue - yes, lets close this one here now and continue there ...

closed 2022-05-17 by hexdump0815