velvet-os/imagebuilder #46: Move to Chrome os kernel 5.4

EMLommers, 2022-05-02:
I noticed on my Chromebook devices with original kernels 4.4 and 4.14 they all moved to kernel chromeos-5.4 branch.
This is stable and working with audio SST and SOF.
Above 5.4 not official, and 5.4 is even working with Debian 11 and Ubuntu Jammy.
Can you look I into this?

hexdump0815, 2022-05-02:
@EMLommers - thanks a lot for this information - which chromeos boards are affected by this move according to your experience? do you have any links for the patches you mentioned? btw. i also noticed, that there is a new mrchromebox firmware out with geminilake support :) ... did not have time to play around with it yet

EMLommers, 2022-05-02:
At least EVE , pixelbook and I guess upwards, Kaby Lake etc.. amd64 boards
Yes, correct Mr. Chromebox has new firmware for APL , GML....
I do not like to flash the firmware and prefer the depthcharge mods
Will provide patch after diffed..

hexdump0815, 2022-05-02:
is there an easy way to check which kernel version is used by a certain chromeos release on a certain system without updating and booting it?

btw. do you maybe have an idea why wifi is not working for me with depthcharge kernel on gemini lake? the driver probes and loads the proper firmware, but then nothing happens and i'm not ending up with a network device ... i was able to do some experiments with an apollo lake rabbid device where wifi works fine if i booted with the mrchromebox legacy firmware and the debian kernel (and i think i even tried a mainline one), but the same debian rootfs resulted in a non working wifi as described above with a mainline kernel and depthcharge booting ... maybe also some bios strings missing or different in the depthcharge case?

EMLommers, 2022-05-02:
The solution would be to automate download the urls of https://dl.google.com/dl/edgedl/chromeos/recovery/recovery.conf?source=linux_recovery.sh , unzip, mount ROOT-A, and check the kernel version.
It could definitely be the case strings are different. Are you using the modprobe.d files and firmware files from chromeos? I use the udev/rules.d , modprobe.d and firmware. Also depending on the hardware ramoops, intel_idle, i915 kernel settings are applied in the kernel.cmdline

EMLommers, 2022-05-03:
Problems can arise from kexec in to an nonchromeos kernel. do you havi wfii depthharging and use same kernel for rootfs?

EMLommers, 2022-05-03:
ChromeOS kernel 5.4 works with KBL / APL. no need for additional kernel. Wifi work on Coral, and EVE. Sound card is correctly detected. now checking audio ucm

Geofferey, 2022-05-03:
@EMLommers Now this is interesting 🤔 I tried building `5.4` inside Debian but the ChromeOS teams seems to have done some custom because I couldn't get an `Image`... That and when I manually attempted to compile `Image`it failed. If you have any tips that would be awesome. I wonder if it's even possible to boot Debian with `5.4` from ChromeOS team.

EMLommers, 2022-05-03:
There is always possibility to use a Makefile from Ubuntu and have to use gcc-8 and binutils < 2.4 or something. Building on ubuntu 22 fails, you have to downgrade.

hexdump0815, 2022-05-03:
@EMLommers - i have added v5.4 support to https://github.com/hexdump0815/linux-chromeos-kernel to my todo list, but it might take some time until i get to it ... maybe in the future we might add an optional v5.4 chromeos kernel to the systems/images where it might be useful (i.e. better feature coverage) like octopus or even trogdor where the mainline gpu driver is used (it does not make sense in my opinion for systems like kukui which use a propriatary mali gpu driver in chromeos).

btw. i did not use any kexec in my test: just mrchromebox legacy firmware and grub on one side and depthcharge and kpart kernel on the other

EMLommers, 2022-05-03:
Just noticed there is also a ChromeOS Release 101 with 5.15 kernel..... They are working hard to catch up!

closed 2022-05-04 by EMLommers