velvet-os/imagebuilder #47: trogdor: lazor: Notes

leezu, 2022-05-03:
# Notes
Here is list of working and non working features on the Acer Chromebook Spin 513 (lazor-r3) with the 5.18.0-rc5 kernel.

# Working
- WiFi
- Internal EMMC installation (https://github.com/hexdump0815/imagebuilder/issues/43)
- USB C
- Touch

# Partial
- Orientation sensor (overly sensitive - changes orientation all the time and breaks external monitor due to "unsupported orientation" errors by the DPU - Can be disabled via `systemctl disable --now iio-sensor-proxy`)
- USB-C DisplayPort (Works well as long as orientation sensor is disabled)

# Broken
- Built-in audio / speakers (Worked with the image from https://github.com/hexdump0815/imagebuilder/releases/tag/211120-02; Broken after upgrade to Debian SID or Kernel upgrade)
- Front camera
- Suspend (Can't start laptop after suspend)
- Built-in microphone

# Untested
- H/W video decode
- Pen input
- KVM
- Bluetooth
- 3D Accelerated graphics (https://github.com/hexdump0815/imagebuilder/issues/41) (?)

Geofferey, 2022-05-03:
@leezu Thanks for this `systemctl disable --now iio-sensor-proxy`. I had also read some documentation on flipping the orientation of the sensor in user land so no changes need be made to kernel. I'll have to dig it up again because I had more pressing issues but it's a thought. Speaker audio is broken for you too? I'll have to see that kernel now... Still haven't figured if it's user side or kernel.

hexdump0815, 2022-05-03:
i really like the idea to have such an issue per system to discuss its current state and problems - i have just renamed the title a bit so that we have an easy to find subject directly related to the system name in the framework - i hope this is ok?

Geofferey, 2022-05-03:
@hexdump0815 we can make lightning ⚡ progress if we keep this up. I still plan to update the doc and scripts. Collecting all this info and coming up with potential fixes. 😊 I still haven't attempted a full build again to spit out images. Will get there.

hexdump0815, 2022-05-03:
@leezu - i have added a link to your notes to the trogdor page: https://github.com/hexdump0815/imagebuilder/commit/31462ce63bb0dca63d59e94817953360b58d926e

hexdump0815, 2022-05-04:
@Geofferey - there is no hurry with anything here, just take the time you need for everything you want to do - i think its better to go slow and continue moving than to go fast an running out of breath too quickly :) ... first step is to identify potential problems, second is finding solutions for them if possible, third to think about including them into the corresponding images and as a long term topic there is trying to improve the quality of the documentation ...

Geofferey, 2022-05-04:
@hexdump0815 I told you I got A LOT of time on my hands! 🤣 But this is for a limited time only. I'm almost done here though and then I can start thinking about the doc 😜 My goal is always to document and fix issues first. I think I've nearly checked them all now except cameras.

hexdump0815, 2022-05-04:
@Geofferey - cameras should either be usb video class devices and already be probed as such in dmesg or they are just some plain cameras for which is done a lot of dsp stuff in closed source blobs most probably and in that case it would be very hard to impossible get them working (same applies to kukui krane = lenovo duet 10.1 - kukui kappa = hp cb 11a and kukui juniper = acer cb311-3h both have usb video class devices and those are working well)

leezu, 2022-06-17:
@hexdump0815 @Geofferey  I gathered further information about why suspend isn't working on this system.

With 5.15.2-stb-cbq from https://github.com/hexdump0815/imagebuilder/releases/tag/211120-02 (boot from usb), issuing `systemctl suspend` brings the system into `PM: suspend entry (deep)`, but the system wakes up again shortly (~1 second) thereafter. Looking at the embedded controller (EC) logs, it's clear that the EC detects an issue and wakes the system back up:

```
[33479.596920 Handle sleep: 1]
[33489.598512 Warning: Detected sleep hang! Waking host up!]
[33489.599167 event set 0x0000000000080000]
[33490.599948 MKBP not cleared within threshold, toggling.]
[33490.671534 Handle sleep: 2]
[33490.678368 Lid Accel ODR: 10000 - roundup 1 from config 1 [AP 0]]
[33490.682236 PD:S3->S0]
[33490.695668 Base Accel ODR: 10000 - roundup 1 from config 1 [AP 0]]
[33490.711119 Gyro ODR: 0 - roundup 0 from config 0 [AP 0]]
```

With kernel 5.18 (boot from encrypted emmc), the EC also detects an issue and attempts to wake the system back up, but the system fails to come back up. Pressing power-button will not help (besides long-press, to completely reset the system):

```
[31045.942099 Unhandled VB reg 11]
[31351.502099 Handle sleep: 1]
[31351.532439 power state 3 = S0, in 0x001e]
RTC: 0x627d214a (1652367690.00 s)
[31351.533411 power state 7 = S0->S3, in 0x001e]
[31351.535166 Lid Accel ODR: 10000 - roundup 1 from config 2 [AP 0]]
[31351.539477 PD:S0->S3]
RTC: 0x627d214a (1652367690.00 s)
[31351.541955 power state 2 = S3, in 0x001e]
[31351.542533 Base Accel ODR: 10000 - roundup 1 from config 2 [AP 0]]
[31351.554454 Gyro ODR: 0 - roundup 0 from config 0 [AP 0]]
[31368.259650 power state 2 = S3, in 0x000e]
RTC: 0x627d215b (1652367707.00 s)
[31368.260616 power state 6 = S3->S0, in 0x000e]
RTC: 0x627d215b (1652367707.00 s)
[31368.266388 power state 3 = S0, in 0x000e]
[31369.259629 MKBP not cleared within threshold, toggling.]
[31370.261480 MKBP not cleared within threshold, toggling.]
[31378.264244 Warning: Detected sleep hang! Waking host up!]
```

Do you have any ideas about what is going wrong?

hexdump0815, 2022-06-17:
@leezu - in theory it should be possible to get it working - at least if running from emmc - see: https://oftc.irclog.whitequark.org/aarch64-laptops/2022-03-25#30771427
```
[12:40](https://oftc.irclog.whitequark.org/aarch64-laptops/2022-03-25#30771427) <macc24> :O !!! i think suspending to what ec calls "S3" finally works on lazor
[16:54](https://oftc.irclog.whitequark.org/aarch64-laptops/2022-03-25#30772012) <steev> nice!
[18:31](https://oftc.irclog.whitequark.org/aarch64-laptops/2022-03-25#30772789) <hexdump0815> macc24: oh - that sounds good - anything special required or just by using a newer kernel?
[18:32](https://oftc.irclog.whitequark.org/aarch64-laptops/2022-03-25#30772791) <macc24> hexdump0815: it just worked on 5.17
```
lets see if we can find out more details on irc ...

robots, 2023-02-07:
Hi, i have just installed trogdor lazor with 6.1.1-stb-cbq kernel. It seems that sleep is working, most features i have tested are working. Sound, wifi, usb-c displayport, emmc install. Very usable system state.

Do you have any pointers on how to make usb-c displayport audio work ?  I tried most alsa sinks, none of them output sound to hdmi/dp

hexdump0815, 2023-02-07:
@robots - i'm happy to hear this and that everything works so well ... i have no real idea regarding what is needed to get displayport audio to work as i do not even have a hdmi display with audio myself. it might be worth a try to ask at the aarch64-laptops irc channel maybe (https://oftc.irclog.whitequark.org/aarch64-laptops/) - maybe someone around there has an idea

best wishes - hexdump

robots, 2023-02-07:
I have been playing with the ucm2 file, uncommenting HDMI section will break pulseaudio, but with plugged in usb-c to hdmi and valid display connection pulseaudio will load correctly, audio plays correctly to TV! Problem is that pulseaudio will not load, when cable is not plugged in, it will just not see "hw:sc7180rt5682max,2" device. I wonder if its possible to tell pulseaudio to ignore this device for a while and rescan it when ready. However unplugging hdmi cable will not break running pulseaudio.
Might need more googling :-) This ucm2+alsa+pulseaudio concept is very new for me.

hexdump0815, 2023-02-07:
i think ucm is not easy to really understand and to deal with in my experience - it takes quite a bit of time to get forward there step by step ... please keep me up to date here in case you find anything out about this ... good luck :)

ps: it will most probably not help much here, but i just want to make you aware of this - https://github.com/hexdump0815/imagebuilder/blob/main/doc/alsa-ucm-pulseaudio-hacking.txt

robots, 2023-02-08:
Soo, very close to solution:

Create file /usr/share/alsa/ucm2/SC7180/HDMI.conf
`
SectionVerb {
        EnableSequence [
        ]

        DisableSequence [
        ]

        Value {
                TQ "HiFi"
        }
}

SectionDevice."HDMI" {
        Comment "HDMI Output"

        EnableSequence [
        ]

        DisableSequence [
        ]

        Value {
                PlaybackPriority 200
                PlaybackPCM "hw:sc7180rt5682max,2"
                JackControl "HDMI Jack"
        }
}
`
Edit sc7180-rt5682-max98357a-1mic.conf in the same folder add SectionUseCase.HDMI:
`
# bla bla original content
# add:
SectionUseCase."HDMI" {
        File "HDMI.conf"
        Comment "HDMI output"
}
`

Now, HDMI output works, if you restart pulseaudio with hdmi cable plugged in. In the mixer you can select output to default, or output to "HDMI output". Even unplugging works. Audio will switch to internal speakers. Plugging the hdmi cable back in will also work, and switch audio back.

What does not work: If pulseaudio is started and hdmi cable is not connected, pulseaudio will try to open the hw:sc7180xxxxx,2 device, and it does not succeed, and will ignore the card forever (untill pulseaudio restart) but at least the internal sound keeps working.

I have also noticed that official alsa-ucm-conf repo contains trogdor/lazor config files that should work with headphones:

https://github.com/alsa-project/alsa-ucm-conf/tree/master/ucm2/Qualcomm/sc7180/rt5682-max98357a

I have no yet tested those. HDMI audio is now priority for me.

JackControl tells pulseaudio to monitor kcontrol signals. JackDev should monitor events, but those are not available in user space without root. So JackControl is better way. To know what should be written there just check:
`
amixer -c 0 contents
`
"HDMI Jack" goes to on when connected, off when disconnected. Same for "Headset Jack" on when connected, off disconnected.

sowa705, 2023-02-09:
Hi, what is the status on power management on this laptop?

robots, 2023-02-09:
sleeping works, resume works also. Not sure how well. I have not noticed anything wrong yet

sowa705, 2023-02-09:
I had an issue with the CPU governor not dropping down frequencies on idle but i managed to get it to work. Still power draw seems to be much higher than chromeOS

hexdump0815, 2023-02-10:
@sowa705 - i think chromeos does some extra tricks to get the power consumption down, but i think even with mainline linux it is very useable (i.e. many hours battery life in use and it should surive a night or a few days even in sleep mode i think)

i so far did not notice that cpu governor issue - what did you do to fix it? maybe switching to schedutil might be better anyway: https://github.com/hexdump0815/imagebuilder/blob/main/files/extra-files/etc/rc.local#L92-L100 - are you running my latest v6.1.1 kernel already?

sowa705, 2023-02-10:
Hi, i switched the governor to ondemand (it was set to performance keeping clocks at max) I'm using the prebuilt image with ubuntu.
imo good battery life is the biggest advantage of this laptop and going from 10 hours to 5 makes it pointless.

Ill be looking into minimizing power draw (hw video accel will probably help a lot)

sowa705, 2023-02-10:
Prebuilt images are quite old tho. Ill try building newer ones

hexdump0815, 2023-02-10:
@sowa705 - now that v6.1 is set as lts kernel i'll most probably start building a new round of images based on v6.1, but it will take a bit of time until i'm done with all of the ones i would like to build ... in case you try to build your own, i just recently tried to write down at least some basic info about how to do it here: https://github.com/hexdump0815/imagebuilder/blob/main/doc/using-the-imagebuilder-framework.md

sowa705, 2023-02-13:
Hi, sorry for derailing the thread but are there any additional steps to make the image bootable? I've built the latest kernel with the guide and jammy image but i cant get it to boot :( Only thing im getting is just that my image is invalid

hexdump0815, 2023-02-13:
@sowa705 - the image resulting from the build should be bootable, but i think i forgot one step in the build info: the bootable kpart kernel image has to be extracted from the kernel.tar.gz for chromebooks using the native kpart chromebook bootloader i.e. this here: https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_trogdor/get-files.sh#L9 ... just doing that step in the imagebuilder-download folder and running the create-image.sh step again should result in a bootable image ... i*ll have to add this to the doc ... if that still does not work, lets open a new issue to try to debug what the problem then is ... btw. aside from that: depending on the usb device used pressing ctrl-u immediately after power-on might be too fast sometimes, it looks like usb needs a moment to initialize first sometimes

best wishes and good luck - hexdump

hexdump0815, 2023-02-13:
@sowa705 - short version: just dd the extracted vmlinux.kpart to the first partition of your sd card after writing your image to it in case you do not want to rebuild it completely ...

hexdump0815, 2023-02-13:
btw. i built a fresh kernel yesterday as well: https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases/tag/6.1.11-stb-cbq ... new image should come within the next days

sowa705, 2023-02-13:
Huge thanks, ill try running my own kernel (6.2 so it may not work) and if it fails ill use your prebuilt one

hexdump0815, 2023-02-18:
@sowa705 - just fyi: new image with v6.1.11 kernel - https://github.com/hexdump0815/imagebuilder/releases/tag/230218-03 ... i did not have the time yet to test it myself, but it should work

thomas725, 2023-03-23:
I've got an Acer Chromebook Spin 513 Lazor / Trogdor device and I'd like to try to install one of your images on it.
So if I understood the process correctly, for that I'll have to:

1. Download for example https://github.com/hexdump0815/imagebuilder/releases/download/230218-03/chromebook_trogdor-aarch64-jammy.img.gz
2. `gunzip chromebook_trogdor-aarch64-jammy.img.gz`
3. `dd if=chromebook_trogdor-aarch64-jammy.img of=/dev/sda status=progress` (where sda is a 4GB usb-stick)
4. boot from that usb stick (does that work without an alternative bootloader? if not, where can I find one compatible with my device?)
5. once booted, write the same image to internal emmc storage: `dd if=~/chromebook-linux.img of=/dev/mmcblk1`
6. reboot from internal storage and see if everything works as hoped?

Is that about it or did I miss anything?

hexdump0815, 2023-03-23:
@thomas725 - it should work as you wrote .. two notes maybe: the jammy image has not yet been tested, but it should work (just if it does not, then maybe give the bookworm one a try) - and the other note: maybe test it a bit first running from usb before installing it to emmc - for that i have tried to put together some notes at https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md ... for getting started please also read: https://github.com/hexdump0815/imagebuilder/blob/main/doc/first-steps.md and https://github.com/hexdump0815/imagebuilder/blob/main/doc/important-information.md (which should not apply to the jammy image, but to the bookworm image) ... btw. there is no special bootloader required as the chromebook bootloader is used which lives in some special flash memorym so does even work when you wipe the emmc ...

good luck and best wishes - hexdump

thomas725, 2023-03-24:
Thank you for your detailed response @hexdump0815 !
I've tried the jammy image, and will try the bookworm next.

Things I didn't like about jammy:
* found something about an "Ubuntu premium subscription" in the settings (last time I used Ubuntu was quite a few years ago..)
* screen resolution could not be reduced, entering custom scale 0.8 hid part of the screen content and made the rest look out of focus
* scrolling with touch in Firefox by trying to drag the empty space instead seemed to select text.

I wonder if maybe a wayland based desktop manager would have better touchscreen support?

hexdump0815, 2023-03-24:
@thomas725 - the way ubuntu goes is also the reason why i have switched completely to debian meanwhile ... for display scaling maybe have a look at the files in /etc/lightdm, there is a little hack using scaling via xrandr ... all the little fine details have to be configured individually as it is too dependend on all the different systems and would be too much work to handle it properly across all image - the idea of the images is to give a good starting point to adjust the system as one likes it then - before it was hours of work of finding out of how to boot it, which kernel patches are required, which kernel config works etc. and now one can start with a working system at least within a few minutes :) ... in case you find out any improvements, feedback here is very welcome as of course i'm open for improving the images over time as long as the changes are doable and do not break the images for other systems

best wishes and good luck - hexdump

linsyking, 2024-02-08:
Tested on my machine:

Hardware decoding works well but no hardware encoding support.

> power

To save power, we can install [tlp](https://linrunner.de/tlp/usage/tlp.html). It seems to extend the working hours. (from 7 - 8 hours to 9 hours+ in my case)

linsyking, 2024-04-06:
When updating to linux 6.8.2, kernel fails to detect battery. 6.7.4 works great.

hexdump0815, 2024-04-07:
@linsyking - it looks like this seems to be some regression or config change - others saw it as well: https://oftc.irclog.whitequark.org/aarch64-laptops/2024-04-07#33086247

linsyking, 2024-07-08:
Hi hexdump, what does "Graphics Hardware Acceleration with https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/17888" mean? How to enable that?

hexdump0815, 2024-07-08:
@linsyking - can you please give a bit more context? i'm not fully understanding what you mean ...

LukIsHere, 2024-07-09:
can confirm that stylus is working on this device as well
more details: https://github.com/hexdump0815/imagebuilder/issues/68#issuecomment-2174602291

glauzon67, 2024-07-15:
@hexdump0815 can you provide a document or something on how to compile and install a kernel on the chromebook lazor. i have tryied the normal way with a kernel source code downloaded on kernel.org and compiled it on the lazor but after im stuck... i know its not a x86 platform but a arm64 so sudo make install not gonna make it work... Thanks

LukIsHere, 2024-07-16:
@glauzon67 you see
general building info is [here](https://github.com/hexdump0815/imagebuilder/blob/main/doc/building-own-kernels.md)
specyfic for this device steps r [here](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/master/config.cbq-6.6.23-stb-cbq+)
info on installation is [here](https://github.com/hexdump0815/imagebuilder/blob/main/doc/installing-a-newer-kernel.md)
it is recommended to compile it on arm64 platform

if you want it to just be ```make install```
you would have to setup depthcharge-tools i managed to [do it myself](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/issues/15#issuecomment-2201276609) but it's not trivial atm (and currently requires kelner partitions to be 64mb)

there is also an [issue with kelner compiled on debian trixie not working](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/issues/15#issuecomment-2211365092) so be aware of that

glauzon67, 2024-07-17:
i managed to compile a kernel and boot it (Linux changeme 6.10.0-stb-cbq #1 SMP PREEMPT Wed Jul 17 13:51:03 EDT 2024 aarch64 GNU/Linux) everything work fine exept all the usb ports get disabled about 30 seconds after boot.
(i know the usb ports was working because my mouse was working before the 30 seconds timeout)
dmesg spit: 

[   35.808200] pp3300_hub: disabling
[   35.812155] usb 1-1: USB disconnect, device number 2
[   35.817652] usb 1-1.1: USB disconnect, device number 3
[   35.847159] usb 1-1.3: USB disconnect, device number 4

Any idea on whats wrong 

Device: Lazor

Thanks

LukIsHere, 2024-07-17:
@glauzon67 
i don't think 6.10 is the most stable kelner to run
but i once had similar issues on corsola chromebook (different cpu, still arm64) when touchpad wasn't working

iirc it was either incorrect/missing .dts or modules not working/loading/existing

still the .config file you generated wasn't really meant for 6.10 (afaik) so it might be configuration issue

i do own this device myself and can try it out

still have u by any chance not pick 6.10 kelner version on purpose? (if yes, you can just ```git reset --hard v6.6.23``` and there should be no problem). be aware that running stuff as critical as kelner from the latest commit aint the best idea :3

hexdump0815, 2024-07-17:
@glauzon67 - it might be a good idea to first start from a known working version like @LukIsHere and if that works maybe move on ... it could be that something is broken in v6.10 (happens from time to time) or it could be that something is not rightz with your kernel build

glauzon67, 2024-07-17:
@hexdump0815 @LukIsHere i used make olddefconfig from 6.1.51-stb-cbq... maybe is my problem...
here is my .config 

ps  sorry for the looong post im new github

#
# Automatically generated file; DO NOT EDIT.
# Linux/arm64 6.10.0 Kernel Configuration
#
CONFIG_CC_VERSION_TEXT="gcc (Debian 12.2.0-14) 12.2.0"
CONFIG_CC_IS_GCC=y
CONFIG_GCC_VERSION=120200
CONFIG_CLANG_VERSION=0
CONFIG_AS_IS_GNU=y
CONFIG_AS_VERSION=24000
CONFIG_LD_IS_BFD=y
CONFIG_LD_VERSION=24000
CONFIG_LLD_VERSION=0
CONFIG_CC_CAN_LINK=y
CONFIG_CC_CAN_LINK_STATIC=y
CONFIG_CC_HAS_ASM_GOTO_OUTPUT=y
CONFIG_CC_HAS_ASM_GOTO_TIED_OUTPUT=y
CONFIG_GCC_ASM_GOTO_OUTPUT_WORKAROUND=y
CONFIG_CC_HAS_ASM_INLINE=y
CONFIG_CC_HAS_NO_PROFILE_FN_ATTR=y
CONFIG_PAHOLE_VERSION=0
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_TABLE_SORT=y
CONFIG_THREAD_INFO_IN_TASK=y

#
# General setup
#
CONFIG_INIT_ENV_ARG_LIMIT=32
# CONFIG_COMPILE_TEST is not set
# CONFIG_WERROR is not set
CONFIG_LOCALVERSION="-stb-cbq"
# CONFIG_LOCALVERSION_AUTO is not set
CONFIG_BUILD_SALT=""
CONFIG_DEFAULT_INIT=""
CONFIG_DEFAULT_HOSTNAME="(none)"
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
CONFIG_SYSVIPC_COMPAT=y
CONFIG_POSIX_MQUEUE=y
CONFIG_POSIX_MQUEUE_SYSCTL=y
# CONFIG_WATCH_QUEUE is not set
CONFIG_CROSS_MEMORY_ATTACH=y
# CONFIG_USELIB is not set
CONFIG_AUDIT=y
CONFIG_HAVE_ARCH_AUDITSYSCALL=y
CONFIG_AUDITSYSCALL=y

#
# IRQ subsystem
#
CONFIG_GENERIC_IRQ_PROBE=y
CONFIG_GENERIC_IRQ_SHOW=y
CONFIG_GENERIC_IRQ_SHOW_LEVEL=y
CONFIG_GENERIC_IRQ_EFFECTIVE_AFF_MASK=y
CONFIG_GENERIC_IRQ_MIGRATION=y
CONFIG_HARDIRQS_SW_RESEND=y
CONFIG_IRQ_DOMAIN=y
CONFIG_IRQ_DOMAIN_HIERARCHY=y
CONFIG_IRQ_FASTEOI_HIERARCHY_HANDLERS=y
CONFIG_GENERIC_IRQ_IPI=y
CONFIG_GENERIC_MSI_IRQ=y
CONFIG_IRQ_MSI_IOMMU=y
CONFIG_IRQ_FORCED_THREADING=y
CONFIG_SPARSE_IRQ=y
[... 7043 lines cut ...]
# CONFIG_MAX11410 is not set
# CONFIG_MAX1241 is not set
# CONFIG_MAX1363 is not set
# CONFIG_MAX34408 is not set
CONFIG_MAX9611=m
# CONFIG_MCP320X is not set
# CONFIG_MCP3422 is not set
# CONFIG_MCP3564 is not set
# CONFIG_MCP3911 is not set
# CONFIG_MEDIATEK_MT6360_ADC is not set
# CONFIG_NAU7802 is not set
# CONFIG_PAC1934 is not set
CONFIG_QCOM_VADC_COMMON=m
# CONFIG_QCOM_SPMI_RRADC is not set
# CONFIG_QCOM_SPMI_IADC is not set
# CONFIG_QCOM_SPMI_VADC is not set
CONFIG_QCOM_SPMI_ADC5=m
# CONFIG_RICHTEK_RTQ6056 is not set
# CONFIG_SD_ADC_MODULATOR is not set
# CONFIG_TI_ADC081C is not set
# CONFIG_TI_ADC0832 is not set
# CONFIG_TI_ADC084S021 is not set
# CONFIG_TI_ADC12138 is not set
# CONFIG_TI_ADC108S102 is not set
# CONFIG_TI_ADC128S052

LukIsHere, 2024-07-17:
please if you want to inclute long text wrap it in ```
and the whole thing put in
```
<details>
<summary>title<\summary>
``
content
``
<\details>
```

like this
<details>
<summary>title<\summary>
```
content
```
<\details>

glauzon67, 2024-07-17:
sorry about that @LukIsHere 

.config 
" #
# Automatically generated file; DO NOT EDIT.
# Linux/arm64 6.10.0 Kernel Configuration
#
CONFIG_CC_VERSION_TEXT="gcc (Debian 12.2.0-14) 12.2.0"
CONFIG_CC_IS_GCC=y
CONFIG_GCC_VERSION=120200
CONFIG_CLANG_VERSION=0
CONFIG_AS_IS_GNU=y
CONFIG_AS_VERSION=24000
CONFIG_LD_IS_BFD=y
CONFIG_LD_VERSION=24000
CONFIG_LLD_VERSION=0
CONFIG_CC_CAN_LINK=y
CONFIG_CC_CAN_LINK_STATIC=y
CONFIG_CC_HAS_ASM_GOTO_OUTPUT=y
CONFIG_CC_HAS_ASM_GOTO_TIED_OUTPUT=y
CONFIG_GCC_ASM_GOTO_OUTPUT_WORKAROUND=y
CONFIG_CC_HAS_ASM_INLINE=y
CONFIG_CC_HAS_NO_PROFILE_FN_ATTR=y
CONFIG_PAHOLE_VERSION=0
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_TABLE_SORT=y
CONFIG_THREAD_INFO_IN_TASK=y

#
# General setup
#
CONFIG_INIT_ENV_ARG_LIMIT=32
# CONFIG_COMPILE_TEST is not set
# CONFIG_WERROR is not set
CONFIG_LOCALVERSION="-stb-cbq"
# CONFIG_LOCALVERSION_AUTO is not set
CONFIG_BUILD_SALT=""
CONFIG_DEFAULT_INIT=""
CONFIG_DEFAULT_HOSTNAME="(none)"
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
CONFIG_SYSVIPC_COMPAT=y
CONFIG_POSIX_MQUEUE=y
CONFIG_POSIX_MQUEUE_SYSCTL=y
# CONFIG_WATCH_QUEUE is not set
CONFIG_CROSS_MEMORY_ATTACH=y
# CONFIG_USELIB is not set
CONFIG_AUDIT=y
CONFIG_HAVE_ARCH_AUDITSYSCALL=y
CONFIG_AUDITSYSCALL=y

#
# IRQ subsystem
#
CONFIG_GENERIC_IRQ_PROBE=y
CONFIG_GENERIC_IRQ_SHOW=y
CONFIG_GENERIC_IRQ_SHOW_LEVEL=y
CONFIG_GENERIC_IRQ_EFFECTIVE_AFF_MASK=y
CONFIG_GENERIC_IRQ_MIGRATION=y
CONFIG_HARDIRQS_SW_RESEND=y
CONFIG_IRQ_DOMAIN=y
CONFIG_IRQ_DOMAIN_HIERARCHY=y
CONFIG_IRQ_FASTEOI_HIERARCHY_HANDLERS=y
CONFIG_GENERIC_IRQ_IPI=y
CONFIG_GENERIC_MSI_IRQ=y
CONFIG_IRQ_MSI_IOMMU=y
CONFIG_IRQ_FORCED_THREADING=y
CONFIG_SPARSE_IRQ=y
# CONFIG_GENERIC_IRQ_DEBUGFS is not set
# end of IRQ subsystem
[... 7045 lines cut ...]
CONFIG_MAX9611=m
# CONFIG_MCP320X is not set
# CONFIG_MCP3422 is not set
# CONFIG_MCP3564 is not set
# CONFIG_MCP3911 is not set
# CONFIG_MEDIATEK_MT6360_ADC is not set
# CONFIG_NAU7802 is not set
# CONFIG_PAC1934 is not set
CONFIG_QCOM_VADC_COMMON=m
# CONFIG_QCOM_SPMI_RRADC is not set
# CONFIG_QCOM_SPMI_IADC is not set
# CONFIG_QCOM_SPMI_VADC is not set
CONFIG_QCOM_SPMI_ADC5=m
# CONFIG_RICHTEK_RTQ6056 is not set
# CONFIG_SD_ADC_MODULATOR is not set
# CONFIG_TI_ADC081C is not set
# CONFIG_TI_ADC0832 is not set
# CONFIG_TI_ADC084S021 is not set
# CONFIG_TI_ADC12138 is not set
# CONFIG_TI_ADC108S102 is not set
# CONFIG_TI_ADC128S052 is not set
# CONFIG_TI_ADC161S626 is not set
CONFIG_TI_ADS1015=m
# CONFIG_TI_ADS7924 is not set
# CONFIG_TI_ADS1100 i

glauzon67, 2024-07-17:
i will join the file haha sorry

[config.txt](https://github.com/user-attachments/files/16273889/config.txt)

LukIsHere, 2024-07-17:
i don't really get that stuff well yet so can't really look at that and diagnose it

but if u r using any config you probably should use the latest (6.6.23-stb-cbq+)

glauzon67, 2024-07-17:
i will give a try

glauzon67, 2024-07-18:
So, i have tryied kernel 6.10.0 with "make olddefconfig" from config-6.6.23-stb-cbq yesterday and ... same thing, all usb ports get disabled after 30 seconds... Today i have compiled kernel 6.9.9 with config-6.6.23-stb-cbq  and so far everything runs great... usb ports, wifi, bt etc... i wil try it for some days and give some updates.

Cheers

hexdump0815, 2024-07-20:
so it looks like either there is a regression in v6.10 or maybe some new kernel config options got introduced which will be required now - lets maybe wait a bit for v6.10 to stabilize and if the problem is then still there maybe bisecting v6.10 against v6.9 might be an option to find out which commit breaks it

glauzon67, 2024-07-25:
After some days of testing of the kernel 6.9.9-stb-cbq... Everything runs fine... no crash but some graphical glitches in gtk apps (ex: Firefox-esr).  ive been using kde-plasma-desktop on wayland and its smooth... Battery life is about 12 hours (from a battery health reporting at 84%) 

pretty usable at the end

cheers

augi33, 2024-08-14:
thanks for all the comments fellas! I successfully installed debian on my old girlfriends spin513. So sick! A tip for someone else if they looking is you need to go into the terminal on chromeOS and enable usb boot. the command is crossystem dev_boot_usb=1. this will enable you to boot from ur usb when your in dev mode. Thanks for the hard work and Great Project!

Bartq, 2024-12-22:
Tried both images chromebook_trogdor-aarch64-jammy.img.gz and chromebook_trogdor-aarch64-bookworm.img.gz but after boot Login and Pass is needed.

linsyking, 2025-02-02:
When I try use kernel 6.13, all USB ports are not working

thenameisluk, 2025-02-02:
@linsyking we do not provide 6.13 kernel and i'd prefer not to touch it before we are done with new batch of images

the problem you are describing sounds like missing kernel modules
but i'm not sure

if you want to run a newer working kernel, i'd advice you to opt for 6.12 https://github.com/hexdump0815/imagebuilder/discussions/267#discussioncomment-11289341 which is already prebuilt and tested to work

linsyking, 2025-02-02:
> [@linsyking](https://github.com/linsyking) we do not provide 6.13 kernel and i'd prefer not to touch it before we are done with new batch of images
> 
> the problem you are describing sounds like missing kernel modules but i'm not sure
> 
> if you want to run a newer working kernel, i'd advice you to opt for 6.12 [#267 (comment)](https://github.com/hexdump0815/imagebuilder/discussions/267#discussioncomment-11289341) which is already prebuilt and tested to work

All 6.10 to 6.12 are not working. Only 6.9 works. Did you observe similar issues on other machines?

thenameisluk, 2025-02-02:
@linsyking no i didn't on other trogdor machines running prebuilt 6.12 kernel
homestar - https://github.com/hexdump0815/imagebuilder/discussions/267#discussioncomment-11295579
wormdingler - https://github.com/hexdump0815/imagebuilder/discussions/267#discussioncomment-11418777
coachz - https://github.com/hexdump0815/imagebuilder/discussions/267#discussioncomment-11699965

if you are saying there is an actual regression with [prebuild 6.12 kernel](https://github.com/hexdump0815/imagebuilder/discussions/267#discussioncomment-11289341) i can get my lazor and check

thenameisluk, 2025-02-02:
there used to be an issue https://github.com/hexdump0815/imagebuilder/discussions/243#discussioncomment-10653865
but it's no longer the case

linsyking, 2025-02-02:
> [@linsyking](https://github.com/linsyking) no i didn't on other trogdor machines running prebuilt 6.12 kernel homestar - [#267 (comment)](https://github.com/hexdump0815/imagebuilder/discussions/267#discussioncomment-11295579) wormdingler - [#267 (comment)](https://github.com/hexdump0815/imagebuilder/discussions/267#discussioncomment-11418777) coachz - [#267 (comment)](https://github.com/hexdump0815/imagebuilder/discussions/267#discussioncomment-11699965)
> 
> if you are saying there is an actual regression with [prebuild 6.12 kernel](https://github.com/hexdump0815/imagebuilder/discussions/267#discussioncomment-11289341) i can get my lazor and check

I figured it out.

Need to change

`CONFIG_USB_ONBOARD_HUB=y`

to

`CONFIG_USB_ONBOARD_DEV=y`

thenameisluk, 2025-02-02:
@hexdump0815 probably something worth adding

[9 more replies]