velvet-os/imagebuilder #48: Odroid U3 Ubuntu/Debian - Missing VLAN 8021q module

AKA-47, 2022-05-05:
Hi.
Im using the image on odroid u3 using kernel 5.10.45-stb-exy+

Installing the vlan package via apt doesnt work to get vlans running because the 8021q kernel module is missing.

Do you have any advise to get this running?

hexdump0815, 2022-05-05:
i think the best would be to build your own kernel directly following my notes from https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/master/readme.exy - it is not that complicated as long as there is some space on the sd card

best wishes and good luck - hexdump

Silverstar, 2022-05-06:
I was able to use the official kernel provided by the backports repository on my u1. 
I needed that because I was missing kernel modules as well and didn't want to compile a kernel on my own so I tried.
You will have to edit the extlinux.conf in the boot partition to use another kernel.
See in my issue #39
But I don't know what else might not work anymore with that kernel, for my tasks everything works.

hexdump0815, 2022-05-22:
i have added the vlan options to the armv7l options now - https://github.com/hexdump0815/kernel-config-options/commit/528c8dbd13031048ba5cb2ff0d07bf5a92f94269

closed 2022-05-22 by hexdump0815