velvet-os/imagebuilder #49: system status: chromebook_kukui: fennel14 (lenovo ideapad chromebook 14)

BenjaminWegener, 2022-05-17:
## Notes on FENNEL14 running 5.18.0-rc7

Here is list of working and non working features on the Lenovo Ideapad Chromebook 14 (aka FENNEL14 sku0). I will update accordingly with detailed issues, logs and proposed fixes. 

### Working
• WiFi
• Screen
• Energy (Battery Status, DPMS)
• USB C
• Webcam (tested with firefox https://experiments.withgoogle.com/collection/ai/move-mirror/view/mirror)
• H/W video decode
• GPU Opengl (tested with firefox https://webglsamples.org/aquarium/aquarium.html)

### Untested
• H/W video encode
• Suspend
• GPU OpenglES 3.1 (should be working with new panfrost driver integrated)

### Broken
• Bluetooth (was working with 5.13.2, maybe config issue)
• Sound (see below)

### Installation
To build the System, get the Image from https://github.com/hexdump0815/imagebuilder/releases/download/210724-03/chromebook_kukui-aarch64-focal.img.gz and flash it to sdcard/pendrive
After boot, open shell with CTRL ALT F2/RIGHT ARROW, login with linux/changeme
connect to wifi with `nmtui`
do following to install LXDE (xfce/gnome doesn't work to this point): `sudo /scripts/extend-rootfs.sh && sudo apt update && sudo install lxde -y`
reboot, in lightdm select LXDE
download this script:
[build_kernel-FENNEL14-5.18-rc7.sh.txt](https://github.com/hexdump0815/imagebuilder/files/8706440/build_kernel-FENNEL14-5.18-rc7.sh.txt)
rename it to `.sh`, make it executable, run it with `sudo ./build_kernel-FENNEL14-5.18-rc7.sh`
after completion, it will ask to write the updated kernel, your installation has to be on /dev/sda!

### PROBLEMS
sound is not working, there were config issues, but after enabling all sound configs, there are still errors, see log: 
[dmesg.txt](https://github.com/hexdump0815/imagebuilder/files/8706477/dmesg.txt)

hexdump0815, 2022-05-17:
let me just add the old issue here as a direct reference for older topics: https://github.com/hexdump0815/imagebuilder/issues/45

regarding your audio issues your idea with CONFIG_SND_SOC_MT8183_DA7219_MAX98357A=y sounds like a good one - please let me know in case it helps, then i'll add it to the additional config options for kukui

btw. i have added fennel14 now as tested at the kukui page and linked this issue here there for it

hexdump0815, 2022-05-17:
ok - the option CONFIG_SND_SOC_MT8183_DA7219_MAX98357A does not exist in current mainline and the options you need (as =m) you should already have in: CONFIG_SND_SOC_RT1015P and CONFIG_SND_SOC_DA7219 ... the problem is more that the currently included ucm files (https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_kukui/extra-files/usr/share/alsa/ucm2 - see the readme there) do not cover fennel14 yet - so you'll have to try to model ucm files based on https://chromium.googlesource.com/chromiumos/overlays/board-overlays/+/refs/heads/master/overlay-jacuzzi/chromeos-base/chromeos-bsp-jacuzzi/files/fennel14/audio/ucm-config/ - for that compare the original chromeos ucm files from the readme mentioned above to the ucm files i made out of them including the accordingly shortened dir- and filenames (https://github.com/hexdump0815/imagebuilder/blob/main/info/generic/alsa-ucm-pulseaudio-hacking.txt might be helpful for this as well) ... this should hopefully give you working audio at some point ... and while working with audio never wear headphones directly and keep mixer levels (if there are any mixers) rather low to avoid bad surprises in case of unexpected high levels

good luck and best wishes - hexdump

hexdump0815, 2022-05-17:
btw. i plan to build new images for kukui after v5.18 and mesa v22.1 are officially released - this should hopefully make the installation on fennel14 (and maybe other kukui systems as well) easier as the lxde step is then hopefully no longer required ...

hexdump0815, 2022-05-28:
i have just finished creating some new kukui images: https://github.com/hexdump0815/imagebuilder/releases/tag/220528-01 - it would be interesting if they will boot cleaner now on fennel14

mattzilla30, 2022-05-29:
Sound still isn't working.

BenjaminWegener, 2022-05-29:
i am still struggeling with that ucm files, i will give update tomorrow. these images i will test today.

hexdump0815, 2022-05-29:
cool - thanks a lot - let me know if you need help with the ucm files

BenjaminWegener, 2022-05-29:
@hexdump0815 maybe, i will create a writeup, what i've done so far including logs etc.

BenjaminWegener, 2022-05-30:
/// POST UNDER CONSTRUCTION ///
- first i tried to implement a custom ucm file, so i thought i'd copy it from into the existing /usr/share/alsa/ucm2/mt8183_mt6358_t/ :
-

hexdump0815, 2022-05-30:
it looks like that for some reason the kernel does not detect any soundcard yet, i.e. we run into problems before ucm comes into play - can you please paste the output of "cat /proc/asound/cards" here? i assume it will be empty in your case ... in case it is empty, then the output of "dmesg | grep -i snd" might be interesting as well ...

hexdump0815, 2022-05-30:
in case you get to a detected sound card, then this might be a good starting point for a fennel14 ucm file: https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_kukui/extra-files/usr/share/alsa/ucm2/mt8183_da7219_m

BenjaminWegener, 2022-05-30:
thank you very much for your reply!
```
linux@changeme:~$ sudo cat /proc/asound/cards
[sudo] password for linux: 
--- no soundcards ---
```

```
linux@changeme:~$ sudo dmesg | grep -i snd
[    0.981323] usbcore: registered new interface driver snd-usb-audio
[   12.997911] mt8183_da7219 mt8183-sound: snd_soc_bind_card: snd_soc_dapm_add_routes failed: -19
```

```
linux@changeme:~$ sudo modprobe mt
Display all 174 possibilities? (y or n)
mt20xx                              mte
mt20xx.mod                          mteswap
mt6311-regulator                    mtk
mt6323-poweroff                     mtk-afe-fe-dai
mt6323-regulator                    mtk-afe-platform-driver
mt6351                              mtk-btcvsd
mt6358                              mtk-cmdq-helper
mt6358-irq                          mtk-cmdq-mailbox
mt6358-regulator                    mtk-cqdma
mt6360-core                         mtk-devapc
mt6360-core.mod                     mtk-devapc.mod
mt6360-regulator                    mtk-efuse
mt6360-regulator.mod                mtk-eint
mt6380-regulator                    mtk-hsdma
mt6397-core                         mtk-infracfg
mt6397-irq                          mtk-mdp
mt6397-regulator                    mtk-mdp.mod
mt6577_auxadc.mod                   mtk-mmsys
mt6660                              mtk-mutex
mt76                                mtk-pm-domains
mt76-connac-lib                     mtk-pmic-wrap
mt76-connac-lib.mod                 mtk-rng
mt76-usb                            mtk-scpsys
mt76-usb.mod                        mtk-sd
mt76.mod                            mtk-smi
mt7601u                             mtk-vcodec-common
mt7601u.mod                         mtk-vcodec-common.mod
mt7615-common                       mtk-vcodec-dec
mt7615-common.mod                   mtk-vcodec-dec-hw
mt7663-usb-sdio-common              mtk-vcodec-dec-hw.mod
mt7663-usb-sdio-common.mod          mtk-vcodec-dec.mod
mt7663u                             mtk-vcodec-enc
mt7663u.mod                         mtk-vcodec-enc.mod
mt76_connac_mac                     mtk-vpu
mt76_connac_mcu                     mtk-vpu.mod
mt76x0-common                       mtk_cec
mt76x0-common.mod                   mtk_disp_aal
mt76x02-lib                         mtk_disp_ccorr
mt76x02-lib.mod                     mtk_disp_color
mt76x02-usb                         mtk_disp_gamma
mt76x02-usb.mod                     mtk_disp_ovl
mt76x02_beacon                      mtk_disp_rdma
mt76x02_debugfs                     mtk_dpi
mt76x02_dfs                         mtk_drm_crtc
mt76x02_eeprom                      mtk_drm_ddp_comp
mt76x02_mac                         mtk_drm_drv
mt76x02_mcu                         mtk_drm_gem
mt76x02_mmio                        mtk_drm_plane
mt76x02_phy                         mtk_dsi
mt76x02_trace                       mtk_hdmi
mt76x02_txrx                        mtk_hdmi_ddc
mt76x02_usb_core                    mtk_iommu
mt76x02_usb_mcu                     mtk_jpeg.mod
mt76x02_util                        mtk_jpeg_core
mt76x0u                             mtk_jpeg_dec_hw
mt76x0u.mod                         mtk_jpeg_dec_parse
mt76x2-common                       mtk_jpeg_enc_hw
mt76x2-common.mod                   mtk_mdp_comp
mt76x2u                             mtk_mdp_core
mt76x2u.mod                         mtk_mdp_m2m
mt8173-afe-pcm                      mtk_mdp_regs
mt8173-afe-pcm.mod                  mtk_mdp_vpu
mt8173-rt5650                       mtk_rpmsg.mod
mt8173-rt5650-rt5514                mtk_scp.mod
mt8173-rt5650-rt5514.mod            mtk_scp_ipi.mod
mt8173-rt5650-rt5676                mtk_thermal
mt8173-rt5650-rt5676.mod            mtk_vcodec_dec_drv
mt8173-rt5650.mod                   mtk_vcodec_dec_hw
mt8183-afe-clk                      mtk_vcodec_dec_pm
mt8183-afe-pcm                      mtk_vcodec_dec_stateful
mt8183-cci-devfreq                  mtk_vcodec_dec_stateless
mt8183-da7219-max98357              mtk_vcodec_enc_drv
mt8183-dai-adda                     mtk_vcodec_enc_pm
mt8183-dai-hostless                 mtk_vcodec_fw
mt8183-dai-i2s                      mtk_vcodec_fw_scp
mt8183-dai-pcm                      mtk_vcodec_fw_vpu
mt8183-dai-tdm                      mtk_vcodec_intr
mt8183-mt6358-ts3a227-max98357      mtk_vcodec_util
mt8183-mt6358-ts3a227-max98357.mod  mtk_wdt.mod
mtd_blkdevs                         mtu3_core
mtd_dataflash                       mtu3_debugfs
mtdblock                            mtu3_dr
mtdchar                             mtu3_gadget
mtdconcat                           mtu3_gadget_ep0
mtdcore                             mtu3_host
mtdpart                             mtu3_plat
mtdsuper                            mtu3_qmu
```
--- to be continued after reboot ---

hexdump0815, 2022-05-30:
"modprobe mt8183-da7219-max98357 ; cat /proc/asound/cards" would be interesting

BenjaminWegener, 2022-05-30:
```
linux@changeme:~$ sudo modprobe mt8183-da7219-max98357 ; cat /proc/asound/cards  
--- no soundcards ---
```
linux@changeme:~$ sudo cat /proc/modules
mt8183_mt6358_ts3a227_max98357 28672 0 - Live 0xffffb5da04f67000
snd_soc_ts3a227e 16384 1 mt8183_mt6358_ts3a227_max98357, Live 0xffffb5da04f62000
ath10k_sdio 32768 0 - Live 0xffffb5da0519d000
ath10k_core 479232 1 ath10k_sdio, Live 0xffffb5da05112000
mtk_vcodec_dec 73728 0 - Live 0xffffb5da050f8000
ath 40960 1 ath10k_core, Live 0xffffb5da05041000
mtk_vcodec_enc 45056 0 - Live 0xffffb5da05021000
v4l2_h264 16384 1 mtk_vcodec_dec, Live 0xffffb5da04f80000
mtk_vcodec_common 20480 2 mtk_vcodec_dec,mtk_vcodec_enc, Live 0xffffb5da04ffe000
mac80211 516096 1 ath10k_core, Live 0xffffb5da05079000
mtk_vpu 20480 1 mtk_vcodec_common, Live 0xffffb5da04f1e000
mtk_jpeg 40960 0 - Live 0xffffb5da04f90000
uvcvideo 106496 0 - Live 0xffffb5da0505e000
videobuf2_dma_contig 24576 3 mtk_vcodec_dec,mtk_vcodec_enc,mtk_jpeg, Live 0xffffb5da05034000
libarc4 16384 1 mac80211, Live 0xffffb5da0502f000
videobuf2_vmalloc 20480 1 uvcvideo, Live 0xffffb5da04f71000
v4l2_mem2mem 40960 3 mtk_vcodec_dec,mtk_vcodec_enc,mtk_jpeg, Live 0xffffb5da04f85000
videobuf2_memops 20480 2 videobuf2_dma_contig,videobuf2_vmalloc, Live 0xffffb5da04f18000
cfg80211 397312 3 ath10k_core,ath,mac80211, Live 0xffffb5da04f9c000
videobuf2_v4l2 32768 5 mtk_vcodec_dec,mtk_vcodec_enc,mtk_jpeg,uvcvideo,v4l2_mem2mem, Live 0xffffb5da04f77000
videobuf2_common 57344 9 mtk_vcodec_dec,mtk_vcodec_enc,mtk_jpeg,uvcvideo,videobuf2_dma_contig,videobuf2_vmalloc,v4l2_mem2mem,videobuf2_memops,videobuf2_v4l2, Live 0xffffb5da0504f000
videodev 237568 7 mtk_vcodec_dec,mtk_vcodec_enc,mtk_jpeg,uvcvideo,v4l2_mem2mem,videobuf2_v4l2,videobuf2_common, Live 0xffffb5da04f27000
mc 53248 6 mtk_vcodec_dec,uvcvideo,v4l2_mem2mem,videobuf2_v4l2,videobuf2_common,videodev, Live 0xffffb5da04ec3000
cros_usbpd_charger 16384 0 - Live 0xffffb5da04ebe000
cros_usbpd_logger 16384 0 - Live 0xffffb5da04ee3000
hci_uart 77824 0 - Live 0xffffb5da04f04000
btqca 20480 1 hci_uart, Live 0xffffb5da04efe000
cros_ec_chardev 16384 0 - Live 0xffffb5da04ef9000
sbs_battery 20480 0 - Live 0xffffb5da04ef3000
btbcm 28672 1 hci_uart, Live 0xffffb5da04eeb000
cros_ec_typec 20480 0 - Live 0xffffb5da03400000
typec 57344 1 cros_ec_typec, Live 0xffffb5da04ed4000
bluetooth 503808 10 hci_uart,btqca,btbcm, Live 0xffffb5da04e42000
ecdh_generic 16384 1 bluetooth, Live 0xffffb5da04e3a000
mtk_scp 20480 1 mtk_vcodec_common, Live 0xffffb5da04e30000
i2c_hid_of 16384 0 - Live 0xffffb5da04e2b000
ecc 36864 1 ecdh_generic, Live 0xffffb5da04e1d000
i2c_hid 28672 1 i2c_hid_of, Live 0xffffb5da04e11000
rtc_mt6397 16384 1 - Live 0xffffb5da04e09000
mtk_rpmsg 20480 1 mtk_scp, Live 0xffffb5da04e00000
mt6577_auxadc 20480 0 - Live 0xffffb5da04df7000
elants_i2c 24576 0 - Live 0xffffb5da04dec000
mtk_wdt 20480 1 - Live 0xffffb5da04de2000
rfkill 36864 7 cfg80211,bluetooth, Live 0xffffb5da033e8000
coreboot_table 16384 0 - Live 0xffffb5da033e0000
mtk_scp_ipi 16384 2 mtk_vcodec_common,mtk_scp, Live 0xffffb5da033d8000
fuse 131072 3 - Live 0xffffb5da04dc1000
ip_tables 32768 0 - Live 0xffffb5da03407000
x_tables 45056 1 ip_tables, Live 0xffffb5da033f4000
ipv6 475136 0 [permanent], Live 0xffffb5da03363000

```

hexdump0815, 2022-05-30:
this looks like problems with the sound driver in the kernel - i think this will take a bit longer - maybe best would be to get some cheap usb audio dongle if required in the meantime ... i'll let you know if i find out anything - thanks a lot for testing

BenjaminWegener, 2022-05-30:
no problem, thank you very much

BenjaminWegener, 2022-05-31:
@hexdump0815 i tested jammy, same result...
i am now digging into the 5.10 original kernel on the internal mmc, found out that there are a few more modules installed:
```
kernel/arch/arm64/kvm/kvm.ko
kernel/arch/arm64/crypto/sha1-ce.ko
kernel/arch/arm64/crypto/sha2-ce.ko
kernel/arch/arm64/crypto/ghash-ce.ko
kernel/arch/arm64/crypto/aes-ce-cipher.ko
kernel/arch/arm64/crypto/aes-ce-ccm.ko
kernel/arch/arm64/crypto/aes-glue-ce.ko
kernel/arch/arm64/crypto/aes-glue-neon.ko
kernel/arch/arm64/crypto/sha256-arm64.ko
kernel/arch/arm64/crypto/aes-arm64.ko
kernel/mm/zsmalloc.ko
kernel/fs/binfmt_misc.ko
kernel/fs/binfmt_script.ko
kernel/fs/binfmt_elf.ko
kernel/fs/compat_binfmt_elf.ko
kernel/fs/mbcache.ko
kernel/fs/quota/quota_v2.ko
kernel/fs/quota/quota_tree.ko
kernel/fs/configfs/configfs.ko
kernel/fs/ext4/ext4.ko
kernel/fs/jbd2/jbd2.ko
kernel/fs/squashfs/squashfs.ko
kernel/fs/ecryptfs/ecryptfs.ko
kernel/fs/esdfs/esdfs.ko
kernel/fs/exportfs/exportfs.ko
kernel/fs/nls/nls_base.ko
kernel/fs/pstore/pstore.ko
kernel/fs/pstore/ramoops.ko
kernel/security/keys/encrypted-keys/encrypted-keys.ko
kernel/crypto/crypto.ko
kernel/crypto/crypto_algapi.ko
kernel/crypto/aead.ko
kernel/crypto/geniv.ko
kernel/crypto/skcipher.ko
kernel/crypto/seqiv.ko
kernel/crypto/echainiv.ko
kernel/crypto/crypto_hash.ko
kernel/crypto/akcipher.ko
kernel/crypto/kpp.ko
kernel/crypto/rsa_generic.ko
kernel/crypto/crypto_acompress.ko
kernel/crypto/cryptomgr.ko
kernel/crypto/hmac.ko
kernel/crypto/crypto_null.ko
kernel/crypto/md5.ko
kernel/crypto/sha1_generic.ko
kernel/crypto/sha256_generic.ko
kernel/crypto/sha512_generic.ko
kernel/crypto/gf128mul.ko
kernel/crypto/ecb.ko
kernel/crypto/cbc.ko
kernel/crypto/cts.ko
kernel/crypto/xts.ko
kernel/crypto/ctr.ko
kernel/crypto/gcm.ko
kernel/crypto/cryptd.ko
kernel/crypto/des_generic.ko
kernel/crypto/aes_generic.ko
kernel/crypto/arc4.ko
kernel/crypto/deflate.ko
kernel/crypto/crc32c_generic.ko
kernel/crypto/crct10dif_common.ko
kernel/crypto/crct10dif_generic.ko
kernel/crypto/authenc.ko
kernel/crypto/authencesn.ko
kernel/crypto/rng.ko
kernel/crypto/drbg.ko
[... 393 lines cut ...]
kernel/net/ipv4/netfilter/ipt_REJECT.ko
kernel/net/ipv4/inet_diag.ko
kernel/net/ipv4/tcp_diag.ko
kernel/net/ipv4/udp_diag.ko
kernel/net/ipv4/tcp_cubic.ko
kernel/net/xfrm/xfrm_algo.ko
kernel/net/xfrm/xfrm_user.ko
kernel/net/xfrm/xfrm_ipcomp.ko
kernel/net/unix/unix.ko
kernel/net/ipv6/ipv6.ko
kernel/net/ipv6/netfilter/ip6_tables.ko
kernel/net/ipv6/netfilter/ip6table_filter.ko
kernel/net/ipv6/netfilter/ip6table_mangle.ko
kernel/net/ipv6/netfilter/ip6table_raw.ko
kernel/net/ipv6/netfilter/nf_defrag_ipv6.ko
kernel/net/ipv6/netfilter/nf_socket_ipv6.ko
kernel/net/ipv6/netfilter/nf_tproxy_ipv6.ko
kernel/net/ipv6/netfilter/nf_reject_ipv6.ko
kernel/net/ipv6/netfilter/ip6t_rpfilter.ko
kernel/net/packet/af_packet.ko
kernel/net/key/af_key.ko
kernel/net/rfkill/rfkill.ko
kernel/virt/lib/irqbypass.ko
```
i will look after the ucm config there too, i will recompile the kernel with CONFIG_SND_MT8192 and some others.

milan475, 2022-05-31:
Hi Guys, thanks for all the work being done here. Bought this Chromebook yesterday to run linux on it. Got to say it runs a lot better then I expected. 

Quite snappy and not sure if the 10h battery life indication is anywhere near correct. But it seems to sip energy.

The lack of sound / bluetooth is however a bit of a shame. I'll do some experiments tonight when I'm done with work for today. I was thinking if it would be possible to copy over the ChromeOS kernel to the kernel partition on the SD card. Using this process: 

https://wiki.debian.org/InstallingDebianOn/Samsung/ARMChromebook#Installing_to_an_SD_card_using_the_ChromeOS_kernel

What do you guys think of this?

BenjaminWegener, 2022-05-31:
> Hi Guys, thanks for all the work being done here. Bought this Chromebook yesterday to run linux on it. Got to say it runs a lot better then I expected.
>
> Quite snappy and not sure if the 10h battery [...]

its possible, but that script is too old IMHO

milan475, 2022-05-31:
The documentation is a bit old indeed. But the process of copying the kernel to an SD-card shouldn't have changed much. I'll see what I can do tonight.

hexdump0815, 2022-05-31:
@BenjaminWegener - lets maybe better discuss the audio topic in a separate issue: https://github.com/hexdump0815/imagebuilder/issues/54

@milan475, @BenjaminWegener - lets maybe better discuss the broken bluetooth in its own issue as well (seems to affect all kukui devices): https://github.com/hexdump0815/imagebuilder/issues/55

milan475, 2022-06-06:
I just tried the new image and tested bluetooth and sound. From what I could see so far it works like a charm. Thanks so much for all the work @BenjaminWegener and @hexdump0815

antonioaneiros, 2022-06-07:
Just a quick note to thank you for your great work. 
I've been testing the latest bullseye image and it works perfectly. Thank you ;-)

nilsonr91, 2023-01-25:
Greetings everyone. I came here looking to completely replace ChromeOS with Linux. I have followed all the steps that I have seen here and I manage to boot the image from the usb but now I don't know what I should do so that it is installed in the emmc and only linux is left working. If someone can guide me I would greatly appreciate it.

hexdump0815, 2023-01-26:
@nilsonr91 - this is what i'm using to install new chromebooks: https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-with-luks-full-disk-encryption.txt ... due to the fact that this document contains a lot more information beyond doing a simple install on a chromenook only it might be very hard to read ... as the question is coming up more and more often, i plan to write some doc for the simple case of just installing to emmc on a chromebook during the next weeks ...