velvet-os/imagebuilder #50: Sound and mic not working on Kappa (Ubuntu Focal SD Card Image)

aneeshlingala, 2022-05-23:
Sound is not working on kappa, I have alsa installed. It shows dummy output for sound and the mic. Please fix it fast.

hexdump0815, 2022-05-23:
@aneeshlingala - i have sound working on my kappa, so in principle it works ... as the linux kernel v5.18.0 and mesa v22.1.0 i'm right now in the process of building new kernels and images and most probably it will work with those then - so please wait a few more days and then the chances are good that it will work ... but the sound from the built in mic will be very low and the mic connector does not work yet, so if you rely on a working mic, better get a cheap usb audio dongle for now ...

aneeshlingala, 2022-05-24:
Oh, thanks!

aneeshlingala, 2022-05-24:
Ok thanks!

aneeshlingala, 2022-05-27:
When can we expect the release? It has been a few days now...

hexdump0815, 2022-05-27:
sorry - this is a spare time project, so it will move on whenever i have time - "a few days" might be within the next week or within the next weeks depending on how well everything is working ...

aneeshlingala, 2022-05-28:
Ok

hexdump0815, 2022-05-28:
new kukui images are ready: https://github.com/hexdump0815/imagebuilder/releases/tag/220528-01

closed 2022-05-29 by aneeshlingala