velvet-os/imagebuilder #51: Installation for krane

Mr-NI, 2022-05-25:
Is there is a script to install the image onto krane? I want to attempt to daily drive Linux on the chromebook

hexdump0815, 2022-05-26:
@Mr-NI - there are the images here: https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_kukui to install it on an sd card to boot from or and i think you should also be able to dump the image to emmc to run it from there (after booting from an sd card as mentioned before). what i'm usually doing is to follow this document: https://github.com/hexdump0815/imagebuilder/blob/main/info/generic/install-to-emmc-with-luks-full-disk-encryption.txt whenever i install on a new chromebook - i admit the document is not that easy to read and it contains information not only for installation on chromebooks, but also for other systems (efi or u-boot booted systems etc.) - i hope it still gives some directions on how to do it ... if you plan to run from emmc, then it is highly recommended to set the gbb flags so that usb booting is enabled by default - more information about all this can be found here: https://github.com/hexdump0815/linux-mainline-on-arm-chromebooks

one last note: i'm currently in the process of uilding fresh images with the latest v5.18 kernel and v22.1 mesa - so it might be worth to wait a few more days until they are ready

best wishes - hexdump

hexdump0815, 2022-05-28:
i have just created some new kukui images: https://github.com/hexdump0815/imagebuilder/releases/tag/220528-01

Mr-NI, 2022-06-05:
is there a script I can use? This is a bit complicated for me

hexdump0815, 2022-06-05:
sorry there is no script - for what exactly you would like to have a script? btw. better wait one or two more days as new kukui images will come soon with a few more fixes ...

Mr-NI, 2022-06-14:
I mean, a script to install linux onto the EMMC