velvet-os/imagebuilder #52: system status: chromebook_kukui: juniper and kappa

hexdump0815, 2022-05-28:
## Notes on juniper and kappa running 5.18.0 (i use one issue for both systems together now, but might spin off separate ones if needed or useful):

Here is list of working and non working features on the acer spin cp311-3h (juniper) and hp chromebook 11a (kappa).

### Working
• wifi
• builtin display
• energy: battery status, dpms, suspend/resume)
• usb a and c for basic usb functionality
• webcam
• gpu opengl: via mesa panfrost (opengl 3.1)
* sound: speaker and headphone jack work, internal mic works with verly low levels

### untested
• hw video encode/decode: most probably not working yet

### broken
• bluetooth: was working with earlier kernels up to v5.15.22, seems to affect all kukui devices with newer kernels
• sound: headset mic not working
* external display: not working yet

### Installation
coming soon

### PROBLEMS
none so far

aneeshlingala, 2022-05-31:
External headphones work in kappa, you just have to manually swtich to headphones and vice versa in pulseaudio volume control.

aneeshlingala, 2022-06-04:
External Mic works with USB headphones in kappa, by switching manually through pavucontrol.

gallee, 2024-05-15:
Hello,
On this device with bookworm, I have from time to time black squares on firefox-esr page rendering 
kernel : 6.1.51
firefox : 115.11.0esr 

Is this a known issue ?

aneeshlingala, 2024-05-15:
I updated to the 6.6 kernel and this didnt happen to me on the new kernel

gallee, 2024-05-16:
Shall I built it myself or is there a version available for download ?

aneeshlingala, 2024-05-16:
the kernel link is
https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/download/6.6.9-stb-mt8%2B/6.6.9-stb-mt8+.tar.gz

You have to extract the gzip archive, copy the files to their respective
locations then dd the file ending in .kpart to the 1st and 2nd partitions
of the root disk (ex. /dev/mmcblk0, /dev/sda, etc.)

gallee, 2024-05-21:
Thanks @aneeshlingala

I installed the newer kernel by following the [installing-a-newer-kernel](https://github.com/hexdump0815/imagebuilder/blob/main/doc/installing-a-newer-kernel.md) guide. 

In your comment, you are saying that the dd copy shall be done **on the 1st and 2nd partitions**. However the guide only mentions dd'ing into the first partition (mmcblk0p1 in my device).
Booting on the new kernel seems to be ok with only one partition .
Am I missing something ?

aneeshlingala, 2024-05-21:
i dd to the first 2 partitions with the second partition being a backup,
but dding to the first partition only is also completely fine

gallee, 2024-08-22:
Hello,

I tried to put my wifi card in monitor mode, that triggers a firmware crash (full dmesg [here](https://gist.github.com/gallee/8ba56a7a28757826409cca4bde41b4d7)

I am on Ubuntu Jammy with `linux-firmware` at `20220329.git681281e4-0ubuntu3.31`

There is some new version of the binary blobs available on the [kernel repository](https://gitlab.com/kernel-firmware/linux-firmware/-/tree/main/ath10k/QCA6174/hw3.0?ref_type=heads) notably the [sdio one](https://gitlab.com/kernel-firmware/linux-firmware/-/blob/main/ath10k/QCA6174/hw3.0/firmware-sdio-6.bin?ref_type=heads)

If I want to give it a try, do you know what is the best procedure to update it ?

hexdump0815, 2024-08-22:
i would suggest to first make a copy of the current version, copy the new firmware to the system (as you might end up without wifi), replace the old one and see if things still work or are better after a reboot

gallee, 2024-08-23:
@hexdump0815  thanks for your answer. 

I managed to have the `WLAN.RMH.4.4.1-00174` version running. However, it also resulted in a crash when trying to put the wlan interface in monitor mode. 

I guess the binary driver does not support monitor mode. 

Cheers !

hexdump0815, 2024-08-23:
could be - thanks for testing

lockieluke, 2024-11-20:
this might be unrelated i'm shimboot to boot into debian on kappa, how do i install the gpu driver for wayland?  also how did you manage to get the speakers working, mine shows as Dummy Output

lockieluke, 2024-11-20:
thanks for confirming can someone make sure that Wayland doesn't use software rendering with this image before i make the move from shimboot to this

thenameisluk, 2024-11-20:
@lockieluke i do own kukui device and can confirm it works with panfrost gpu driver

the operation isn't irreversible anyway so why do you hesitate
you can always backup ur date on google account

how to start :
https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/readme.md

lockieluke, 2024-11-20:
because currently i'm using https://github.com/ading2210/shimboot i don't have a spare sd card to test this without wiping my current installation

thenameisluk, 2024-11-20:
@lockieluke yes the stuff works on the latest image

you can even install it directly onto chomebook memory and have it default os if you set gbb flags

lockieluke, 2024-11-20:
can i dual boot this off an sd card on a managed device?  many thanks

thenameisluk, 2024-11-20:
@lockieluke yes both dual boot and install directly

lockieluke, 2024-11-20:
I don't think installing on a **managed**(enrollment enabled + developer mode disabled) device is supported while https://github.com/ading2210/shimboot support this

aneeshlingala, 2025-02-08:
> I don't think installing on a **managed**(enrollment enabled + developer mode disabled) device is supported while https://github.com/ading2210/shimboot support this

If you do try, you could face consequences
I would recommend reading company/school policy if this is not your device.