velvet-os/imagebuilder #53: system status: chromebook_kukui: krane (lenovo chromebook duet 10.1)

hexdump0815, 2022-05-28:
## Notes on krane running 5.18.0:

Here is list of working and non working features on the lenovo chromebook duet 10.1 (krane).

### Working
• wifi
• builtin display
• energy: battery status, dpms, suspend/resume
• usb c for basic usb functionality
• gpu opengl: via mesa panfrost (opengl 3.1)
* sound: speaker works, internal mic works with verly low levels

### untested
• hw video encode/decode: most probably not working yet

### broken
• bluetooth: was working with earlier kernels up to v5.15.22, seems to affect all kukui devices with newer kernels
• webcams: not working yet, unclear if they can be made working as they are just simple cameras and no usb video class cameras like on other chromebooks
* external display: not working yet

### Installation
coming soon

### PROBLEMS
none so far

alpenamilch, 2023-06-01:
Is this https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md a correct way of installing to emmc on ideapad duet 1.gen 4GB/128GB?

hexdump0815, 2023-06-02:
yes - that should be the correct way

vulpes2, 2023-12-02:
@hexdump0815 I have some updated status based on a more recent kernel + userspace combination, it would be great if you can update the issue description with these. I think GNOME's tablet-style UI works pretty well on this device, so I've replaced xfce with it. It's still a little sluggish but it's somewhat tolerable.

## Software setup

- kernel: [6.5.5](https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/tag/6.5.5-stb-mt8%2B)
- distro: Debian sid
- mesa: 23.2.1 (from the Debian sid repo)
- window protocol: wayland
- desktop environment: gnome

## Working

- Wi-Fi (including WPA3-SAE and 802.11w) and Bluetooth (PCM offloading is untested)
- Built-in display (orientation, brightness controls)
- Energy: battery status, dpms, suspend/resume
- IIO sensors: ambient light sensor (auto brightness), accelerometer (untested)
- USB: basic USB host and charging only
- Touchscreen: works out of the box, orientation is correct and follows screen rotation on GNOME
- Graphics: OpenGL works out of the box with panfrost, zero screen tearing with wayland.
- Audio: speakers work fine

## Needs work

Will investigate in these when I have more time.

- Hardware video decoding/encoding: firmware is loaded but userspace support is missing. Need to test if `v4l2-request` works at all on this platform.
- Pen digitizer: axes are swapped, needs an accel matrix config. Pen pressure works but registers a touch before the pen touches the screen, needs a way to add an offset before it's usable.
- IIO Sensors: gyroscope (auto-screen orientation) works but is 90 degrees offset from the physical orientation
- Trackpad: very low sensitivity and annoying to use, not a hardware issue as it worked fine previously on Cadmium 

## Broken
- Audio: internal microphones are extremely quiet and completely unusable
- Cameras: likely CSI cameras, no idea if that subsystem of the soc has any mainline support at all
- USB-C display output: the ec fails to configure the USB-C port with the following error message: `No valid DP mode provided.`

## Notes

1. IIO sensors require `iio-sensor-proxy` to be installed and running for them to work with GNOME.
2. Hardware compositing needs to be manually enabled on both Firefox and Chromium. I have only tested wayland so there are no x11 instructions unfortunately. <br>
Make sure both Firefox and Chromium are running in wayland by setting the environment variable `MOZ_ENABLE_WAYLAND=1` for Firefox, and set the flag `ozone-platform-hint` to `auto` for Chromium. Additionally, you need to set `gfx.webrender.all` to `true` in Firefox. After that, you can verify hardware acceleration at `about:support` in Firefox and `chrome://gpu` in Chromium. <br>
Unfortunately I have observed increased system freezes when browsers are using hardware accelerated graphics, but it does make web browsing a lot more tolerable on this device.

vulpes2, 2023-12-03:
I haven't had time to check out kernel 6.6 yet, but I did try to fix the digitizer with some level of success. The pressure range is now correct but rotation still doesn't work yet. Once I figure that out I'll send the quirks to libinput so we don't have to deal with downstream configs anymore. The udev rules can't be submitted to systemd because they're too generic, I'll have to make it more specific to the device before sending them to systemd .

vulpes2, 2023-12-03:
I was unable to get digitizer rotation to work properly, but at least it works properly in one single orientation now, which is better than nothing I guess.

Here's the libwacom config that makes the digitizer configurable with gnome-control-center or kcm-wacomtablet. Add the file as `/etc/libwacom/google-krane.tablet` and then run `libwacom-update-db`, the digitizer should show up in the GUI tools afterwards. Ideally there should be a custom stylus entry that specifies zero buttons, but that can come at a later date. 
```ini
[Device]
Name=hid-over-i2c 27C6:0E30 Stylus
ModelName=
DeviceMatch=i2c:27c6:0e30
Class=ISDV4
Width=5.35433
Height=8.54331
IntegratedIn=Display;System
Styli=@generic-no-eraser

[Features]
Stylus=true
Touch=false
```

This is the libinput quirks file that fixes the trackpad and pen digitizer pressure range, place at  `/etc/libinput/local-overrides.quirks` and reboot. This will at least make the digitizer usable, tested with xournal++ and it seems to work pretty well.
```ini
[Google Chromebook Krane Trackpad]
MatchUdevType=touchpad
MatchName=Google Inc. Hammer
MatchBus=usb
MatchDeviceTree=*krane*
ModelChromebook=1
AttrPressureRange=20:10

[Google Chromebook Krane Stylus Digitizer]
MatchUdevType=tablet
MatchDeviceTree=*krane*
MatchBus=i2c
ModelChromebook=1
AttrPressureRange=1100:1000
```

vulpes2, 2023-12-09:
Tested v6.6 with your configs on krane and nothing seems horribly broken so far compared to v6.5. The high pitched noise was still a problem as of v6.5. I made a small change to the dt on top of your patches,

janamach, 2023-12-10:
Hi there,

Piggy-backing on this discussion as I just installed 230917-01 ubuntu jammy on my Lenovo IdeaPad Duet Chromebook 10.1 (MT8183), running GNOME 42.9 (alternating between Xorg and Wayland) and kernel version 6.1.51-stb-mt8+.

@vulpes2, your instructions [above](https://github.com/hexdump0815/imagebuilder/issues/53#issuecomment-1837517427) worked beautifully on my system with Lenovo USI pen! Tested by doodling in inkscape and krita. The latter one amazed me: I thought this USI pen was defect as it never worked properly on Chrome OS, but with Krita is works so incredibly well, it's an absolute pleasure to draw!

Other than that, the system seems to run as one would generally expect (comparing with other computers I own with Intel processors running different versions of Ubuntu and Debian), no major disasters so far. Hardware acceleration works as described above, tested with firefox and youtube videos (huge difference).

### Broken

* Internal mic very quiet as reported above
* Webcam is not detected
* Screen rotation is doing weird things (all on GNOME):
    * With Xorg, the screen is rotated at login, to put it in the correct orientation I use `xrandr -o left`. Then when I turn it into either of the two possible portrait modes, I need to use either  `xrandr -o normal` or `xrandr -o inverted`. If I flip it upside-down in landscape mode, then it's  `xrandr -o right`. My temporary workaround is getting it into the desired orientation and locking screen rotation.
    * On Wayland it starts normally when logging in default orientation (i.e., landscape), but turning it once messes up the behavior is a similar way as on Xorg.

Do you think upgrading to kernel 6.5 or 6.6 would fix the screen rotation issue?

vulpes2, 2023-12-10:
> Manually locking the screen and then unlocking it rotates the screen rotates the screen to portrait mode.

By default the sensor orientation is 90 degrees off. Please make sure you have added this [udev rule](https://wiki.postmarketos.org/wiki/Lenovo_IdeaPad_Duet_Chromebook_(google-krane)) mentioned above to correct the sensor mount orientation. Keep in mind that even if auto-rotation is disabled on GNOME when the device is not in tablet mode (keyboard folded to the back or detached), it will still auto-rotate your screen to the default/upright orientation according to the sensor when the machine is woken up with the keyboard attached, which is why you're seeing an incorrectly rotated screen after waking up the device from standby. The udev rule will address that and all the orientation issues should be gone after you add it, I don't think upgrading the kernel is necessary.

janamach, 2023-12-11:
Thank you for explaining! Adding that udev rule has indeed fixed the rotation issue.

> auto-rotation is disabled on GNOME when the device is not in tablet mode

Ah, tablet mode! What confused me was Xorg, it was rotating the screen regardless of the keyboard position, I thought that was the expected behavior. Now it all makes a lot of sense. With the orientation being fixed, Gnome on Wayland behaves as a laptop with the keyboard attached and "unfolded", and as a tablet when the keyboard is detached or folded to the back.

> Thank you for the feedback, I was particularly interested in the USI pen because I was not 100% sure if it was just my USI pen or digitizer being weird or not. Given it's not a defect on my device, I will be able to send these quirks to libinput now.

Happy if I can help! Anything else I should test?

vulpes2, 2023-12-11:
@hexdump0815 I've opened a few pull requests to do some basic cleanup but most of these aren't specific to krane.
- https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/pull/11
- https://github.com/hexdump0815/kernel-config-options/pull/4
- https://github.com/hexdump0815/imagebuilder/pull/183

The libinput quirks for krane and wormdingler will be submitted to libinput very soon, but the udev rule will probably have to remain downstream for now since udev doesn't seem to be able to apply dt based filtering with hwdb like libinput can with the quirks database, so if you don't have ACPI it seems to be all-or-nothing unfortunately. On top of that, the udev rule from the pmos wiki seems to be a hack too, I found out that the property was applied to every single input event devices on the system, which doesn't necessary break anything but probably still not a good practice nonetheless.

@janamach If you're on GNOME you may encounter this bug too, where the touchscreen will become unresponsive and appear frozen until you press Esc on a physical keyboard. It's reportedly fixed in 45 but I don't know if that will be backported to older releases: https://gitlab.gnome.org/GNOME/mutter/-/issues/3181

janamach, 2023-12-11:
@hexdump0815 I really appreciate your initiative with the images! At this point I still lack a lot of knowledge for being able to port a device myself, so I admire those who can. This said, I now own a super awesome tablet all thanks to your project. Thank you :)

@vulpes2 I have not encountered that bug specifically yet, but I've observed the following weirdness on GNOME:
* Connecting a monitor through HDMI->USB-c disables touchpad and physical keyboard. The functionality is restored after reboot. The monitor is detected by the tablet, but not the other way around, so there is no actual HDMI output, but tablet thinks there is. So there are two bugs in one :)
* I have observed GNOME randomly refusing to revert to desktop mode after being used in tablet mode. This is fixed by closing the lid (or sending to sleep) for a moment.

What works consistently is the USI pen, I am still amazed. I wanted to trash that thing, it was frustratingly useless on Chrome OS and every web search would lead me to posts of other people complaining. Some said they sent theirs back and received a functional replacement from Lenovo, so I thought mine is just broken (or from a bad batch). But now am convinced it's all software. Weird that Lenovo and Google couldn't get it to work properly, this pen had so much hidden potential.

janamach, 2023-12-12:
As I see. External display output is not a big deal for me, I wasn't planning on using it like that anyway.

Not sure if my USI pen drains the battery like that. I did have to replace it a couple of times, I'll keep an eye on it now. The main problem with it was that it was "too sensitive": it would write and click on things while the tip was not touching the screen. IIRC, this distance to the screen when it would start doing things was also not consistent. People reported that different hacks worked for them, e.g., removing the battery for some time and inserting it back in while pressing on the tip, or inserting a spacer ring just above the tip and so on. None of that really worked for me, it was too uncomfortable to use. I also did not find a good use for Chrome OS, so I left the whole thing lying in the corner for months. Until now ;-)

External display output is not a big deal for me, I wasn't planning on using it like that anyway, a linux tablet with a working pen is **the** thing that was missing in my life :-) Would be nice to have the mic and the cameras working though. Anyone working on those issues?

hexdump0815, 2023-12-12:
@vulpes2 - thanks a lot for the pull requests - they all looked good to me and i already merged them

external display via usb-c never really worked with mainline on kukui as far as i know - the patch floating around seems to only add the corresponding dt pieces from the chromeos kernel, but i think the type-c code in chromeos has some extras needed for that, which are not in mainline ... i'm not sure, but maybe this looks like something getting closer there with mainline: https://patchwork.kernel.org/project/dri-devel/cover/20230331091145.737305-1-treapking@chromium.org/ - i'm not 100% sure and did not play around with it myself yet though

tech-with-mo, 2024-03-16:
> I was unable to get digitizer rotation to work properly, but at least it works properly in one single orientation now, which is better than nothing I guess.
>
> Here's the libwacom config that makes [...]

So for this I found a really simple fix.

First you need to identify the Stylus using `xinput list` After that note down the ID. In my case it was 14. Now paste this command into you're Terminal. `xinput set-prop 14 "Coordinate Transformation Matrix" 0 -1 1 1 0 0 0 0 1
` Remember to replace _14_ with the actual ID of you're Stylus. After that, test.

cheungyau2022, 2024-04-07:
Hi I am not sure if this is the right place to ask but I recently got a Lenovo Chromebook Duet 10.1 and it came with Ubuntu 22.04 installed by someone. I want to know if there is a way to wipe the Linux system and restore it to Chrome OS? I did some searching but have yet found anything useful. I tried MrChromebox's script but it says ARM64 devices are not supported. Can someone post a link to any guide related to my issue? Thank you very much!

vulpes2, 2024-04-07:
Follow the instructions here: https://support.google.com/chromebook/answer/1080595?hl=en, ignore the "less invasive steps", start from `Recovery option 2: Use USB drive`.

osfanbuff63, 2024-05-13:
> ## Software setup
> 
> * kernel: [6.5.5](https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/tag/6.5.5-stb-mt8%2B)
> 
> * distro: Debian sid
> 
> * mesa: 23.2.1 (from the Debian sid repo)
> 
> * window protocol: wayland
> 
> * desktop environment: gnome
> 
> (truncated from https://github.com/hexdump0815/imagebuilder/issues/53#issuecomment-1837137317)

Is there any way to get a bootable image of this so that we don't have to try to get all that set up after installing?

janamach, 2024-05-14:
@osfanbuff63 are you looking for this or something else?

https://github.com/hexdump0815/imagebuilder/blob/main/systems%2Fchromebook_kukui%2Freadme.md

osfanbuff63, 2024-05-23:
Those still run on XFCE last time I tried, but I'll check it again to make sure

LukIsHere, 2024-06-28:
so i got my hands on duet 10.1
was very cheap for some reason :3 
but is have a slight problem with it
[image]
i can't seam to be able to set gbb flags for it
i tried [unplugging the battery](https://forum.chrultrabook.com/t/lenovo-duet-ideapad-google-krane-disable-write-protect/1840) but it doesn't work
[image]
anyone know where can i find a write protection screw or sth?
there is no hardware maintenance manual (i could find) so 'm out of ideas

hexdump0815, 2024-06-28:
@LukIsHere - i also ran into this problem that the battery-disconnect option does not seem to work for krane ... you can ccd open it via suzyqable (see: https://github.com/hexdump0815/linux-mainline-on-arm-chromebooks) and in case you do not have one (i would really recommend to get/build one if you deal with as many chromebooks as you do now :) ) then maybe search around a bit for "sh1mmer" - with that approach and some modifications i  was even able to get linux booted from usb on a krane with dead emmc (and thus no regular way to ccd open it due to the battery problem you mentioned) :)

LukIsHere, 2024-06-28:
there is no way i can order one at reasonable price, and building on
o my
it sounds scary and looks even scarier
i've never used resistors in my entire life
can probably order them online but the site says 1% and shop says 5%, i don't want to mess it up so will probably have to visit my local electronics store so ppl there can help me get the right parts
will give it a shot ^~^
_Sidenote. i think i managed to get depthcharge tools working but it's producing images at 46mb (kinda over 32mb, it's so bloated :3) so i have to do fresh emmc install to test if these even work_

hexdump0815, 2024-06-29:
@LukIsHere - too large images sound like kernel and initrd are not heavily enough compressed - see  https:/https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-with-luks-full-disk-encryption.txt#L213-L231 and /github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-with-luks-full-disk-encryption.txt#L235-L285 - this way it can be made small enough (i.e. < 32m) - no idea if there is a way to teach dethcharge tools this - i did not look into it yet

LukIsHere, 2024-06-29:
lack of compressing is least of my concerns since it doesn't boot
also we should probably move this discussion to a separate issue

tech-with-mo, 2024-07-01:
Hi, are we expected to get any new updates for the new Ubuntu releases (23, 24)?

hexdump0815, 2024-07-01:
@tech-with-mo - i plan to build newer version, but just do not have the time for it right now - i already added ubuntu noble support to the framework, but found some problems on my first test images which i'll have to sort out first ... what @LukIsHere suggested might work for some setup or not for others (zstd compressed kernel modules in ubuntu noble being the biggest problem - see https://github.com/hexdump0815/kernel-config-options/commit/592147c6c98944aed337af6003d79fc8e3f524c6 ) ... newer images will come, its just not clear yet when

hexdump0815, 2024-07-02:
@LukIsHere - that was quick for the suzyqable, nice to hear that you got one working - for krane (and tablets in general - i would say in in general for all chromebooks actually even) the pp button is the power button (on the tablet part for krane) - just a quick press is enough usually

ttyUSB* devices should also show up on debian, but will not on kukui devices (and maybe other chromebooks with mainline kernel too) as the usb stack seems to still be a bit buggy for them, at least buggy enough for not working well with the suzyqable ... one thing to also keep in mind is that the orientation in which the suzyqable is plugged into the target device (chromebook to ccd open) matters - it only works in one direction/orientation

good luck and best wishes - hexdump

LukIsHere, 2024-07-18:
@hexdump0815
i tested it
it's bronke on trogdor
- acer spin 513
- lenovo ideapad duet 5
kukui
- lenovo 10e

but works on corsola
- lenovo ideapad slim 3

would it be good idea to use the chromebook spin of kelner for kukui/trogdor?

hexdump0815, 2024-07-18:
@LukIsHere - what exactly is broken? (i lost the overview a bit) ... it does not work to use the trogdor kernel for kukui as they still need different sets of patches (trogdor close to none, but kukui need still quite a few) and as put together at the moment also the required dtb files for the other wardware are not included.

LukIsHere, 2024-07-18:
@hexdump0815 it simply doesn't appear when ```ls /dev/ttyU*```
i know it's not cable orientation issue cause i used usb-A cable port (if present, on ideapad slim 3 and acer spin) and it worked on corsola

hexdump0815, 2024-07-20:
@LukIsHere - it could be that the usb stack has a problem on both kukui and trogdor - i remember that kukui was always a bit problematic regarding usb but i never tried trogdor ... what might be worth a try it to give anewer kernel like v6.10 a try on kukui - i did some tests with v6.9 some weeks ago and usb looked a bit better, but sadly the display support was broken on krane (which i tested it on)

thenameisluk, 2024-08-02:
yeah, squable isn't the only thing not working i have also **usb cd reader** that works only on corsola
but **requires** ```sudo modprobe isofs``` to be run in order to read the files
will give v6.9 a shot when i finish working on waydroid

thenameisluk, 2024-08-05:
6.9 doesn't fix the suzyqcable, 6.10 even breaks usb :3
the device detects something is plugged in
```
root@chluk:/compile/source/linux-stable-cbq# lsusb 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 0bda:5414 Realtek Semiconductor Corp. USB2.1 Hub
Bus 001 Device 003: ID 18d1:5052 Google Inc. Hammer
Bus 001 Device 033: ID 18d1:5014 Google Inc. Cr50
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0bda:0414 Realtek Semiconductor Corp. USB3.2 Hub
```
but there still is no tty
```
root@chluk:/compile/source/linux-stable-cbq# ls /dev/ttyU*
ls: nie ma dostępu do '/dev/ttyU*': Nie ma takiego pliku ani katalogu <- no such file or directory
```

hexdump0815, 2024-08-06:
looks like usb on kukui is not in a good shape right now - i'll have to look deeper into it once i find a bit more time

vulpes2, 2024-09-24:
Touchpad quirks for krane, wormdingler and coachz have been submitted to libinput. The pen pressure issue is not the fault of the digitizer and cannot be upstreamed, because the Levovo USI Pen is defective. Many people have been experiencing similar issues in ChromeOS as well with this specific pen. Will try to get the digitizer rotation and gyro offset fixed upstream when I have more time.

vulpes2, 2024-10-06:
Touchpad quirks for krane, wormdingler and coachz have been merged and will be in the next libinput release. There are a few more things that needs to get done, will get to them at some point.

- [ ] Submit tablet information to libwacom (krane, wormdingler, coachz)
- [ ] Upstream the `ACCEL_MOUNT_MATRIX` udev rules (krane, wormdingler, coachz)
- [ ] Fix pen digitizer rotation and ensure auto-rotate works (krane, wormdingler, coachz)
- [ ] Look into the mic gain situation for krane, gain setting might be applied in either dt or by a daemon in userspace on ChromeOS

vulpes2, 2024-10-06:
Looked into the mic gain situation, there is nothing we can tweak in the codec, CRAS is applying a 20db software gain via the `IntrinsicSensitivity "-2600"` directive in their UCM config. I'm not sure if there is a way to apply a hidden 20db gain with ALSA so the volume controls will work properly. Docs: https://chromium.googlesource.com/chromiumos/third_party/adhd/+/refs/heads/main/docs/ucm.md#audio-nodes-cras

hexdump0815, 2024-10-06:
@vulpes2 - first a big thank you for all your effort and double so for bringing all that to mainline of the corresponding projects which i would say is absolutely amazing ... regarding the mic gain boost, did you see my note in the readme of kukui to maybe use "pactl set-source-volume 1 400%" - 400% (i think that was the limit, but not completely sure anymore) would at least be 12db, still not 20db - maybe the more and more upcoming pipewire offers more options for such a fixed gain within the audio chain ... the target for the next round of images would be debian trixie where we should maybe try to switch to pipewire for audio if easily possible and if it would help for such things ...

vulpes2, 2024-10-06:
I did see the notes about the 400% volume with pulseaudio, I'm hoping there's a way to "hide" that 20db gain somewhere, so the end user still gets a "normal" 0-100% volume control that actually works as intended. I skimmed through the existing upstream UCM configs and they all use hardware gain controls, which is not available on this particular audio setup. PipeWire doesn't have the ability to apply audio gain transparently either.

EnderRobber101, 2025-01-04:
Is there a new version based on gnome instead of xfce? 
[image]

EnderRobber101, 2025-01-05:
Can i update simply but clicking on the upgrade to ubuntu lts 24?

thenameisluk, 2025-01-05:
@EnderRobber101 no there is no server image

updating to newer ubuntu cause problems atm so i would avoided it, you can use latest debian without any issues

xx1182, 2025-01-26:
> [@EnderRobber101](https://github.com/EnderRobber101) no there is no server image
> 
> updating to newer ubuntu cause problems atm so i would avoided it, you can use latest debian without any issues

(1) Restored ChromeOS by pressing UP+DOWN+POWER
(2)When is the first boot pressed ALT F3 for tty3 and logged with chronos
Then
(4)Followed this insteuctions witouth setting gbb flags
https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/readme.md

... https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/installation/basic-installation.md

With downloaded the latest release
https://github.com/hexdump0815/imagebuilder/releases/tag/230917-01

///
Some stuff did not work but saw that in the docs there was something about fixing the touchpad etc planning to do it later...
///

DID do-release-upgrade AND WIFI NOT WORK ANYMORE.

Can the docs be cleaned up?
Cause also, 

> Is this https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md a correct way of installing to emmc on ideapad duet 1.gen 4GB/128GB?

Is old link

vulpes2, 2025-03-06:
Anyone who's got a working system right now, could you please submit a hardware info dump at libwacom to get this device properly supported upstream? Thanks. https://github.com/linuxwacom/wacom-hid-descriptors/tree/master?tab=readme-ov-file#contributing

oznogon, 2025-03-07:
> Anyone who's got a working system right now, could you please submit a hardware info dump at libwacom to get this device properly supported upstream? Thanks. https://github.com/linuxwacom/wacom-hid-descriptors/tree/master?tab=readme-ov-file#contributing

Done: https://github.com/linuxwacom/wacom-hid-descriptors/issues/469#issue-2903821424

vulpes2, 2025-03-07:
Thanks, forgot to mention this was already sorted out shortly after I posted the original comment.

DRCRecoveryData, 2025-05-21:
Does chromebook_kukui have lastest build?

thenameisluk, 2025-05-21:
@DRCRecoveryData 
define "lastest"

if you want current stable debian (bookworm) https://github.com/velvet-os/imagebuilder/releases/tag/230917-01
if you want next to be stable debian (trixie) https://github.com/velvet-os/imagebuilder-testimages/releases/tag/spring-release-2025-testimages-round-01

DRCRecoveryData, 2025-05-22:
> [@DRCRecoveryData](https://github.com/DRCRecoveryData) define "lastest"
>
> if you want current stable debian (bookworm) https://github.com/velvet-os/imagebuilder/releases/tag/230917-01 if you want [...]

gnome not good on duet 1, i think install https://plasma-mobile.org/ should be better ui

thenameisluk, 2025-05-22:
@DRCRecoveryData ???
the **images** doesn't **come with** gnome but with **xfce** and there is a good reason for that
while it looks dated it works across all systems and is way less resource intensive than gnome/kde/whatever else

if you want a diffrent desktop environment just install it : https://github.com/velvet-os/velvet-os.github.io/blob/main/postinst/switch-de.md

pupontech, 2025-09-02:
Noob here 
Got the latest trixie up and running 
I didnt do the gbb flags dont know if that effects these problems
I did updates and all
everything is working fine so far except I cannot use wayland sessions it just boots me right out back to lightdm . Not sure if this is something I broke or if its just broken.
I can troubleshoot it if you walk me through what to do but honestly couldn't figure out from googling how to properly.

Thanks for the awesome work

hexdump0815, 2025-09-02:
@pupontech - i think there were reports of broken gnome and maybe kde with trixie (i'm using xfce, so cannot comment) - iirc @thenameisluk has wayland gnome working after going to debian sid via upgrade from the trixie image ... lets hope the problem fixes will also be back ported to trixie at some point as well

EnderRobber101, 2025-10-07:
Hi, I am using https://github.com/velvet-os/imagebuilder-testimages/releases/tag/spring-release-2025-testimages-round-01 and the rotation if 90 degrees off, it is similar to https://github.com/velvet-os/imagebuilder/issues/202 . does the same fix work?

thenameisluk, 2025-10-07:
@EnderRobber101 
depends on device you have
but it's worth a shot

alternatively there is https://velvet-os.github.io/postinst/touch-input-rotate.html for x11

thenameisluk, 2025-10-07:
you can also try
```
0, 1, 0; -1, 0, 0; 0, 0, -1
```
if the values there don't work

EnderRobber101, 2025-11-06:
One question about the kernel, i saw on the raspberry pi CONFIG_ZRAM is enabled to compress ram to have more usable. Since this device only has 4GB, wouldn't it be useful. (does it require a specific chip feature?) Thanks!

hexdump0815, 2026-05-19:
@BeNach0C - you can plug in the usb-a end into another linux computer and then you should get /dev/ttyUSB0...2 there if everything is plugged in properly and you can use some terminal program like minicom to enter the commands or do it via echo i think as mentioned in the chromebook docs ... good luck