velvet-os/imagebuilder #54: problem: chromebook_kukui: fennel14: audio not working

hexdump0815, 2022-05-31:
it looks like the audio driver is not probed properly on fennel14 and a proper ucm file for it needs to be created and tested.

milan475, 2022-05-31:
@BenjaminWegener Are you sure you have kernel 5.10 installed in ChromeOS? I have 4.19 and also have none of the modules in /lib/modules that you are listing.

BenjaminWegener, 2022-06-01:
@milan475 its on dev channel i think, had to reset it lately, didnt use the chromeos since then. yes, verified 5.10. the modules are gzipped. aplay lists the mt8183_da7219_rtsomething as card.

BenjaminWegener, 2022-06-01:
i managed to switch the old `MT8183_MT6358_TS3A227E_MAX98357A` with the new `MT8183_DA7219_MAX98357A` by compiling the module from source and installing it using
[sound-FENNEL14.sh.txt](https://github.com/hexdump0815/imagebuilder/files/8813243/sound-FENNEL14.sh.txt)
but i didnt test the ucm yet.
 
[dmesg.txt](https://github.com/hexdump0815/imagebuilder/files/8813257/dmesg.txt)

milan475, 2022-06-01:
Awesome! Does cat /proc/asound/cards give you anything now?

milan475, 2022-06-01:
I still wonder if het has anything to do with the kernel modules anyway though. If I list the /lib/modules/[KERNEL_VER]/ directory on ChromeOS I get this output. All the MT8183 sound ko files are nowhere to be found
[mods.txt](https://github.com/hexdump0815/imagebuilder/files/8813291/mods.txt)
.

BenjaminWegener, 2022-06-01:
i think they are in -`rw-r--r--. 1 root root 8161 May 24 04:25 snd-sof-mt8195.ko.gz`

no, got no soundcard yet, lsmod shows the old module, have to blacklist it first.

BenjaminWegener, 2022-06-01:
lsmod doesnt shw built-in mods, so `cat /lib/modules/$(uname -r)/modules.builtin` should show some.
i got the new mod installed, but still no sound. same error with pinctrl_paris something

BenjaminWegener, 2022-06-01:
i am new to this, so it's a guess:
in `/compile/source/linux-stable-mt8/arch/arm64/boot/dts/mediatek/` there it says

```
			input-enable;
			bias-pull-up;
		};

		touch_rst_l {
			pinmux = <PINMUX_GPIO156__FUNC_GPIO156>;
			output-high;
		};
	};

	trackpad_pins: trackpad-pins {
		trackpad_int {
			pinmux = <PINMUX_GPIO7__FUNC_GPIO7>;
			input-enable;
			bias-disable; /* pulled externally */
		};
	};

	vddio_mipibrdg_en: vddio-mipibrdg-en {
		pins1 {
			pinmux = <PINMUX_GPIO37__FUNC_GPIO37>;
			output-low;
		};
	};

	volume_button_pins: volume-button-pins {
		voldn-btn-odl {
			pinmux = <PINMUX_GPIO6__FUNC_GPIO6>;
			input-enable;
			bias-pull-up;
```
so i can imagine, the trackpad requests pin 7 and doesnt clear it before the sound driver is loaded, hence the error

milan475, 2022-06-01:
In case it's helpful: same output under chrome OS:

``
pin 0 (GPIO0): 11012000.spi (GPIO UNCLAIMED) function func7 group GPIO0
pin 1 (GPIO1): 11012000.spi (GPIO UNCLAIMED) function func7 group GPIO1
pin 2 (GPIO2): 11012000.spi (GPIO UNCLAIMED) function func7 group GPIO2
pin 3 (GPIO3): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 4 (GPIO4): 4-0058 (GPIO UNCLAIMED) function func0 group GPIO4
pin 5 (GPIO5): volume-buttons pinctrl_paris:325 function func0 group GPIO5
pin 6 (GPIO6): volume-buttons pinctrl_paris:326 function func0 group GPIO6
pin 7 (GPIO7): 2-002c (GPIO UNCLAIMED) function func0 group GPIO7
pin 8 (GPIO8): (MUX UNCLAIMED) (GPIO UNCLAIMED)
pin 9 (GPIO9): 3-005c (GPIO UNCLAIMED) function func0 group GPIO9
``

Seems like "2-002c" is able to claim the pin under ChromeOS...

BenjaminWegener, 2022-06-01:
i was mistaken, after kernel recompile following dmesg appeared:
```
linux@changeme:~$ dmesg | grep -i pin
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1031424
[    0.005019] arch_timer: CPU4: Trapping CNTVCT access
[    0.005588] arch_timer: CPU5: Trapping CNTVCT access
[    0.006089] arch_timer: CPU6: Trapping CNTVCT access
[    0.006590] arch_timer: CPU7: Trapping CNTVCT access
[    0.015170] pinctrl core: initialized pinctrl subsystem
[   14.599065] mt8183-pinctrl 10005000.pinctrl: pin GPIO8 already requested by 2-0015; cannot claim for 2-002c
[   14.600137] mt8183-pinctrl 10005000.pinctrl: pin-8 (2-002c) status -22
[   14.604063] mt8183-pinctrl 10005000.pinctrl: could not request pin 8 (GPIO8) from group GPIO8  on device pinctrl_paris
[   23.202746] mt8183_da7219 mt8183-sound: ASoC: Failed to add route TDM_OUT_PINCTRL -> direct -> I2S Playback
linux@changeme:~$ dmesg | grep -P audio\|snd\|sound\|audio
[    1.802366]   No soundcards found.
[   14.314004] mt6358-sound mt6358-sound: mt6358_platform_driver_probe(), dev name mt6358-sound
[   14.568250] mt8183-audio 11220000.audio-controller:mt8183-afe-pcm: No cache defaults, reading back from HW
[   14.570013] mt8183-audio 11220000.audio-controller:mt8183-afe-pcm: mtk_afe_combine_sub_dai(), num of dai 20
[   23.185240] debugfs: Directory '11220000.audio-controller:mt8183-afe-pcm' with parent 'mt8183_da7219_rt1015p' already present!
[   23.202722] mt8183_da7219 mt8183-sound: ASoC: no sink widget found for I2S Playback
[   23.202746] mt8183_da7219 mt8183-sound: ASoC: Failed to add route TDM_OUT_PINCTRL -> direct -> I2S Playback
[   23.202752] mt8183_da7219 mt8183-sound: snd_soc_bind_card: snd_soc_dapm_add_routes failed: -19
```
so the elan trackpad has its own problem, but that isnt related to the sound
but i have hope, because lsmod show following:
```
linux@changeme:~$ lsmod
Module                  Size  Used by
snd_seq                73728  0
snd_seq_device         20480  1 snd_seq
mt8183_da7219_max98357    28672  0
ath10k_sdio            32768  0
ath10k_core           479232  1 ath10k_sdio
mtk_vcodec_dec         73728  0
mtk_vcodec_enc         45056  0
v4l2_h264              16384  1 mtk_vcodec_dec
mtk_vcodec_common      20480  2 mtk_vcodec_enc,mtk_vcodec_dec
mtk_vpu                20480  1 mtk_vcodec_common
mtk_jpeg               40960  0
uvcvideo              106496  0
ath                    40960  1 ath10k_core
videobuf2_dma_contig    24576  3 mtk_jpeg,mtk_vcodec_enc,mtk_vcodec_dec
mac80211              516096  1 ath10k_core
v4l2_mem2mem           40960  3 mtk_jpeg,mtk_vcodec_enc,mtk_vcodec_dec
videobuf2_vmalloc      20480  1 uvcvideo
videobuf2_memops       20480  2 videobuf2_vmalloc,videobuf2_dma_contig
videobuf2_v4l2         32768  5 mtk_jpeg,mtk_vcodec_enc,uvcvideo,mtk_vcodec_dec,v4l2_mem2mem
libarc4                16384  1 mac80211
videobuf2_common       57344  9 mtk_jpeg,mtk_vcodec_enc,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,uvcvideo,mtk_vcodec_dec,v4l2_mem2mem,videobuf2_memops
hci_uart               77824  0
videodev              237568  7 mtk_jpeg,mtk_vcodec_enc,videobuf2_v4l2,uvcvideo,videobuf2_common,mtk_vcodec_dec,v4l2_mem2mem
cfg80211              397312  3 ath,mac80211,ath10k_core
i2c_hid_of             16384  0
btqca                  20480  1 hci_uart
btbcm                  28672  1 hci_uart
snd_soc_mt8183_afe    118784  0
cros_ec_typec          20480  0
cros_ec_chardev        16384  0
sbs_battery            20480  0
cros_usbpd_charger     16384  0
cros_usbpd_logger      16384  0
mc                     53248  6 videodev,videobuf2_v4l2,uvcvideo,videobuf2_common,mtk_vcodec_dec,v4l2_mem2mem
typec                  57344  1 cros_ec_typec
snd_soc_da7219         65536  1 mt8183_da7219_max98357
bluetooth             503808  10 btqca,hci_uart,btbcm
snd_soc_mtk_common     24576  1 snd_soc_mt8183_afe
i2c_hid                28672  1 i2c_hid_of
elants_i2c             24576  0
elan_i2c               32768  0
ecdh_generic           16384  1 bluetooth
ecc                    36864  1 ecdh_generic
mtk_scp                20480  1 mtk_vcodec_common
rfkill                 36864  7 bluetooth,cfg80211
mtk_rpmsg              20480  1 mtk_scp
coreboot_table         16384  0
mt6577_auxadc          20480  0
mtk_scp_ipi            16384  2 mtk_vcodec_common,mtk_scp
snd_soc_mt6358         57344  0
rtc_mt6397             16384  1
snd_soc_bt_sco         16384  0
mtk_wdt                20480  1
snd_soc_rt1015p        16384  0
fuse                  131072  3
ip_tables              32768  0
x_tables               45056  1 ip_tables
ipv6                  483328  0
```

milan475, 2022-06-01:
[config.txt](https://github.com/hexdump0815/imagebuilder/files/8816229/config.txt)

This is the dump I made earlier.

BenjaminWegener, 2022-06-02:
some news...
the error message with `[   23.202746] mt8183_da7219 mt8183-sound: ASoC: Failed to add route TDM_OUT_PINCTRL -> direct -> I2S Playback` led me to hack the `/compile/source/linux-stable-mt8/sound/soc/mediatek/mt8183/mt8183-da7219-max98357.c` just to see if the error message would change.
```
static const struct snd_soc_dapm_route mt8183_da7219_max98357_dapm_routes[] = {
	{"Speakers", NULL, "Speaker"},
//	{"I2S Playback", NULL, "TDM_OUT_PINCTRL"},
};
```
so the good news, i get 
```
linux@changeme:~$ lsmod
Module                  Size  Used by
bridge                253952  0
mt8183_da7219_max98357    28672  0
stp                    20480  1 bridge
llc                    20480  2 bridge,stp
ath10k_sdio            32768  0
ath10k_core           479232  1 ath10k_sdio
mtk_vcodec_dec         73728  0
ath                    40960  1 ath10k_core
v4l2_h264              16384  1 mtk_vcodec_dec
mtk_vcodec_enc         45056  0
mtk_vcodec_common      20480  2 mtk_vcodec_enc,mtk_vcodec_dec
mac80211              516096  1 ath10k_core
mtk_jpeg               40960  0
mtk_vpu                20480  1 mtk_vcodec_common
uvcvideo              106496  0
videobuf2_dma_contig    24576  3 mtk_jpeg,mtk_vcodec_enc,mtk_vcodec_dec
libarc4                16384  1 mac80211
videobuf2_vmalloc      20480  1 uvcvideo
v4l2_mem2mem           40960  3 mtk_jpeg,mtk_vcodec_enc,mtk_vcodec_dec
videobuf2_memops       20480  2 videobuf2_vmalloc,videobuf2_dma_contig
videobuf2_v4l2         32768  5 mtk_jpeg,mtk_vcodec_enc,uvcvideo,mtk_vcodec_dec,v4l2_mem2mem
videobuf2_common       61440  9 mtk_jpeg,mtk_vcodec_enc,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,uvcvideo,mtk_vcodec_dec,v4l2_mem2mem,videobuf2_memops
cfg80211              397312  3 ath,mac80211,ath10k_core
videodev              237568  7 mtk_jpeg,mtk_vcodec_enc,videobuf2_v4l2,uvcvideo,videobuf2_common,mtk_vcodec_dec,v4l2_mem2mem
mc                     53248  6 videodev,videobuf2_v4l2,uvcvideo,videobuf2_common,mtk_vcodec_dec,v4l2_mem2mem
cros_ec_typec          20480  0
hci_uart               57344  0
cros_ec_lid_angle      16384  0
typec                  57344  1 cros_ec_typec
snd_soc_mt8183_afe    118784  2
btqca                  20480  1 hci_uart
cros_ec_sensors_core    20480  1 cros_ec_lid_angle
btrtl                  24576  1 hci_uart
snd_soc_da7219         65536  2 mt8183_da7219_max98357
snd_soc_mtk_common     24576  1 snd_soc_mt8183_afe
cros_usbpd_charger     16384  0
cros_usbpd_logger      16384  0
cros_ec_chardev        16384  0
bluetooth             503808  10 btrtl,btqca,hci_uart
joydev                 28672  0
elants_i2c             24576  0
mtk_scp                20480  1 mtk_vcodec_common
ecdh_generic           16384  1 bluetooth
ecc                    36864  1 ecdh_generic
mtk_rpmsg              20480  1 mtk_scp
rfkill                 36864  7 bluetooth,cfg80211
rpmsg_core             20480  1 mtk_rpmsg
mtk_scp_ipi            16384  2 mtk_vcodec_common,mtk_scp
mtk_wdt                20480  1
mt8183_cci_devfreq     16384  0
coreboot_table         16384  0
snd_soc_rt1015p        16384  1
snd_soc_bt_sco         16384  1
fuse                  131072  3
ip_tables              32768  0
x_tables               45056  1 ip_tables
ipv6                  483328  1 bridge

linux@changeme:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Dummy [Dummy], device 0: Dummy PCM [Dummy PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: Loopback [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: Loopback [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 3: mt8183da7219rt1 [mt8183_da7219_rt1015p], device 0: Playback_1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: mt8183da7219rt1 [mt8183_da7219_rt1015p], device 1: Playback_2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: mt8183da7219rt1 [mt8183_da7219_rt1015p], device 2: Playback_3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: mt8183da7219rt1 [mt8183_da7219_rt1015p], device 7: Playback_HDMI (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```
no sound currently in the mixer, haven't had time to look at the ucm.

BenjaminWegener, 2022-06-02:
https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/pull/3

BenjaminWegener, 2022-06-02:
```
linux@changeme:~$ cat /proc/asound/cards
 0 [Dummy          ]: Dummy - Dummy
                      Dummy 1
 1 [Loopback       ]: Loopback - Loopback
                      Loopback 1
 2 [VirMIDI        ]: VirMIDI - VirMIDI
                      Virtual MIDI Card 1
 3 [mt8183da7219rt1]: mt8183_da7219_r - mt8183_da7219_rt1015p
                      mt8183_da7219_rt1015p

linux@changeme:~$ amixer 
Simple mixer control 'Master',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 65536
  Mono:
  Front Left: Playback 84026 [128%] [on]
  Front Right: Playback 84026 [128%] [on]
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch cswitch-joined
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 65536
  Front Left: Capture 0 [0%] [on]
  Front Right: Capture 0 [0%] [on]

```

hexdump0815, 2022-06-02:
ok - the naming is as i expected it so my draft ucm file from https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_kukui/extra-files/usr/share/alsa/ucm2/mt8183_da7219_m at least is properly named ... can you please paste the "amixer -c 3" output?

BenjaminWegener, 2022-06-02:
```
linux@changeme:~$ amixer -c 3
amixer: Mixer hw:3 load error: Invalid argument
```

hexdump0815, 2022-06-02:
oh - my naming is wrong: its "mt8183_da7219_r" - so "_r" and not "_m" ... which kernel (cromeos vs mainline) and which audio module source (chromeos vs mainline) is this now?

hexdump0815, 2022-06-02:
oh - i see - from the pull request its mainline/mainline which is good

hexdump0815, 2022-06-02:
not sure if it helps: found this - https://patchwork.kernel.org/project/linux-mediatek/patch/20220316014059.19292-1-linmq006@gmail.com/ ... mybe some of the corresponding mt8186 patches might give some hint: https://patchwork.kernel.org/project/linux-mediatek/list/?series=&submitter=&state=*&q=da7219&archive=both&delegate=

BenjaminWegener, 2022-06-05:
sound is working, haven't tested mic etc.
commits needed i will create tomorrow
to make sound work, run this script:
```
#!/bin/bash

if [[ $EUID -ne 0 ]]; then
    echo "$0 is not running as root. try using sudo."
    exit 2
fi

echo "this script builds and installs the kernel sound module mt8183-da7219-max98357a on the FENNEL14 sku0 chromebook"
echo "be sure to use the exact image on an usbdrive/sdcard mounted as /dev/sda!"
echo "image can be downloaded at https://github.com/hexdump0815/imagebuilder/releases/tag/220528-01"
echo "this can take over an hour to finish!"

echo "
extending the rootfs..."
/scripts/extend-rootfs.sh

echo "
uninstalling old module..."
modprobe -rf mt8183_mt6358_ts3a227_max98357
modprobe -rf snd-soc-ts3a227e
rm -rf /lib/modules/5.18.0-stb-mt8+/kernel/sound/soc/mediatek/mt8183/mt8183-mt6358-ts3a227-max98357.ko
rm -rf /lib/modules/5.18.0-stb-mt8+/kernel/sound/soc/mediatek/mt8183/mt8183-da7219-max98357.ko
sed -i -e 's/modprobe mt8183-mt6358-ts3a227-max98357/modprobe mt8183-da7219-max98357/g' /etc/rc.local

echo "
installing build dependencies..."
apt update
apt install -y sed libncurses-dev dbus-x11 liblz4-tool vboot-kernel-utils build-essential flex bison gettext libiberty-dev libpci-dev libelf-dev libssl-dev libslang2-dev libperl-dev python-all-dev liblzma-dev libzstd-dev libcap-dev libnuma-dev libbabeltrace-dev systemtap-sdt-dev libdw-dev apt-utils binutils-dev git wget libunwind-dev

echo "
downloading kernel sources to build new module..."
cd /tmp
mkdir sound-fix
cd sound-fix
wget https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.18.tar.xz
tar -xvf linux-5.18.tar.xz
cd linux-5.18

echo "
patching mt8183-da7219-max98357a.c..."
sed -i -e 's/{"I2S Playback", NULL, "TDM_OUT_PINCTRL"},/\/\/{"I2S Playback", NULL, "TDM_OUT_PINCTRL"},/g' sound/soc/mediatek/mt8183/mt8183-da7219-max98357.c

echo "
building module..."
make clean -j8
make kernelrelease -j8
make oldconfig -j8
sed -i -e 's/CONFIG_LOCALVERSION="-stb-mt8"/CONFIG_LOCALVERSION="-stb-mt8+"/g' .config
make prepare -j8
make modules_prepare -j8
make Image dtbs modules -j8

echo "
installing module..."
cp sound/soc/mediatek/mt8183/mt8183-da7219-max98357.ko /lib/modules/5.18.0-stb-mt8+/kernel/sound/soc/mediatek/mt8183/
depmod

echo "
creating alsa profile..."
mkdir /usr/share/alsa/ucm2/mt8183_da7219_r/
echo "
Syntax 2

Comment "Kukui internal card"

SectionUseCase."HiFi" {
	File "HiFi.conf"
	Comment "Default"
}
" > /usr/share/alsa/ucm2/mt8183_da7219_r/mt8183_da7219_rt1015p.conf
echo "
SectionVerb {
	EnableSequence [
		cdev "hw:mt8183da7219rt1"
		cset "name='Headphone Volume' 0"
#		cset "name='UL3_CH1 ADDA_UL_CH1' 0"
#		cset "name='UL3_CH2 ADDA_UL_CH2' 0"
		cset "name='I2S3_CH1 DL1_CH1' 1"
		cset "name='I2S3_CH2 DL1_CH2' 1"
		cset "name='MTKAIF_DMIC' 1"
		cset "name='Mt6358 Mic Type Mux' 2"
		cset "name='UL3_CH1 ADDA_UL_CH1' 1"
		cset "name='UL3_CH2 ADDA_UL_CH2' 1"
	]
	DisableSequence [
	]
}
SectionDevice."Headphone" {
	Value {
		PlaybackPCM "hw:mt8183da7219rt1,0"
		JackDev "mt8183_da7219_rt1015p Headset Jack"
		JackSwitch "2"
	}

	ConflictingDevice [
		"Speaker"
	]

	EnableSequence [
		cdev "hw:mt8183da7219rt1"
		cset "name='I2S3_CH1 DL1_CH1' 1"
		cset "name='I2S3_CH2 DL1_CH2' 1"
	]
	DisableSequence [
		cdev "hw:mt8183da7219rt1"
		cset "name='I2S3_CH1 DL1_CH1' 0"
		cset "name='I2S3_CH2 DL1_CH2' 0"
	]
}
SectionDevice."Speaker" {
	Value {
		PlaybackPCM "hw:mt8183da7219rt1,0"
		DspName "speaker_eq"
	}

	ConflictingDevice [
		"Headphone"
	]

	EnableSequence [
		cdev "hw:mt8183da7219rt1"
		cset "name='I2S3_CH1 DL1_CH1' 1"
		cset "name='I2S3_CH2 DL1_CH2' 1"
	]
	DisableSequence [
		cdev "hw:mt8183da7219rt1"
		cset "name='I2S3_CH1 DL1_CH1' 0"
		cset "name='I2S3_CH2 DL1_CH2' 0"
	]
}

SectionDevice."Internal Mic" {
	Value {
		CapturePCM "hw:mt8183da7219rt1,5"
		CaptureChannelMap "0 1 -1 -1 -1 -1 -1 -1 -1 -1 -1"
		IntrinsicSensitivity "-600"
	}
	EnableSequence [
		cdev "hw:mt8183da7219rt1"
		cset "name='MTKAIF_DMIC' 1"
		cset "name='Mt6358 Mic Type Mux' 2"
		cset "name='UL3_CH1 ADDA_UL_CH1' 1"
		cset "name='UL3_CH2 ADDA_UL_CH2' 1"
	]
	DisableSequence [
		cdev "hw:mt8183da7219rt1"
		cset "name='MTKAIF_DMIC' 0"
		cset "name='Mt6358 Mic Type Mux' 0"
		cset "name='UL3_CH1 ADDA_UL_CH1' 0"
		cset "name='UL3_CH2 ADDA_UL_CH2' 0"
	]
}
" > /usr/share/alsa/ucm2/mt8183_da7219_r/HiFi.conf
echo "
alsaucm -n -b - <<EOM
open mt8183_da7219_r
reset
set _verb HiFi
set _enadev Speaker
EOM
" > /etc/rc.local

echo "
Do you wish to reboot?"
select yn in "Yes" "No"; do
    case $yn in
        Yes ) 
            reboot;
            break;;
        No ) exit;;
    esac
done

```
@milan475 could you test it please?

hexdump0815, 2022-06-05:
this is very nice - congratulations! ... i was planning to build new kukui images with v5.18.1 and mesa 22.1.1 these days - i think i'll wait until your patches and ucm files are in as well so that those images will support fennel14 out of the box then too :) ... but anyway - take the time you need for preparing the patches - there is no hurry

hexdump0815, 2022-06-05:
@BenjaminWegener - i have pulled in your audio fix into the mt81xx kernel tree as it seems to be the only fix required as it seems to work for you now ... please let me know if i should simply copy and paste the ucm files from your comment above (and credit you for it of course) or if you prefer to create a pull request for it

hexdump0815, 2022-06-05:
i built a new kernel including this fix in case anyone wants to try it: https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/tag/5.18.1-stb-mt8%2B

the ucm steps from above are still required

BenjaminWegener, 2022-06-06:
@hexdump0815 thanks very much!
here is an untested script to use for the ucm and kernel updating. will test it on a fresh install asap.
```
#!/bin/bash

if [[ $EUID -ne 0 ]]; then
    echo "$0 is not running as root. try using sudo."
    exit 2
fi

echo "this script updates the kernel and sound module mt8183-da7219-max98357a on the FENNEL14 sku0 chromebook."
echo "be sure to use the exact image on an usbdrive/sdcard mounted as /dev/sda!"
echo "image can be downloaded at https://github.com/hexdump0815/imagebuilder/releases/tag/220528-01"

echo "
extending the rootfs..."
/scripts/extend-rootfs.sh

echo "
uninstalling old module..."
modprobe -rf mt8183_mt6358_ts3a227_max98357
modprobe -rf snd-soc-ts3a227e
rm -rf /lib/modules/5.18.0-stb-mt8+/kernel/sound/soc/mediatek/mt8183/mt8183-mt6358-ts3a227-max98357.ko

echo "
activating new module..."
sed -i -e 's/modprobe mt8183-mt6358-ts3a227-max98357/modprobe mt8183-da7219-max98357/g' /etc/rc.local
sed -i -e 's/exit 0/ /g' /etc/rc.local

echo "
creating alsa profile..."
mkdir /usr/share/alsa/ucm2/mt8183_da7219_r/
echo "
Syntax 2

Comment \"Kukui internal card\"

SectionUseCase.\"HiFi\" {
	File \"HiFi.conf\"
	Comment \"Default\"
}
" > /usr/share/alsa/ucm2/mt8183_da7219_r/mt8183_da7219_rt1015p.conf
echo "
SectionVerb {
	EnableSequence [
		cdev \"hw:mt8183da7219rt1\"
		cset \"name='Headphone Volume' 0\"
#		cset \"name='UL3_CH1 ADDA_UL_CH1' 0\"
#		cset \"name='UL3_CH2 ADDA_UL_CH2' 0\"
		cset \"name='I2S3_CH1 DL1_CH1' 1\"
		cset \"name='I2S3_CH2 DL1_CH2' 1\"
		cset \"name='MTKAIF_DMIC' 1\"
		cset \"name='Mt6358 Mic Type Mux' 2\"
		cset \"name='UL3_CH1 ADDA_UL_CH1' 1\"
		cset \"name='UL3_CH2 ADDA_UL_CH2' 1\"
	]
	DisableSequence [
	]
}
SectionDevice.\"Headphone\" {
	Value {
		PlaybackPCM \"hw:mt8183da7219rt1,0\"
		JackDev \"mt8183_da7219_rt1015p Headset Jack\"
		JackSwitch \"2\"
	}

	ConflictingDevice [
		\"Speaker\"
	]

	EnableSequence [
		cdev \"hw:mt8183da7219rt1\"
		cset \"name='I2S3_CH1 DL1_CH1' 1\"
		cset \"name='I2S3_CH2 DL1_CH2' 1\"
	]
	DisableSequence [
		cdev \"hw:mt8183da7219rt1\"
		cset \"name='I2S3_CH1 DL1_CH1' 0\"
		cset \"name='I2S3_CH2 DL1_CH2' 0\"
	]
}
SectionDevice.\"Speaker\" {
	Value {
		PlaybackPCM \"hw:mt8183da7219rt1,0\"
		DspName \"speaker_eq\"
	}

	ConflictingDevice [
		\"Headphone\"
	]

	EnableSequence [
		cdev \"hw:mt8183da7219rt1\"
		cset \"name='I2S3_CH1 DL1_CH1' 1\"
		cset \"name='I2S3_CH2 DL1_CH2' 1\"
	]
	DisableSequence [
		cdev \"hw:mt8183da7219rt1\"
		cset \"name='I2S3_CH1 DL1_CH1' 0\"
		cset \"name='I2S3_CH2 DL1_CH2' 0\"
	]
}

SectionDevice.\"Internal Mic\" {
	Value {
		CapturePCM \"hw:mt8183da7219rt1,5\"
		CaptureChannelMap \"0 1 -1 -1 -1 -1 -1 -1 -1 -1 -1\"
		IntrinsicSensitivity \"-600\"
	}
	EnableSequence [
		cdev \"hw:mt8183da7219rt1\"
		cset \"name='MTKAIF_DMIC' 1\"
		cset \"name='Mt6358 Mic Type Mux' 2\"
		cset \"name='UL3_CH1 ADDA_UL_CH1' 1\"
		cset \"name='UL3_CH2 ADDA_UL_CH2' 1\"
	]
	DisableSequence [
		cdev \"hw:mt8183da7219rt1\"
		cset \"name='MTKAIF_DMIC' 0\"
		cset \"name='Mt6358 Mic Type Mux' 0\"
		cset \"name='UL3_CH1 ADDA_UL_CH1' 0\"
		cset \"name='UL3_CH2 ADDA_UL_CH2' 0\"
	]
}
" > /usr/share/alsa/ucm2/mt8183_da7219_r/HiFi.conf

echo "
alsaucm -n -b - <<EOM
open mt8183_da7219_r
reset
set _verb HiFi
set _enadev Speaker
EOM

exit 0
" >> /etc/rc.local

echo "
downloading the kernel...
"
apt update
apt install wget -y
cd /tmp
mkdir mnt
wget https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/download/5.18.1-stb-mt8%2B/5.18.1-stb-mt8+.tar.gz
echo "Do you wish to install the kernel? YOUR INSTALLATION HAS TO BE ON /dev/sda!!! THIS CAN NOT BE UNDONE AND MAY DAMAGE YOUR DEVICE!!!"
select yn in "Yes" "No"; do
    case $yn in
        Yes ) 
            mount /dev/sda4 /tmp/mnt;
            mount /dev/sda3 /tmp/mnt/boot;
            cd mnt;
            tar xzf 5.18.1-stb-mt8+.tar.gz;
            dd if=boot/vmlinux.kpart-5.18.1-stb-mt8+ of=/dev/sda1;
            umount /tmp/mnt/boot /tmp/mnt;
            echo "kernel successfully written...";
            break;;
        No ) exit;;
    esac
done

echo "Do you wish to reboot?"
select yn in "Yes" "No"; do
    case $yn in
        Yes ) 
            reboot;
            break;;
        No ) exit;;
    esac
done
```

hexdump0815, 2022-06-06:
i have added the ucm files and adjusted the rc.local in the imagebuilder (https://github.com/hexdump0815/imagebuilder/commit/89676f88f4385cc06856c2e5df460855dc877989) and will build a new image based on this one now

hexdump0815, 2022-06-06:
new bullseye image is up now: https://github.com/hexdump0815/imagebuilder/releases/tag/220606-01 - it would be nice if you could test boot it to see if audio now works out of the box on fennel14 too with it ... as soon as this is checked i'll build the jammy image too and upload it as well

BenjaminWegener, 2022-06-06:
Will do

BenjaminWegener, 2022-06-06:
@hexdump0815 i am happy to say, everything works now out of the box, including bluetooth and sound! thanks...
i think, this and the bluetooth issue can be closed

hexdump0815, 2022-06-06:
@BenjaminWegener - i'm happy to hear this - thanks a lot for making this work and for the testing ... uploading the jammy image right now ... so welcome to fennel14 to the fully supported kukui devices club :)

btw. you mentioned in your status github issue that hardware video decoding is working - i think it is still done in software and the cpu is fast enough to easily do it ... there is some hw video decoder support for the mediatek socs in the tree i think, but things are not wired up properly yet (and not sure if and when they will) so that normal apps like firefox etc. can make use of it as far as i know

hexdump0815, 2022-06-06:
new images with working fennel14 audio are out, so lets close this issue ...

closed 2022-06-06 by hexdump0815

BenjaminWegener, 2022-06-06:
> @BenjaminWegener - i'm happy to hear this - thanks a lot for making this work and for the testing ... uploading the jammy image right now ... so welcome to fennel14 to the fully supported kukui [...]

@hexdump0815 maybe, i dont know. i have a few things i am free to test now, learned a lot about kernel and stuff the past days. first will be the open source arm mali driver to test, i know, its not free as in GNU but i want to know exactly what the mali 72 is capable of...