velvet-os/imagebuilder #55: problem: chromebook_kukui: bluetooth no longer working in v5.18

hexdump0815, 2022-05-31:
bluetooth was working well in v5.15 but seems to be broken in v5.18

BenjaminWegener, 2022-06-01:
mazbe it has something to do with https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_kukui/extra-files/lib/firmware/qca ?

aneeshlingala, 2022-06-01:
Sadly, this is broken for all kukui devices.

aneeshlingala, 2022-06-01:
Also on kappa, Blueman doesn't detect any adapter, must be a kernel module or related issue.

hexdump0815, 2022-06-01:
it is not related to the firmware i guess as it is working fine with up to v5.17 ... as v5.17 is working and v5.18 is broken it should be possilbe to git bisect this problem between those to kernel versions - just a matter of effort :) ... maybe i'll give it a try over the weekend

aneeshlingala, 2022-06-01:
sure

aneeshlingala, 2022-06-01:
is there newer bluetooth firmware from 5.17 for bluetooth? If so, try using that firmware in the 5.18 kernel, this issue might be related to this: https://www.spinics.net/lists/stable/msg491263.html

hexdump0815, 2022-06-01:
that would be https://github.com/gregkh/linux/commit/0ea9fd001a14ebc294f112b0361a4e601551d508 and it looks like it was fixed shortly after with https://github.com/gregkh/linux/commit/0ea53674d07fb6db2dd7a7ec2fdc85a12eb246c2 ... so i guess this was a different problem

aneeshlingala, 2022-06-01:
If you post the updated image, can we update from a older version?

hexdump0815, 2022-06-01:
@aneeshlingala - there is currently no automated way to update the kernel - best is to just compile a new one or untar a newer build if one is available and install it

hexdump0815, 2022-06-01:
problem kind of solved: drivers/tty/serial/8250/8250_mtk.c needs to be reverted to the v5.17 state to make bluetooth work for kukui devices on v5.18 ... i'll add a proper patch to the repo and build a new kernel during the next days

aneeshlingala, 2022-06-02:
I think you can use dd to get two images from the two kernel partitions while booting in the live USB/SD card and flash those two images to the kernel partitons on eMMC and chroo into the eMMC install and run` sudo update-initramfs -u`

BenjaminWegener, 2022-06-02:
do we know which of the [patches](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/drivers/tty/serial/8250/8250_mtk.c?h=v5.18.1) is responsible?

hexdump0815, 2022-06-03:
i have added the patch now: https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/blob/master/misc.cbm/patches/5.18.0/mt8183-fix-bluetooth.patch ... i think it were 3 commits for the mtk serial driver - maybe some refactoring which broke things in the end - you can see the commits via a git log of the patched file

BenjaminWegener, 2022-06-05:
@hexdump0815 thank you very much! can confirm, its working now using your patch from the repo

hexdump0815, 2022-06-06:
new images with fixed bluetooth are out, so lets close this issue ...

closed 2022-06-06 by hexdump0815