velvet-os/imagebuilder #57: discussion: features to implement

BenjaminWegener, 2022-06-07:
i was thinking to test some features not included as of now and maybe create tutorials or scripts to implement them.
may wishes are:

- squashfs for snapcraft
- mali gpu driver for opencl / vulkan
- widvine drm for netflix/disney etc.
- CONFIG_ASHMEM, CONFIG_ANDROID and CONFIG_JOYSTICK for anbox - to play android games maybe

hexdump0815, 2022-06-07:
thanks for your ideas - maybe some comments from my side about them: in general i'm trying to keep the systems rather basic and simple as trying to make it perfect for everyone is impossible anyway ... i tried to add features which (at least in my opinion) really added some benefit for nearly everyone like zstd compressed zswap by default (as it helps with low memory systems and minimizes disk writes while still offering more virtual memory than for instance zram), btrfs with zstd compression (making even a 16gb emmc really useable for a full system with some space left and on top having internal checksumming, i.e. you get notified when you storage starts to die slowly and it will not do so silently) or the tiny load average bar in the top right of the default xfce menubar (it helped me so many times to have an idea if the system is overloded or still has reserves left) ... besides that i would like to try to encourage people to do things like building a kernel, mesa etc. themself and try to provide at least some sketches (with still a lot of room for improvement :) ) about how to do that ... enough foreword - to your points:

- squashfs for snap support - i already tried to add the required options - in case there is something missing i'm more than willing to include any missing minimal set of required options (to https://github.com/hexdump0815/kernel-config-options) to make that work - i think snap and flatpack should be supported out of the box (just as docker already is)
- mali gpu drivers - they are supported for systems where there is no lima/panfrost available (snow and peach chromebooks for instance) and i would prefer to avoid adding more of them than needed (i suffered quite a bit with them, their quality and the required hacks in the past :) ) - vulkan and opencl should come in the near future for panfrost and freedreno (which already has some support) at least - but putting some notes on how to get them running or the kernel modules compiled would be completely fine (some old example from pre panfrost times: https://github.com/hexdump0815/linux-mainline-and-mali-rockchip-rk33xx-kernel/blob/master/readme.rkc-mali-rk3399)
- widivine could be easily handled via some scripts to install/uninstall what is required to /scripts - downside is that (as far as i know and until now) its limited to 32bit arm and requires some chroot hacks etc.
- the anbox options should be fine if they do not break anything else or compatibility and if they do not compromise security in any way - otherwise some notes on how to get all this work would be preferred (similar approaches might be used for box86, box64 etc.)

so basically there are currently three ways to add things:

- adding them to the base images: only well tested and reliable things which are useful for a lot of users and which will be easy to maintain over time on a variety of systems should go here
- adding scripts for them to /scripts to easily add and cleanly remove them if that is easily possible
- adding notes to the info subdir giving hints or tutorials on how to do more complex or more special things - preferrably in better quality than what i have dumped there so far maybe :)

best wishes - hexdump

BenjaminWegener, 2022-06-08:
thanks or your much appreciated answer, as always!

- testing the gpu with mali driver, reporting back. will share a tutorial, as panfrost is the way to go in the future.
- as for the option with squashfs and android i will test it myself and open a pull request once its ready
- with widevine you are right, i will look into it, but isnt my main concern.

BenjaminWegener, 2022-06-09:
@hexdump0815 just wanted to give some status report:
-anbox needs some kernel patches for dropped android support in 5.18, will see if it works
-mali gpu needs some non free driver, still fighting with some configs to install it, maybe will never succeed
-squashfs needs some extra config options to work, succeeded, can install chromium for example

hexdump0815, 2022-06-09:
@BenjaminWegener - thanks for the update:
- anbox looks more some tutorial/notes style topic then i think
- mali gpu will be hard and please keep in mind that you will not be able to use opengl directly in regular xorg with it as it cannot handle the propietary driver - my recommendation would be to not invest too much time in it as the chance to fail is very high
- squashfs is very welcome to get into the base images, i.e. kernel-config-options and the working chromeium snap installation sounds promising
best wishes - hexdump

hexdump0815, 2022-06-09:
maybe lets keep the issue open as long as it is being worked on ...

BenjaminWegener, 2022-06-09:
https://github.com/hexdump0815/kernel-config-options/pull/1

hexdump0815, 2022-06-10:
@BenjaminWegener - thanks a lot -i pulled it in - it will take a while until it appears in images ... i plan to build new kernels and some new images maybe later in v5.18 or when v5.19 is out ... but all those will have the options enabled then

BenjaminWegener, 2022-06-11:
@hexdump0815 thanks alot, much appreciated.
-i dropped the mali driver, its not worth it, glmark2 is fast enough, as for machine learning support, arm relased the tflite package https://github.com/ARM-software/armnn#quick-start-guides, should run on panfrost opengles 3.1 (opencl alternative), ill write it up once its tested
-anbox is incompatible with arm64 it seems, dropped that too
- lxc is an container environment, needs three kkernel configs added, ill try to boot a raspi image, maybe ill get chromium with widevine to work

hexdump0815, 2022-06-11:
@BenjaminWegener - sounds good

not sure if you have seen it: there are at least some code fragments available for the neuronal part of the soc - not sure if its possible to make anything useful with it, but it might be worth a look: https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_kukui/todo.txt#L1-L8

hexdump0815, 2022-06-11:
just discovered: there are patches for apu support for mt8183 as well: https://patchwork.kernel.org/project/linux-mediatek/cover/20220304161514.994128-1-abailon@baylibre.com/ ... but i guess that is only the kernel driver and there is no usermode code to interface with it available ...

linsyking, 2024-07-16:
> thanks for your ideas - maybe some comments from my side about them: in general i'm trying to keep the systems rather basic and simple as trying to make it perfect for everyone is impossible anyway ... i tried to add features which (at least in my opinion) really added some benefit for nearly everyone like zstd compressed zswap by default (as it helps with low memory systems and minimizes disk writes while still offering more virtual memory than for instance zram), btrfs with zstd compression (making even a 16gb emmc really useable for a full system with some space left and on top having internal checksumming, i.e. you get notified when your storage starts to die slowly and it will not do so silently) or the tiny load average bar in the top right of the default xfce menubar (it helped me so many times to have an idea if the system is overloded or still has reserves left) ... besides that i would like to try to encourage people to do things like building a kernel, mesa etc. themself and try to provide at least some sketches (with still a lot of room for improvement :) ) about how to do that ... enough foreword - to your points:
> 
> * squashfs for snap support - i already tried to add the required options - in case there is something missing i'm more than willing to include any missing minimal set of required options (to https://github.com/hexdump0815/kernel-config-options) to make that work - i think snap and flatpack should be supported out of the box (just as docker already is and kvm should be in this category too where technically possible)
> * mali gpu drivers - they are supported for systems where there is no lima/panfrost available (snow and peach chromebooks for instance) and i would prefer to avoid adding more of them than needed (i suffered quite a bit with them, their quality and the required hacks in the past :) ) - vulkan and opencl should come in the near future for panfrost and freedreno (which already has some support) at least - but putting some notes on how to get them running or the kernel modules compiled would be completely fine (some old example from pre panfrost times: https://github.com/hexdump0815/linux-mainline-and-mali-rockchip-rk33xx-kernel/blob/master/readme.rkc-mali-rk3399)
> * widivine could be easily handled via some scripts to install/uninstall what is required to /scripts - downside is that (as far as i know and until now) its limited to 32bit arm and requires some chroot hacks etc. on 64bit arm
> * the anbox options should be fine if they do not break anything else or compatibility and if they do not compromise security in any way - otherwise some notes on how to get all this work would be preferred (similar approaches might be used for box86, box64 etc.)
> 
> so basically there are currently three ways to add things:
> 
> * adding them to the base images: only well tested and reliable things which are useful for a lot of users and which will be easy to maintain over time on a variety of systems should go here
> * adding scripts for them to /scripts to easily add and cleanly remove them if that is easily possible
> * adding notes to the info subdir giving hints or tutorials on how to do more complex or more special things - preferrably in better quality than what i have dumped there so far maybe :)
> 
> best wishes - hexdump

I think zram is helpful (with writeback), are there any reasons you disable zram support in kernel config?

hexdump0815, 2024-07-16:
@linsyking - in the images zswap is used instead, which is in my opinion the better option ... but as long as the kernel options do not conflict i would have no problem to enable the zram options as well