velvet-os/imagebuilder #58: system status: chromebook_kukui: fennel (lenovo chromebook flex 3 - mt8183 version)

hexdump0815, 2022-06-09:
## notes on fennel running 5.18.1:

here is list of working and non working features on the mt8183version of the lenovo chromebook flex 3 (fennel). i just had a short moment to do a test boot and the system tested was the sku6 version.

### working
- wifi
- bluetooth
- builtin display
- energy: battery status, dpms, suspend/resume
- usb a and c for basic usb functionality
- webcam
- gpu opengl: via mesa panfrost (opengl 3.1)

### untested
- hw video encode/decode: most probably not working yet

### broken
- sound: in theory it should be possible to make it work with the fennel14 settings as it seems to have the same audio setup, but the time was too short for testing
- external display: not working yet

### installation
coming soon

### problems
none so far

hexdump0815, 2022-06-21:
sorry - i'm mostly offline right now, but will be back in a few days

regarding audio, you might check if you have a sku6 version as well (cat /proc/device-tree/model) and if this is the case then maybe try to change the matching line in rc.local (https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_kukui/rc-local-additions.txt#L27) to your model string ... sku6 and fennel14 should have the exact same audio hardware and it should work, but one never knows ...

if you have another sku version or if it still does not work then this might give some ideas about how to debug it: https://github.com/hexdump0815/imagebuilder/issues/54

regarding the sleep issue i'll have to check later - it worked for me, but everything is possible to break again or for certain systems :)

Pawlicker, 2022-06-22:
Changing the model string did in fact fix it and audio now works. I'm still having sleep issues, when the laptop is in sleep mode for "too" long it seems to have issues with the SD card seemingly being unmounted. I'll see if I can determine why that's the case.

Pawlicker, 2022-06-22:
It's definitely connected as /dev/sda while the onboard storage is stored on mmcblk. I'm not sure it's a good idea to transfer the OS over to the MMC or how I would do it on these newer Chromebooks. It also seems to only happen after a few hours and not right away, and well Chromebooks do not run the OS from USB. 

Even changing that s2idle only keeps it at the deep version and worse; I get permission denied issuing the command itself as root.

hexdump0815, 2022-06-23:
this "permission denied" with s2idle sounds very strange - this should simply work ... what does "cat /sys/power/mem_sleep" say for you? are you using the latest images with the latest 5.18.1 kernel? i just tested it on my kappa system (running off emmc) and there suspend/resume works perfectly fine even for hours in deep mode

regarding installing onto emmc i would recommend to carefully read all of https://github.com/hexdump0815/linux-mainline-on-arm-chromebooks to get the basic ideas ... on the 64bit arm chromebooks which are all using the default and installed depthcharge bootloader the installation to emmc is quite riskless as chromeos can easily be reinstalled from a downloaded google recovery image for the corresponding device if desired - there is no rom flashing etc. required like for intel chromebooks with the efi firmware

i'm always using https://github.com/hexdump0815/imagebuilder/blob/main/info/generic/install-to-emmc-with-luks-full-disk-encryption.txt as base to copy and paste the steps whenever i setup to emmc from one of the images - the doc needs to be rewritten and is hard to read as there is a lot of other not chromebook relevant information mixed in (efi, u-boot), but it has proven to work well many times and will even give you an encrypted root disk, which i think is a very good idea for a portable device

good luck and best wishes - hexdump

Pawlicker, 2022-06-25:
I'll mess with trying an eMMC install some later, I've been somewhat busy the past week. I noticed in dmesg there was a second or two before the SD card was remounted when I left the laptop closed overnight for testing purposes with no apps open and it took a little longer for the login box to come up. This was marked with some write/read errors right before it's remounted.

I found another minor GPU bug with color corruption. I'll take a screenshot of it later and see if it shows up there. I might file a freedesktop bug and whatnot.

hexdump0815, 2022-07-03:
@Pawlicker - i have adjusted the audio module loading in /etc/rc.local now so that it should work for fennel (and a lot of other systems) out of the box for future images - https://github.com/hexdump0815/imagebuilder/commit/fbb7823bb1e637dabfb24d44467531ff1668d1f1

deiol, 2022-08-30:
I've got a Lenovo Flex 3 and from the ChromeOS shell, /proc/device-tree/model outputs "Google fennel sku7 board." Interestingly, when booting to VelvetOS, /proc/device-tree/model outputs "Google fennel14 sku0 board." I haven't gotten sound to work yet, dmesg shows `mt8183_da7219 mt8183-sound: ASoC: failed to instantiate card -6` and the output device is "Dummy Output." Modifying rc.local to load mt8183-mt6358-ts3a227-max98357 resolves the "failed to instantiate card" error in dmesg, but I still get Dummy Output. I will update if I make any progress.

hexdump0815, 2022-08-30:
@deiol - that wrong sku version is a bit confusing asthe kernel image includes all possible dtb files and the bootloader should pick the right one - maybe some naming got mixed up in the dtb files ...

regarding audio: the first point to achieve is to get some device showing up with "cat /proc/asound/cards" and it looks like you got to that point already - correct? if that is there you'll need to create a proper ucm file based on the chromeos ucm file - for that the following links should give some information on how to move on with this:
- https://github.com/hexdump0815/imagebuilder/blob/main/info/generic/alsa-ucm-pulseaudio-hacking.txt
- https://github.com/hexdump0815/imagebuilder/issues/61#issuecomment-1193085113
- https://github.com/hexdump0815/imagebuilder/issues/54

let me know in case more help is required ... in case you get it working a pull request with the working ucm files would be very nice, so that i can add it to the tree ( https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_kukui/extra-files/usr/share/alsa/ucm2 and https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_kukui/rc-local-additions.txt#L30-L74 )

good luck and best wishes - hexdump

deiol, 2022-09-08:
@hexdump0815 thank you for the informative post! I've solved the wrong sku version and submitted a pull request on your linux-mainline-mediatek-mt81xx-kernel repo.

After fixing that and updating rc.local to account for the new sku, I'm getting output when I cat /proc/asound/cards. Still no audio though, so I'll look into creating a ucm file. I'll submit a pull request if I get it working.

hexdump0815, 2022-09-09:
@deiol - cool - thanks a lot - i have put your pull request on my todo list - this is indeed a much better way to get the list of the dtb's and i'll also apply to it quite a few more kernels and in other places ... you already tested it exactly in the way you submitted it in the pr - right?

looking forward to you getting your sound working - good luck!

deiol, 2022-09-13:
Correct, I tested building the kernel exactly as I submitted in the PR (multiple times now) with success.

I got sound working, your UCM file `mt8183_mt6358_ts3a227_max98357.conf` worked however I had to rename it to `mt8183_mt6358_t.conf`, I've submitted a PR. I'll note that the default volume was VERY loud. I'm still looking into a few other oddities;

- deleting `mt8183_mt6358_t.conf` and sound continues work after restart of pulseaudio or computer (although in this case "Speaker" shows up as "Headphones")
- channels in alsamixer are a bit of a mess
- headphones work, but output doesn't switch from speakers automatically

In case folks come across this thread in the future, note the ALSA/UCM/PulseAudio hacking doc recently moved;
https://github.com/hexdump0815/imagebuilder/blob/main/doc/alsa-ucm-pulseaudio-hacking.txt

hexdump0815, 2022-09-13:
@deiol - thanks a lot for the update and the pr's - i hope to find the coming weekend or the days after a bit more time to have a closer look at this ... the headphone jack switching needs some extra effort i think if its possible with mainline and regular alsa/ucm right now at all, but at some point we'll get that going too ... thanks for the hint with the moved docs

thanks a lot and best wishes - hexdump

hexdump0815, 2022-09-16:
@deiol - fyi: your pr's are merged now - thanks a lot for your contribution ... i took out the (as i think unused) "-F" option from the mkimage cmd and replaced the shortened ucm conf file with a symlink to the long filename which should work as well i think ... i think it depends on the ucm version (depending on the distribution used) if the long filename can be used or a short one is required

best wishes - hexdump