velvet-os/imagebuilder #60: system status: chromebook_asurada

hexdump0815, 2022-07-10:
this issue is just a placeholder to potentially give feedback if my untested experimental images from https://github.com/hexdump0815/imagebuilder/releases/tag/220710-01 are working at all and in case they are working to note down what works and what not ...

caljorden, 2022-12-30:
@hexdump0815 

Just a note, I successfully ran this release on my Asurada/Sipherion.  It booted right up to a full GUI.  Here is what I can tell you from the past hour or so of using it:

Works:
- Software rendering GUI (llvmpipe)
- Keyboard
- Trackpad
- WiFi
- Webcam
- Battery Meter
- Screen Brightness adjustments
- USB A port and USB-C ports
- KVM (Virtualization)

Not working:
 - Audio - kernel dmesg shows a bunch of ASoC errors ("Playback_1: ASoC: no backend DAIs enabled for Playback_1" repeats a large number of times with other messages).

Note: My system has 8GB of RAM, and all of it shows up properly.

Overall Summary: Very usable, very snappy system.  I would highly recommend it.

hexdump0815, 2022-12-30:
@caljorden - thanks a lot for testing and reporting the info back ... regarding audio and some more info maybe useful asurada/spherion as well maybe have a look at https://github.com/hexdump0815/imagebuilder/issues/61 for inspiration

hexdump0815, 2023-01-11:
@caljorden - i have built an experimental image with a v6.1 kernel which has working panfrost gpu support (but touchpad and sound does not work) - in case you want to give it a try: https://github.com/hexdump0815/imagebuilder/releases/tag/230109-01

Pet0203, 2023-08-08:
> @caljorden - i have built an experimental image with a v6.1 kernel which has working panfrost gpu support (but touchpad and sound does not work) - in case you want to give it a try: https://github.com/hexdump0815/imagebuilder/releases/tag/230109-01

Can confirm GPU works. Would it be possible to fix touchpad and sound? I can try any images you want me to try. Would also be a nice learning opportunity for me!

hexdump0815, 2023-08-09:
it might be worth experimenting with mainline linux on asurada meanwhile - see: https://oftc.irclog.whitequark.org/aarch64-laptops/2023-07-26#32341129;

Pet0203, 2023-09-12:
@hexdump0815 would it be possible for you to release a new kernel/image for asurada with the latest stable kernel? There have been some fixes for the WiFi card stability on the mainline Linux tree that could potentially solve my biggest issue with Linux on Spherion. I've spent a few days trying to compile my own kernel but they refuse to boot. Thanks!

hexdump0815, 2023-09-12:
@Pet0203 - i might look into that, but it will take a bit of time - i first plan to do something completely different and then i plan to have a look at a mainline kernel for corsola/asurada/cherry maybe in the range of october to december, maybe targetting v6.6 as a potential next lts kernel ... maybe at that time mainline will even be a bit easier to deal with on those devices

Pet0203, 2023-11-16:
Can confirm mainline v6.7.0-rc1 is running on Spherion. My previous issues with the WiFi drivers timing out have been solved!
Audio no longer spams dmesg with errors, but there are still like 40 different sinks in the Audio Settings. ALSA detects the card correctly and loads the ucm config, but both pipewire and pulseaudio are getting the pcms instead of the "devices" defined in the ucm config.

Sleep is still not working, probably config error because the kernel has been patched for suspend/resume. Overall the system feels a lot more snappy and it feels like a gain on power efficiency, nothing I can confirm.

Last time I said that my kernels refused to boot. As it turned out, they didn't because I was able to SSH in. The issue was that the display output now runs on dp-aux as an edp panel.

hexdump0815, 2023-11-18:
@Pet0203 - thanks a lot for the detailed report - this sounds similar to what was reported on irc a while ago: https://oftc.irclog.whitequark.org/aarch64-laptops/2023-07-26#32341129 and https://oftc.irclog.whitequark.org/linux-cros-arm/2023-09-04#32456692 ... do you use any extra patches and/or kernel config options besides pure mainline/defconfig? i guess you are aware, but just in case you are not: https://patchwork.kernel.org/project/linux-mediatek/list/ is a good place to watch out for potential further patches for mt8192 ... for rthe sound stuff i guess some more work on the ucm files might be required to get it working - first step is to get some audio driver to be probed successfully (i.e. something in /proc/asound/cards) which it seems you have already and then ucm config is required to make it work properly ...

i plan to try to get some kind of mainline working on corsola (mt8186) soon and will most probably run into similar problems there and that is not as far developed as asurada in mainline yet even as it seems - my first tries with the collabora chromeos mainline tree from https://gitlab.collabora.com/google/chromeos-kernel on corsola so far only resulted in a hanging kernel at some point into the boot with a few errors :)

Pet0203, 2023-11-20:
Yeah I will give the UCM configs and puleaudio a second look at a later time. Really at this point the trackpad would be the most important thing for me to get working. For some reason the elan_i2c driver refuses to bind to the trackpad. I've dried using driverctl but to no avail. Really don't know what I should try next.

As for the config, I just used the defconfig for arm64 from [cros](https://chromium.googlesource.com/chromiumos/overlays/chromiumos-overlay/+/HEAD/eclass/cros-kernel/arm64_defconfig), excluded all platforms but Mediatek SoCs. Figured it was a good place to start when I didn't know what to include to get the display working. No patches though :)

Pet0203, 2024-11-04:
Hi again! Thought I could mention that I fixed both the trackpad, speakers and also some Bluetooth issues while on Debian. Nowadays I've switched to Arch and suspend suddenly decided to start working aswell.

@hexdump0815 your repos have been an enormous help, thank you for your work!

hexdump0815, 2024-11-04:
@Pet0203 - its nice to hear that you have it working better now ... could you maybe give some details what fixed the problems you mentioned, so that i can include ithe fixes into future images?

Pet0203, 2025-02-09:
Sorry, I saw your message but I had lots of other stuff going on at that time.  Sadly I lost my solutions that fixed sound and bluetooth on debian when I switched to arch. Really gotten into this stuff but I'm still just essentially playing around. 

### Trackpad:
The fixes for the trackpad have been merged to mainline. It was just a matter of using hid-over-i2c instead of elan, because spherion has a synaptics trackpad not elan.

### Bluetooth:
Currently this is standing in my way, was way worse previously with a bunch of errors
 `[   10.978329] Bluetooth: hci0: Opcode 0x0c03 failed: -110`

### DP Alt Mode
I'm currently running Wens mtk-wip branch and have tried the DP alt mode. I could get a 640x480 image so that's something for sure but not there yet, or maybe I'm just missing something. Still getting a failed to switch mode error in dmesg which is weird.

### Suspend:
It wasn't fixed, it just didn't hang the system. Currently getting this through ectool: `Warning: Detected sleep hang! Waking host up!` So about 5 seconds after going to S3 the EC sends a wakeup. I think I used [this](https://www.kernel.org/doc/Documentation/ABI/testing/debugfs-cros-ec) to basically ignore it. Now it seems like the CPU maintains S3 at least? I just got ectool to work but still struggling with SuzyQ. If I remember correctly it worked with the chromeos kernel so I will try and find what made it work there.

### GPU
As for the GPU, Panfrost seems to be working fine. Mesa made some changes one day that borked my laptop. It was just pure coincidence that I installed arch when it worked, as neither the versions before or after it seemed to work. Haven't tried updating in a while though. 
<details>
  <summary>Can confirm that 24.2.5 works though!</summary>

```
OpenGL vendor string: Mesa
OpenGL renderer string: Mali-G57 (Panfrost)
OpenGL core profile version string: 3.1 Mesa 24.2.5 (git-3b9fcb7e4d)
OpenGL core profile shading language version string: 1.40
OpenGL core profile context flags: (none)
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 24.2.5 (git-3b9fcb7e4d)
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.1 Mesa 24.2.5 (git-3b9fcb7e4d)
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
OpenGL ES profile extensions:
```
</details>

### FW-Devlink
Last year, Linux started to default to fw_devlink=rpm. My system didn't like that. I found that adding `driver_async_probe=* deferred_probe_timeout=2` to cmdline worked really well. I also set the config option `CONFIG_FW_DEVLINK_SYNC_STATE_TIMEOUT=y`, although I'm not sure I fully understand what it does. I'm not a driver dev. 

### Sound
The driver doesn't register the card through ALSA. Figuring this out has been a bit of a lower priority. Going to try different drivers.

hexdump0815, 2025-02-09:
@Pet0203 - thanks a lot for the detailed feedback - re suspend: maybe s2idle can be used as a workaround? on modern hardware it can reach days of suspend sometimes ... but of course having real suspend would be better

re audio: did you have audio working properly at any point? (i.e. is it theoretically possible - would be better than nothing, even if it no longer works for you)

re wens wip-mtk tree: is that the one from the kernel git server?

dinkumdingo, 2025-12-27:
@hexdump0815 I've tested all the images on my spherion and the trackpad only works on the first bullseye image. My trackpad shows up as `hid-over-i2c` on the first image, but doesn't show at all when calling `xinput list` on the newer experimental ones. Messing around with the `synaptics` and `libinput` packages didn't help.

Anyway, the audio works fine for me on the autumn test-image.

hexdump0815, 2026-11-08:
just fyi: it looks like someone has vibe coded panvk mesa support for the mali g57 (for an allwinner a523 in that case) - maybe it is something to play around with in case someone has some time and is open for some experimentation on the mt8192 :)

https://github.com/ut-slayer/mesa-panvk-g57
via: https://oftc.catirclogs.org/panfrost/2026-11-07#35867073

Pet0203, 2026-11-08:
That's just what I was about to start doing, thanks for saving me some tokens! I actually bought a SuzyQ-cable which helped me find out why the EC never went to sleep. Also flashed customized depthcharge with full disk-encryption and my own keys. Thanks for getting me into this hobby!

hexdump0815, 2026-11-09:
@Pet0203 - happy to hear that you got infected by the mainline linux on chromebooks virus - luckily its a rather harmless and nice one :)

please let us know here if you get anywhere regarding panvk (or not) and also please share if you found/find out any more about the suspend issues besides that suspend_timout_ms workaround - as far as i understand it the problems come from that the ec waits for all devices to tell it that they are ready for suspend and some or even one seems to be still missing with mainline preventing it from successfully entering sleep - i think this describes the same problem on trogdor devices where i recently saw it as well: https://oftc.catirclogs.org/aarch64-laptops/2026-07-25#35538940

do you have some links or more info about that customized depthcharge?

Pet0203, 2026-11-09:
> @Pet0203 - happy to hear that you got infected by the mainline linux on chromebooks virus - luckily its a rather harmless and nice one :)
>
> please let us know here if you get anywhere regarding panvk [...]

I disabled cros-ec-rpmsg in dt, modified mtk-nor-spi driver to init again again and force resume to fix the resume hang that I got. Then I had to use .prepare/.unprepare in panfrost driver to actually get the GPU to suspend.

Also fixed dp through usb-c.

I have patches for the kernel and depthcharge on a private repo, I can make some mirror repos on github for you and others.

hexdump0815, 2026-11-09:
@Pet0203 - that would be very nice if you could make those patches available somewhere on github as they might be quite useful for others as well.

hexdump0815, 2026-11-09:
@Pet0203 - btw i just ran across this mailing list thread while analyzing the mt8183 audio not working with v7.3 and i think it also mentions the mt8192 power-domain / suspend topic - might be an interesting read: https://lists.infradead.org/pipermail/linux-arm-kernel/2026-July/1156501.html

in the end it helped me to get audio working again on mt8183 :)