velvet-os/imagebuilder #61: system status: chromebook_cherry

hexdump0815, 2022-07-10:
this issue is just a placeholder to potentially give feedback if my untested experimental images from https://github.com/hexdump0815/imagebuilder/releases/tag/220710-01 are working at all and in case they are working to note down what works and what not ...

zhxt, 2022-07-22:
Hi, thanks a lot for your efforts! 
I had tried. It's working pretty well, except audio and video hardware acceleration.

What works and what not: 
- [x] Display
- [x] Camera
- [x] WiFi
- [x] Bluetooth (can connect to other devices, sending file failed)
- [x] Video (soft decode)
- [ ] Video Codec (hardware acceleration) 
- [ ] Audio (no sound, pulseaudio shows zero sinks/sources)

hexdump0815, 2022-07-22:
thanks a lot for testing - that is actually way better than i expected :)

for audio: what does "cat /proc/asound/cards" show?

zhxt, 2022-07-23:
- The output looks like:
```
cat /proc/asound/cards
 0 [m8195r10195682s]: m8195_r1019_568 - m8195_r1019_5682s
                      m8195_r1019_5682s

```
- and, found a [ucm config](https://chromium.googlesource.com/chromiumos/overlays/board-overlays/+/HEAD/overlay-cherry/chromeos-base/chromeos-bsp-cherry/files/tomato/audio/ucm-config/m8195_r1019_5682s.tomato/) for this card in chromeos source code.

- also, seems something wrong with ALSA ASoC driver. 
```
dmesg -l err
[    1.275644] mtk_cmdq 10320000.disp_mailbox: failed to get gce clk: 0
[    1.282094] mtk_cmdq 10330000.mdp_mailbox: failed to get gce clk: 0
[    1.614156] mtk-tphy soc:t-phy@11c40000: failed to get ref_clk(id-1)
[    1.620657] mtk-tphy soc:t-phy@11c50000: failed to get ref_clk(id-1)
[    1.627173] mtk-tphy soc:t-phy@11e40000: failed to get ref_clk(id-1)
[    1.634685] mt8195-pinctrl 10005000.pinctrl: pin GPIO19 already requested by 10005000.pinctrl; cannot claim for 112f0000.pcie
[    1.646073] mt8195-pinctrl 10005000.pinctrl: pin-19 (112f0000.pcie) status -22
[    1.653346] mt8195-pinctrl 10005000.pinctrl: could not request pin 19 (GPIO19) from group GPIO19  on device pinctrl_paris
[    1.664380] mtk-pcie 112f0000.pcie: Error applying setting, reverse things back
[    1.735591] mtk-mmsys 1c01a000.clock-controller: error -2 can't parse gce-client-reg property (0)
[    1.751979] mtk-mmsys 1c01a000.clock-controller: error -2 can't parse gce-client-reg property (0)
[    1.985198] mediatek-disp-ovl-adaptor mediatek-disp-ovl-adaptor.6.auto: Failed to get id. type: 1, alias: -19
[    2.006930] mediatek-disp-ovl-adaptor mediatek-disp-ovl-adaptor.6.auto: Failed to get id. type: 1, alias: 5
[    2.359772] mt8195-pinctrl 10005000.pinctrl: pin GPIO8 already requested by 10005000.pinctrl; cannot claim for 11e00000.i2c
[    2.370906] mt8195-pinctrl 10005000.pinctrl: pin-8 (11e00000.i2c) status -22
[    2.377949] mt8195-pinctrl 10005000.pinctrl: could not request pin 8 (GPIO8) from group GPIO8  on device pinctrl_paris
[    2.388635] i2c-mt65xx 11e00000.i2c: Error applying setting, reverse things back
[    2.804271] mtk-wdt 10007000.watchdog: IRQ index 0 not found
[    3.315961] mt8195_mt6359_rt1019_rt5682 mt8195-sound: mt8195_mt6359_rt1019_rt5682_dev_probe snd_soc_register_card fail -517
[    3.497683] mediatek-mutex 1c016000.disp_mutex0: error -2 can't parse gce-client-reg property (0)
[    3.513994] mediatek-mutex 1c101000.disp_mutex0: error -2 can't parse gce-client-reg property (0)
[    3.531353] mediatek-drm-dp 1c500000.edp_tx: Failed to find panel or bridge: -517
[    3.548313] mediatek-drm-dp 1c600000.dp_tx: Failed to get dptx clock: -2
[    3.961959] hub 4-0:1.0: config failed, hub doesn't have any ports! (err -19)
[    4.105580] hub 6-0:1.0: config failed, hub doesn't have any ports! (err -19)
[    4.322563] xhci-mtk 11200000.xhci: drop_ep_quirk high-speed ep1in intr, mpkt:1, interval:12/256ms
[    4.863472] mediatek-drm-dp 1c500000.edp_tx: Failed to find panel or bridge: -517
[    5.332063] mediatek-drm-dp 1c500000.edp_tx: Failed to get dptx clock: -2
[    5.529826] usb 1-1-port1: Cannot enable. Maybe the USB cable is bad?
[    6.524343] usb 1-1-port1: Cannot enable. Maybe the USB cable is bad?
[    6.797317] systemd[1]: Failed to find module 'autofs4'
[    7.928819] usb 1-1-port1: Cannot enable. Maybe the USB cable is bad?
[    8.374035] remoteproc remoteproc0: request_firmware failed: -2
[    8.446301] mtk-vdec-comp 18002000.vcodec_lat_soc: IRQ index 0 not found
[    8.446303] mtk-vdec-comp 18002000.vcodec_lat_soc: Failed to get irq resource
[    8.446303] mtk-vdec-comp 18002000.vcodec_lat_soc: Failed to register irq handler.
[    8.526550] mt7921e 0000:01:00.0: ASIC revision: 79610010
[    8.631498] remoteproc remoteproc0: request_firmware failed: -2
[    8.766286] [MTK_V4L2][ERROR] fops_vcodec_open:176: vpu_load_firmware failed!
[    8.783682] remoteproc remoteproc0: request_firmware failed: -2
[    8.783687] [MTK_V4L2][ERROR] fops_vcodec_open:321: failed to load firmware!
[    8.898379] usb 1-1-port1: Cannot enable. Maybe the USB cable is bad?
[    8.898810] usb 1-1-port1: unable to enumerate USB device
[    9.035493] mt7921e 0000:01:00.0: Firmware init done
[   11.143603] Bluetooth: read_supported_features() hci0: Failed to read MSFT supported features (-56)
[   12.086842]  DL2_FE: ASoC: no backend DAIs enabled for DL2_FE
[   12.092589]  DL2_FE: ASoC: dpcm_fe_dai_prepare() failed (-22)
[   12.098391]  DL2_FE: ASoC: no backend DAIs enabled for DL2_FE
[   12.104135]  DL2_FE: ASoC: dpcm_fe_dai_prepare() failed (-22)
[   12.104170]  DL2_FE: ASoC: no backend DAIs enabled for DL2_FE
[   12.104171]  DL2_FE: ASoC: dpcm_fe_dai_prepare() failed (-22)
[   12.104260]  DL2_FE: ASoC: no backend DAIs enabled for DL2_FE
[   12.114790]  DL2_FE: ASoC: dpcm_fe_dai_prepare() failed (-22)
[   12.114895]  DL2_FE: ASoC: no backend DAIs enabled for DL2_FE
[   12.120585] remoteproc remoteproc0: request_firmware failed: -2
[   12.126359]  DL2_FE: ASoC: dpcm_fe_dai_prepare() failed (-22)
[   12.132285] [MTK_V4L2][ERROR] fops_vcodec_open:321: failed to load firmware!
[   12.141120]  DL2_FE: ASoC: no backend DAIs enabled for DL2_FE
[   12.151773]  DL2_FE: ASoC: dpcm_fe_dai_prepare() failed (-22)
[   12.151822]  DL2_FE: ASoC: no backend DAIs enabled for DL2_FE
[   12.163436]  DL2_FE: ASoC: dpcm_fe_dai_prepare() failed (-22)
[   12.163472]  DL2_FE: ASoC: no backend DAIs enabled for DL2_FE
[   12.170478] remoteproc remoteproc0: request_firmware failed: -2
[... cut ...]
[   12.267338]  DL2_FE: ASoC: no backend DAIs enabled for DL2_FE
[   12.273079]  DL2_FE: ASoC: dpcm_fe_dai_prepare() failed (-22)
[   12.278886]  DL2_FE: ASoC: no backend DAIs enabled for DL2_FE
[   12.284626]  DL2_FE: ASoC: dpcm_fe_dai_prepare() failed (-22)
[   12.290492]  DL2_FE: ASoC: no backend DAIs enabled for DL2_FE
[   12.296228]  DL2_FE: ASoC: dpcm_fe_dai_prepare() failed (-22)

```
- The codec driver looks like missing firmware `[MTK_V4L2][ERROR] fops_vcodec_open:176: vpu_load_firmware failed!`

hexdump0815, 2022-07-23:
the chromeos ucm config would be a good start i guess - i have collected some notes about getting audio working here: https://github.com/hexdump0815/imagebuilder/blob/main/info/generic/alsa-ucm-pulseaudio-hacking.txt ...

there is some limitation in the length of the ucm dir name, so it should go to /usr/share/alsa/ucm2/m8195_r1019_568 (see also the output of the /proc/alsa/cards) ... i think it is a good idea to put a "Syntax 2" on top of m8195_r1019_5682s.tomato.conf and to comment out all devices except the speakers at the beginning

to check if the proper ucm file gets loaded it is a good idea to run "strace alsaucm -c myconfig reload" (will have to add it to the notes mentioned above)

in general when trying to bring up audio never use headphones directly until you have it working as you might get high volume level in the process which is not good for the ears

regarding video accel: i guess the chance is low to get this working with a regular linux without a lot of extra patches i guess (but i might be wrong)

good luck and best wishes - hexdump

hexdump0815, 2022-07-23:
this might be a good read about getting audio to work: https://github.com/hexdump0815/imagebuilder/issues/54

zhxt, 2022-07-24:
Thanks for the detail info, will try.

zhxt, 2022-07-24:
Wow! That worked, audio is working now ! hahaha

What works and what not: 
- [x] Display
- [x] Camera
- [x] WiFi
- [x] Bluetooth (can connect to other devices, sending file failed)
- [x] Video (soft decode)
- [ ] Video Codec (hardware acceleration) 
- [x] Audio
- [x] Speaker
- [x] Mic ( need to adjust volume `pactl set-source-volume 1 400%` as the note said)

Many many thanks!

hexdump0815, 2022-07-24:
@zhxt - that sounds very good!

could you please create a pull request with your working ucm files or upload/post them somewhere, so that i can include them?

zhxt, 2022-07-24:
@hexdump0815 PR #64 created :D

hexdump0815, 2022-07-24:
@zhxt - thanks a lot - pull request is already merged :)

regarding the default to headphone what i have learned over time is that the last entry in the ucm file is used by default as it seems - so maybe you can get this fixed by just reordering the entries to have the speaker last ... same applies for the mics

this is the ugly hack for kukui chromebooks to initially set the devices: https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_kukui/rc-local-additions.txt#L29-L73

best wishes - hexdump

zhxt, 2022-07-25:
@hexdump0815 thanks, reorder the entries did the trick :-D

for initialize: maybe add a `BootSequence` section will work, I will test later.

and btw, do you think is it possible to get GPU work ?

hexdump0815, 2022-07-25:
@zhxt - i think in newer ucm versions (syntax 3 and 4 i think) they restructured things completely with proper init sections (so no more enadev etc. required) and maybe there are more keywords ... chromeos has some extra keywords in their ucm files which are specific to their cras (chrome-audio-server i think) and not really part of reglar ucm

i think gpu support will come with mainline support via the panfrost project, but all that is in very early stages and will still need quite a few months until it might start to get useable, so most probably some patience is required for this

zhxt, 2022-07-26:
@hexdump0815 tried, `BootSequence` still need to call `aslactl init`.

and yeah, seems cras uses a separate program called `sound_card_init` triggered by udev  rules to do the initialization.

for GPU, I was thinking, chromeos came with GPU support, and debian uses the same kernel,  the point is cros is 32bit, but debian is 64bit, so if the user space binary blobs can be get from somewhere,  will it work on debian ?

I found that there is a  [overlay-cherry64](https://chromium.googlesource.com/chromiumos/overlays/board-overlays/+/refs/heads/main/overlay-cherry64/) directory, maybe there will be a 64bit cros in the future :)

hexdump0815, 2022-07-26:
the mali blobs are not really useable for linux as they are for the chromeos windows system or low level drm (not sure) and thus have a completely different api than what xorg or wayland are using ... but initial mali valhal support is there in panfrost, so its just a matter of time that it will be useable

hexdump0815, 2022-07-26:
i have reordered the devices in the ucm files in the repo now as well: https://github.com/hexdump0815/imagebuilder/commit/b659e964f86b4fd4adae3f0d17897f8f62b21627

zhxt, 2022-07-28:
> the mali blobs are not really useable for linux as they are for the chromeos windows system or low level drm (not sure) and thus have a completely different api than what xorg or wayland are using [...]

yeah, need to wait  for now.
and btw, how about kukui ? It has a bifrost  arch GPU, seems panfrost already support it .

hexdump0815, 2022-07-28:
kukui is in very good shape - with my latest images nearly everything is working quite well except of hw video decoding, external monitors and on some devices the cameras if no usb video class devices are used ... otherwise gpu is working with panfrost, wifi, bt, suspend/resume, sound all works quite well with little bugs here and there maybe :) - see for example https://github.com/hexdump0815/imagebuilder/issues/52

hexdump0815, 2022-07-28:
@zhxt - i read your conversation on panfrost irc - good luck for trying mainline - please let me know here if you get anywhere near a working kernel with this ...

zhxt, 2022-07-29:
> kukui is in very good shape - with my latest images nearly everything is working quite well except of hw video decoding, external monitors and on some devices the cameras if no usb video class [...]

Sounds great! 

> @zhxt - i read your conversation on panfrost irc - good luck for trying mainline - please let me know here if you get anywhere near a working kernel with this ...

yeah, I haven't tried that kernel yet. I will let you know if I get any progress :D

zhxt, 2022-10-26:
Hey, finally got some time to try. :laughing: 
The kernel seems work fine , but I got no GUI with a black screen.
I can login with ctrl+alt+fn, the [log](https://gist.github.com/zhxt/18755f8377d04d733b0d3a1d5cfa4efd) shows lightdm service failed to start, still need more time to find out the reason.

hexdump0815, 2022-10-26:
@zhxt - from irc i remember some trouble with the /dev/dri device numbering (https://oftc.irclog.whitequark.org/panfrost/2022-10-24 and the next days) - maybe try something like https://github.com/hexdump0815/imagebuilder/blob/main/files/extra-files/etc/X11/xorg.conf.d.samples/13-panfrost-swapped-dri-nodes.conf

best wishes and good luck - hexdump

zhxt, 2022-10-27:
@hexdump0815 yeah, I saw that too, but I didn't find any log related Xorg,  when I start lightdm manually from terminal, it just jump to the black screen with a flicking cursor.

zhxt, 2022-10-27:
@hexdump0815 wow, tried the patch, that did the trick. thanks.
A new problem is when login with plasma wayland,  it just goes back to login window  again after a few seconds while enter password.
Have you tried plasma-wayland on other devices?

hexdump0815, 2022-10-29:
@zhxt - are you using the latest version of mesa?

zhxt, 2022-10-29:
> @zhxt - are you using the latest version of mesa?

yeah, I rebuild mesa 22.2 taken from bookwarm .

hexdump0815, 2023-01-11:
@zhxt - i have built an experimental image with a v6.1 kernel which has working panfrost gpu support - it has not yet been tested on mt8195/tomato (only on mt8192 so far) and it would be interesting to hear if it is working - in case you want to give it a try: https://github.com/hexdump0815/imagebuilder/releases/tag/230109-01

bruceloco, 2023-06-19:
following the untimely demise of my CP513-1H was I was 90% of the way setting it up, i got the CP513-2H
I tested this image and it takes ages to boot.
Like strangely slow, I will try to build a an image based on the chromebook defconfig mixed with yours
there are strange modules like nvidia and other things like apparmor that take time and are not fixed
will attempt a 6.3 linux build again now for the Kompanio 1380

bruceloco, 2023-12-05:
And I am back
I used the mt9 config file, some changes, did not copy all the dtbs for the CP513-2H, and built kernel 6.6.2
Getting something past 6.4 to get panfrost support.
Also I flashed the kpart to partition 0, filesystem is on partition 2
I did not do the initramfs as I did not think it necessary and I am not there yet.
When I boot to the USB device, I get a blank screen, screen turns on but there is nothing there, no console.
Any clues on how to debug it?

hexdump0815, 2023-12-06:
@bruceloco - first would be trying to figure out if it boots at all - for that it might be good to add something like "dmesg > /root/dmesg.txt; shutdown -h now" to rc.local and then you can see if the system will shutdown and if the file exists afterwards ... if that is working then it is a good idea to make sure (with some well working kernel) that you end up with a working network connection when fully booted (best lower level like usb ethernet as it is easier than when using wifi) - if that works you may login via ssh even if the screen, kdb etc. do not work and then you can move on debugging from there ... alternatives would be to get or rather build nowadays a suzyqable yourself which can give you a serial console on chromebooks (see https://github.com/hexdump0815/linux-mainline-on-arm-chromebooks) ... in some situations also something like https://github.com/Sonicadvance1/linux/issues/27#issuecomment-823668871 might be helpful ...

bruceloco, 2023-12-06:
@hexdump
The first cherry image you provided booted.Will try the first approach on dumping the dmesg to a file and I was halfway through a homebrew suzyqable.Does the double serial work as a suzyqable?

hexdump0815, 2023-12-07:
@bruceloco - i just remembered, that on tomato maybe reading around this one here might be useful: https://oftc.irclog.whitequark.org/aarch64-laptops/2023-10-06#32545749 - this mentions a patch required for the display to work: https://lore.kernel.org/all/20230816190427.2137768-1-nfraprado@collabora.com/#r ... not sure if this is your problem, but worth looking into maybe ... in general that user on irc seems to have tomato running quite well with mainline or the collabora chromeos-close-to-mainline tree from https://gitlab.collabora.com/google/chromeos-kernel/-/tree/for-kernelci ... maybe some of this information is useful for you to move forward ... best wishes and good luck

hexdump0815, 2023-12-07:
i think colabora is doing the mainlining of the mediatek chromebook kernels for google, so they use a kernel which is always updated to the latest mainline dev kernel version and has some of the chromeos special patches in it on top - they use this kernel to upstream the mt81xx support to mainline ... it might be a good option to try if pure mainline is still not reliable enough on cherry tomatoes ...

hexdump0815, 2023-12-08:
i just noticed that o forgot to answer the question about the dual usb-serial vs suzyqable from above: suzyqable is special for chromebooks and gives a full serial console while the dual us-bserial will only give serial output if usb is at least working

bruceloco, 2023-12-08:
@hexdump0815 

Thanks for that tip, that guy is one of the contributors to this git: https://git.kernel.org/pub/scm/linux/kernel/git/chunkuang.hu/linux.git/commit/?h=mediatek-drm-fixes
marginal success, still crashed but this is now up to me to setup the services to load missing firmwares using the modem.
I did see the penguin and the log though so it's a major step up :)

hexdump0815, 2023-12-09:
@bruceloco - that are very good news - please keep me/us up here in case of any further progress

bruceloco, 2023-12-10:
@hexdump0815 
what are the absolute CONFIG settings to enable the console TTY?
Is there anything on a config that must not be set and a group of things that have to be set?
I can use your config, but your config has a lot of things not required for the "cherry".
I enabled the coreboot console, framebuffer console and using the google config I get a blank screen.
Not sure if this could be a backlight setting, otherwise I need to make a suzyqable.
Thanks

When I sort this out I will provide you with a cherry tomato config that is working on the DRM fixes branch of the 6.7 RC1 from the above git repo.

hexdump0815, 2023-12-10:
you mentioned that you saw the penguins - right? that means the console initially came on i guess - at some point it is taken over by the drm system - maybe this is where it breaks? does it still crash, i.e. no dmesg if you write it to the filesystem, sync, shutdown via rc.local?

bruceloco, 2023-12-10:
@hexdump0815  that was building with your defconfig
for some reason now when I build it removes options, example, copying the config.mt9 to .config and making olddefconfig removes the framebuffer devices and other things
how do I force linux to build a particular config file?

bruceloco, 2023-12-10:
ignore that, i was being an idiot, there are some dependencies that I needed to follow for options enabling
if they are not followed, a make will clean the enabled options without the dependencies so the .config was missing stuff
now its all runninng
I am on void linux musl with the above kernel and the mt9 config :)
I will shave the mt9 config to remove a lot of the stuff to make a leaner kernel as well

bruceloco, 2023-12-13:
@hexdump0815 got a cherry config
Wifi is working, haven't installed Sway or anything of the sort so running of the console.
One wifi snag - bad assert on wiphy config in mac80211 - get_tx_power-> need to dig deeper to see what device actually did it.
It might not be setup at the time that assert is called.
Once I got Sway(stepping stone to Enlightment 0.25.4) and sound tested will post it here.

bruceloco, 2023-12-14:
@hexdump0815 I fixed the wifi bad assert, Chun-Kuang Hu did not pull the code from nl80211.c where they lock the wiphy mutex.
I sent him the fix, but its direct email so I might be ignored.
I will test this kernel version working forward and give you feedback on it.

bruceloco, 2023-12-16:
@hexdump0815 
based on the latest tag of drm fixes
https://git.kernel.org/pub/scm/linux/kernel/git/chunkuang.hu/linux.git/tag/?h=mediatek-drm-fixes-20231211

adding this fix otherwise wifi produce error messages
https://github.com/torvalds/linux/commit/103317670e6bf2542309db28d52444a83d84ed28

A fix for audio as its loading the wrong firwware file
/sound/soc/sof/mediatek/mt8195/mt8195.c is wrong
(https://github.com/hexdump0815/imagebuilder/files/13690787/config.zip)

		.sof_tplg_filename = "sof-mt8195-mt6359-rt1019-rt5682-dts.tplg"
should be
		.sof_tplg_filename = "sof-mt8195-mt6359-rt1019-rt5682.tplg"

Attaching the current .config
[config.zip]

hexdump0815, 2023-12-17:
@bruceloco - thanks a lot for this information - i hope to have time to look a bit closer at all this next week as maybe it helps me getting corsola (mt8186) forward a bit with mainline as well

bruceloco, 2023-12-17:
@hexdump0815 it's just contributing, i would have not gone far without you so I have to pay it forward :)
I will also check how to use git to get a latest kernel - mainline to latest + mediatek directory specific fixes(this can be applied to qualcomm, etc if you know the repos/branches
It will be some sort of git recipe in a script.
One thing I don't know is how to set the resolution in the console, setting fb modes does nothing except for reducing the tty area, the resolution stays the same.
since this is panfrost it has a specific something, not sure its a video buffer, some special frame buffer or if it is setting hdmi modes because the text is really small, its extremely tiring to read.
I am still battling Enlightenment.org loading(some stupidity i have done with the libs folder) and musl - it has no system locales which does not help Enlightenment.
I had sway come up(just the image) no config or text menus or anything useful, did not bother to configure it.

The advantage is that I do my builds on my chromebook in a chroot and ssh into the terminal so i can have an actual terminal and a browser for searches. so its easy to cat the output of modules and things that are loaded, compare the dmesg output, etc

hexdump0815, 2023-12-17:
i just want to say that i'm very happy with such kinds of contribution as it helps a lot to bring the idea to run mainline linux on chromebooks forward quite a bit if more and more people start researching how to get and keep it working on all the different hardware - so all your effort is very welcome

bruceloco, 2023-12-18:
@hexdump0815 
Another thing to consider is since we are moving to later kernels with newer options we benefit from newer firmwares:

bruceloco, 2023-12-18:
and for extra fun - we have a non changeable mode 2256x1504 (3:2)
Which normal people can't read and you can't change modes in the framebuffer but you can set the font:
Install the terminus-font for you distro
follow this guide:
https://wiki.archlinux.org/title/Linux_console#Fonts

I no longer feel blind

oddajpierscien, 2024-08-20:
Hello, is there any chance for updated image for Cherry? :D I couldn't resolve freezes on experimental image when I've connected to wifi.

thenameisluk, 2024-08-20:
@oddajpierscien yes but [likely around february](https://github.com/hexdump0815/imagebuilder/issues/209#issuecomment-2251317062)
hexdump [doesn't have a lot of time currently](https://github.com/hexdump0815/imagebuilder/issues/129#issuecomment-2211262173)

there is a chance you can find a solution in this issue or
that non experimental image version from https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_asurada/readme.md will work

the thing is this device is currently very much experimental and not a lot of ppl who know how to improve it have it

if you want something working fairly well i would recommend getting [kukui](https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_kukui/readme.md) or (newer) [trogdor](https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_trogdor/readme.md) chromebooks
you can get some nice prices for (especially used) models on you local shopping app or ebay

hexdump0815, 2026-11-08:
just fyi: it looks like someone has vibe coded panvk mesa support for the mali g57 (for an allwinner a523 in that case) - maybe it is something to play around with in case someone has some time and is open for some experimentation on the mt8195 :)

https://github.com/ut-slayer/mesa-panvk-g57
via: https://oftc.catirclogs.org/panfrost/2026-11-07#35867073