velvet-os/imagebuilder #65: [BUG] High pitch noise - Debian 11 - kappa

aneeshlingala, 2022-08-02:
A high pitch noise comes sometimes when system wakes up from sleep. And it can only be stopped by turning off the system. Please fix this

hexdump0815, 2022-08-03:
i so far not have observed this on my kappa or anorther kukui device, but i have observed it on an intel celeron chromebook running linux as well - i think what helped there was to play some sound to stop that strange noise ... it might be worth a try to add playing a sound file with a few seconds of silence on wakeup to avoid something like this ... i'll have a look at this, but it will take a few days i guess ... can you meanwhile maybe verify if playing some sound helps to stop the noise? maybe just have a browser tab with a youtube video around which you start playing as soon as the noise appears again on wakeup.

aneeshlingala, 2022-08-03:
Sure, could it be something related to the ucm file ported from chromeOS, or it could be alsa freaking out.

hexdump0815, 2022-08-03:
i do not think that it is related to ucm, i guess its more a problem with the reinitialization of the audio hardware after resume (maybe some timing issue, so that it only happens from time to time)

aneeshlingala, 2022-08-04:
Oh, still haven't got the high-pitched noise yet, I will make a comment when it happens.

aneeshlingala, 2022-08-07:
Well, I got the High-Pitched noise and muting didn't help, I couldn't play a video in time as it stopped 6 seconds after.

hexdump0815, 2022-08-07:
what did stop 6 seconds after?

some more ideas to debug the problem:
- what happens if you put the system back into sleep again immediately and wake it up a little later?
- to rule out pulesaudio try: "systemctl --user stop pulseaudio.service" as the logged in user and check if that stops the noise (i guess it will not)
- try to reload the alsa ucm config (run as root): "alsaucm -c mt8183_mt6358_t reload" or "alsaucm -c mt8183_mt6358_t reset"
- try to unload the sound driver (run as root): "rmmod mt8183_mt6358_ts3a227_max98357 snd_soc_mt8183_afe snd_soc_mtk_common snd_soc_ts3a227e snd_soc_mt6358 snd_soc_max98357a"

good luck and best wishes - hexdump

aneeshlingala, 2022-08-08:
Sure, will try when im free

aneeshlingala, 2022-08-26:
Sadly, i'm not using the device anymore, so I gave it to someone and reinstalled chromeOS on it, so I will close this issue.

aneeshlingala, 2022-09-17:
I got the laptop back again.

aneeshlingala, 2022-10-16:
Well, I played a video when this happened a few days ago, it did not stop the high pitched noise

hexdump0815, 2022-10-16:
ok - thanks for the report - no real idea yet how to get this resolved - will maybe try to daily drive my kappa soon to see if i run into the same problem with it

aneeshlingala, 2022-10-16:
Sure.

Thanks, Aneesh

hexdump0815, 2022-10-28:
@aneeshlingala - just a little update: i used my kappa for a week now as my main daily system and was not able to reproduce the problem - no strange noise and i did use it with sound, without sound, put it into sleep mode a few times per day etc. ... this was with debian bullseye upgraded to bookworm with the normal xfce desktop

hexdump0815, 2022-10-29:
ok - got it this morning myself as well while resuming from suspend - putting it back into suspend and resume once more fixed the problem (i.e. no more strange sound) ... it looks like the problem was there the following message showed up in dmesg:
```
mt8183-audio 11220000.audio-controller:mt8183-afe-pcm: mtk_afe_pcm_pointer hw_ptr err
```

aneeshlingala, 2022-10-30:
Glad to hear it, I am currently using chromeOS in a need of stability, I
will reinstall soon.

Thanks,
Aneesh

hexdump0815, 2022-11-14:
i have seen this on juniper now as well - so it sems to affect multiple kukui devices, maybe all with this audio setup or maybe even all kukui devices

aneeshlingala, 2023-05-26:
On kernel 6.1.11, I have not experienced the noise. Closing the issue.

hexdump0815, 2023-05-26:
yes also i did not yet notice it anymore for a long time since using the v6.1.11 kernel, so i can confirm your observation

aneeshlingala, 2023-06-25:
Hello Hexdump0815,

Reopening the issue. Few days ago, the high pitched noise came back. This is on the kernel 6.1.11-stb-mt8+.

Thanks,
Aneesh

hexdump0815, 2023-06-26:
oh - that is bad news then - but as the cause of the problem is still unknown there is no other option than the workaround to put system to sleep and wake it up again ... lets see, hopefully some kernel update one day will fix it

i'm using my kappa as daily driver again right now using the v6.1.11 kernel and did not run into the problem again yet.

aneeshlingala, 2023-06-29:
Good news:
It happened again, but I have noticed it only happens for a few seconds instead.

willmasters, 2023-11-28:
I also get the crazy long beep after resuming on krane in the September release, it's very loud and distressing. X)

Can't really take it out the house because of it. People will probably thing there is a fire alarm going off. :)

hexdump0815, 2023-11-28:
@willmasters - i so far only got it on kappa and never on krane yet and that the suspend/resume workaround does not help on your device does not sound very good ... sadly i still have no idea how to get rid of it

willmasters, 2023-11-28:
Thanks @hexdump0815 . :) And no worries, if I find out anything further - I will post it here.

aneeshlingala, 2023-11-28:
One extra thing:

This happens on Postmarketos and my custom Arch Linux Image.

hexdump0815, 2024-01-02:
just a little update: i'm running currently some v6.6 kernel on my kappa and i haven't heard the bad sound for a long time - i'm not sure if its really gone or if i'm just lucky, but maybe newer kernels will bring some help with this - lets see what time will tell :)

aneeshlingala, 2024-01-02:
Lets hope it has gone over time... And also, happy new year

hexdump0815, 2024-01-09:
happy new year as well

it looks like the problem maybe manifests in a different way with newer kernels which is less annoying: i noticed that sometimes something seems to be wrong with the audio clocks resulting in audio or videos with audio are being played extremely slow - the "fix" to make it work is the same es in the past: just put the device into sleep and wake it back up and things are fine again ... so maybe some of the clocks are not restored properly on suspend/resume from time to time ...

aneeshlingala, 2024-02-10:
i have also tested the new 6.6 kernel, and I haven't gotten any high pitch sound. I observed this has been fixed from kernel ver. 6.5.5. closing the issue

hexdump0815, 2024-06-01:
today i got the noise once even with my v6.6.9 kernel, but that was after months of intensive use of my kappa chromebook as daily driver, so it looks like its not completely gone, but very very rare meanwhile ...

aneeshlingala, 2024-06-01:
Weird, have been daily drive ring my kappa with KDE and there has been no
sound, also tried MATE, no high pitched sound as well. This might be
related to DE but I'm not too sure because it also happens on postmarketOS
with phosh

aneeshlingala, 2025-04-10:
This has happened to me again on 6.12.20 cbm kernel, I will attach a video.

aneeshlingala, 2025-04-10:
I wasn't able to upload a file on GitHub, I have uploaded the video on a separate site. [https://jumpshare.com/embed/QaBZ3VWExICFTH30rYax](https://jumpshare.com/embed/QaBZ3VWExICFTH30rYax)

aneeshlingala, 2025-04-10:
Again, the logs show `mt8183-audio 11220000.audio-controller:mt8183-afe-pcm: mtk_afe_pcm_pointer hw_ptr err ` being spammed in dmesg

thenameisluk, 2025-04-11:
yeah we still don't know the cause of this issue

but if you run (when it happens)
```
systemctl restart --user pipewire
```
it should get rid of the noise (for the time being)

sad thing is we can't even git bisect this since there was no version without this issue afaik

aneeshlingala, 2025-04-11:
Issue has been on and off for 3 years. It was fine until 6.6 kernel. Most
likely cbm merge did something?

aneeshlingala, 2025-04-11:
However, this only happened once so we still need to see if it is a rare occurrence like 6.6

hexdump0815, 2025-04-11:
@aneeshlingala - i guess it was always there, it just happens so rarely and randomly that sometimes one might think its gone - and then it comes back one day :) ... for me i have not run into it for a long time (using my kappa as daily driver), but from time to time the audio clocks seem to be off (i.e. videos play back extremely slow) and i have to suspend/unsuspend to make it work again ... will have to give the pipewire/pulseaudio restart a try ...

hexdump0815, 2025-05-02:
just to have it as a reference here - this one might be related: https://patchwork.kernel.org/project/linux-mediatek/cover/20250424102509.1083185-1-wenst@chromium.org/ ... in the recent times (=months) i got the high pitch sound only once or twice but a few more times situations where the sound was either extremely slow (another suspend/resume usually fixed that) or not working at all (reboot required to get the sound back) ... in the non working case i noticed some memory allocation error in the sound driver in the logs, so it might be related to that patch series which somewhere also mentions memory starvation ... as i read it this patch series is still wip and not yet ready completely for mt8183, but definitely something to keep an eye on over the next months ...

hexdump0815, 2025-05-14:
it also happens with pulseaudio, it can just take quite a while until it happens

thenameisluk, 2025-05-14:
@hexdump0815 given that the noise only occures after suspense could this fix it?

at ```/etc/systemd/system/fix-pipewire.service```
```
[Unit]
Description=stop high pitch
After=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target

[Service]
ExecStart=systemctl restart --user -M linux@ pipewire

[Install]
WantedBy=suspend.target hibernate.target hybrid-sleep.target suspend-then-hibernate.target
```

the ```systemctl restart --user pipewire``` is already known to get rid of the noise when it occures
(and the ```-M linux@``` part if to do it for specyfic user since this one is ran by root)

it is a hack but unless someone finds a solution it might be better than nothing

hexdump0815, 2025-05-15:
@thenameisluk - i'm still using pulseaudio and there restarting oes not seem to help (tried it in a few cases), but for pipewire we should at least document what you sugested as a workaround (or point to your last post in the kukui specific doc maybe)

roomfullofcommits, 2026-11-14:
I also have the high pitch noise and i think i noticed a pattern that i havent seen mentioned before.  
  
Device: Krane
Kernel: 6.18.34
Debian Trixie
using niri with noctalia

my workaround has been to use noctalias hooks to stop pipewire-pulse when locking the device and restarting it when unlocking. this still results in the noise being played but only for about a second. it might be possible to improve this by having a bigger delay between stopzing pipewire and suspending the device, but i havent tried that.

but heres a pattern ive noticed: the noise only appears if there is audio playing when suspending the device. if there is no audio playing at the time of suspending it, the noise will not show up.

archisman-panigrahi, 2026-11-15:
Someone please try installing Opencode (it is an AI agent with free models that can fix issues in your computer) and asking it to fix it. It solved microphone issue for me in another chromebook (I'll soon share the solution in an appropriate (different) thread).