velvet-os/imagebuilder #66: ntfs3

sirzur, 2022-08-09:
I have been trying to mount an NTFS drive on a Odroid U3 using the latest Odroid release.  Normally I mount using ntfs-3g in conjunction with fuse.  However the included packages are ntfs-3g and fuse3.  f I install the fuse package, ntfs-3g and fuse3 is removed.  I understand that the current way to mount ntfs disk is using ntfs3 now, but I think it is not configured in your kernel/packages.  Can I request that your next release for Odroid U3 include ntfs3 configured.

hexdump0815, 2022-08-12:
i'll put it onto my todo list for the kernel options, but it might take some time

sirzur, 2022-08-13:
Thank you for your hard work and I look forward to it

sirzur, 2022-08-14:
In my OP, I mentioned that I originally started down this path in order to mount an NTFS USB drive on the Odroid U3 using the images generated by your hard work (thanks).  After a few days of mucking around, being a complete newbie at kernel compilation, I managed to compile a kernel with fuse enabled.  All I did was included the following 3 lines into one of the scripts that was merged into the main config

CONFIG_FUSE_FS=y
CONFIG_CUSE=y
CONFIG_VIRTIO_FS=y

I suspect that the last two lines are not necessary, but they were sub-options which appeared when I ticked yes to enable fuse_fs.  Another option to include these 3 lines will be to create another config file with just these 3 lines and append the path of that file to the end of your merge command.

In any case, what I am saying is that I have managed to mount the USB drive.  I also wanted to say that when poking at the menuconfig file under the filesystem section, there does not seem to be a choice to enable or modularise a ntfs3 file system, so my original request may be a bit suss.

Also please consider just including the 3 lines when compiling kernels in future, especially for the U3 :).  Thanks again

hexdump0815, 2022-08-14:
@sirzur - the next kernel builds should have it - https://github.com/hexdump0815/kernel-config-options/commit/e27a51f91989c0c9c2d445b42118334011870279 :)

hexdump0815, 2022-08-14:
@sirzur - a new kernel with those options enabled is now available here: https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases/tag/5.19.1-stb-exy%2B

i did not test it yet - it would be nice if you could give it a try and let me know if it works well for you

best wishes and have a good week - hexdump

hexdump0815, 2022-08-15:
the procedure would be:
cd /
tar xzf 5.19.1-stb-exy+.tar.gz
update-initramfs -c -k 5.19.1-stb-exy+ (only required for setup with encrypted rootfs, but good practice anyway)

now edit /boot/extlinux/extlinux.conf and add a new menu entry for the new kernel (without and typos) and point the DEFAULT entry on top to the label of that new entry ... then reboot and hope :) ... switching back to the old kernel goes then by simply switching the DEFAULT entry back to its label

sirzur, 2022-08-16:
OK, thanks for the detailed instructions.  I will try that after I sort out why my btrfs rootpart file system is getting corrupted all the time.  Might be time to buy another microsd card reader

sirzur, 2022-08-16:
OK, I managed to boot into the new kernel with your instructions.  But the interesting thing is that in the extlinux.conf file, to get the kernel to boot, I had to shift the section with the new kernel immediately after the MENU TITLE.  It was as if the DEFAULT switch was not registered.  But I checked and rechecked the label, so I don't know what the issue is.  In any case, I have booted up into the latest and fuse works.  I will continue to watch it over the next few days to see whether it remains stable

hexdump0815, 2022-08-16:
thanks for testing - the DEFAULT entry should work - maybe really an editor issue?

sirzur, 2022-08-17:
Yeah, maybe it is an editor issue, although  find winscp pretty good as an editor.  I will play around with nano over the next couple of days and see whether I can figure it out

sirzur, 2022-08-18:
Well.  Call me blind.  It turns out I made a mistake despite all the previous checks.  I decided to label do the menu labels with version numbers, with a "v" in front to denote the kernel.  The DEFAULT had a capital "V" while the LABELS had a small "v".  Corrected that and it all works.  All good.  Big thumbsup.  Been running for a couple of days now with the new kernel and seems stable so far - which is more than can be said for my eyesight