velvet-os/imagebuilder #67: SNOW install

betolley, 2022-08-10:
I have a xe303C12 SNOW.  I can boot several other images but I have been trying to install the ubuntu and several of the images from this project.  I really want ubuntu.  I get a U-Boot prompt and it appears to try to boot from the mmcblk0 instead of the usb.  I tried to boot usb using usbboot usb 0:1  I tried 0:1->0:4 and it wouldn't boot with any of them.  So I installed a image to the mmcblk0 that worked from another project then DD'd the Ubuntu install from my USB drive and it boots and works.  Any idea what is up the the U-Boot?  All of the images from the project do that to me.  I also tried 5 different USB drives and all do the same thing when I boot them.

hexdump0815, 2022-08-12:
@betolley - i think the chainloaded u-boot used by the images does not support usb and only sd card. i guess the other images which worked for you are not using u-boot, but boot a kpart style fit kernel directly and that why it works for them. it should most probably work by dd'ing the vmlinux.kpart to the first partition of the usb stick after dd'ing the image to the usb stick.

betolley, 2022-08-12:
Ty for telling why it was happening.  I ended up DDing from crosh/shell and it is running great.  Two questions.  When I run Ubuntu I can't install Chrome because no support for squashfs/snap in the kernel so I tried Debian and it works great except the touch pad doesn't work in Debian but does in ubuntu but I can install chrome.  Mouse works fine on Debian.  It is the version of the xe303c12 I thing rev 5.  It has a Atmel maxtouch touchpad.  Anything I can do to get it working?

hexdump0815, 2022-08-14:
@betolley - which image did you use exactly in the end? is your snow rev4 or rev5? - if not sure: what is the manufacturing date printed on the back side label?

betolley, 2022-08-15:
https://github.com/hexdump0815/imagebuilder/releases/tag/220619-01
Bullseye from here.  It works great except no sound from speaker.  Headphones work great.
Chromebook is rev 5..

hexdump0815, 2022-08-15:
i do not own a rev5 snow, but a friend of mine has one - will try to test it there, but it might take some time as he is on vacation now ...

betolley, 2022-08-16:
Let me know if you want me to test anything.

hexdump0815, 2022-08-17:
audio is hard to test remotely

hexdump0815, 2022-09-16:
@betolley - today i had the chance to test my latest bullseye image with a snow rev5 of a friend and sound works fine - did you follow this line from the chromebook snow readme: "after first boot please run /scripts/fix-snow-audio.sh as root to fix the ucm audio config files for the corresponding system and reboot"?

jenneron, 2022-11-12:
Please try this u-boot tree https://gitlab.com/exynos5-mainline/u-boot

hexdump0815, 2022-11-12:
@jenneron - thanks for the info, noted it down for the next round of snow work