velvet-os/imagebuilder #68: chromebook_trogdor: system status: homestar (lenovo duet 5 chromebook)

hexdump0815, 2022-08-14:
## notes on homestar running 5.19.1:

here is list of working and non working features on the lenovo duet 5 chromebook.

### working
- wifi
- builtin display
- energy: battery status
- usb c for basic usb functionality
- gpu opengl: via mesa freedreno (opengl 3.3)

### untested
- bluetooth
- energy: dpms, suspend/resume
- external display
- hw video encode/decode: most probably not working yet

### broken
- sound: i got audio out of both the speakers and out of connected headphones using the lenovo adapter in a proof of concept way, but there is still work to be done for a proper out of the box ucm and pulseaudio setup
- webcams: not working yet, unclear if they can be made working as they are just simple cameras and no usb video class cameras like on other chromebooks (not deeply checked yet)

### installation
coming soon

### problems
none so far

bsammon, 2022-11-30:
Just got a Duet 5 a few days ago, and made a boot-thumbdrive out of your image yesterday.  Boots up and so-far-so-good.

hexdump0815, 2022-11-30:
@bsammon - thanks a lot for the feedback ... i plan to look ater newer kernels and images a bit after linux kernel v6.1 is out - looks like there might be some patches required to make it work nicely on snapdragon 7c from what i read on the irc channel ... i hope to have something ready around new year maybe

bsammon, 2022-12-25:
Discovery: trying to modify the partition table of the boot-thumbdrive with parted (most attempts were actually the gparted frontend, but IIRC one attempt was commandline parted) renders the thumbdrive unbootable.  Each attempt was tried on a recently-created and recently-working thumbdrive setup.  Modifying the partition table with fdisk was successful, however.

bsammon, 2022-12-25:
Since I can't really find a relevant forum (or mailing list/etc) for discussing this (non-ChromeOS/crostini/crouton-linux-on-Duet), I'm thinking of posting my discoveries/questions here on an ongoing basis.  Let me know if you'd like me to stop or set limits or something.

hexdump0815, 2022-12-26:
@bsammon - that is the idea of those per device threads and issues here, so you are welcome to continue giving feedback here about what you notice ... regarding your parted issue you might have a look at this: https://oftc.irclog.whitequark.org/linux-cros-arm/2022-12-13#31712113 ... if you just want to extend the filesystem to full size, for that there is a script /scripts/extend-rootfs.sh to be run as root for exactly that

bsammon, 2022-12-27:
Yeah, I didn't want to grow the filesystem to fill the drive, just expand it a few gigs, leaving space for other partitions.  I did experiment with customizing the extend-rootfs script; don't remember if that's the approach that I ended up using.

While the parted information in that IRC thread was confusing (What's the "-f" argument to parted?  Undocumented?), the [Kali Linux forum thread](https://forums.kali.org/showthread.php?36002) linked from there had some interesting info.

My current thinking:  If you're going to use parted, learn how to use cgpt first.

hexdump0815, 2022-12-28:
@bsammon - i plan to maybe switch to a more simply parted approach at some point in time when i find some time to change and test it ... let me know in case you find any working solution

bsammon, 2022-12-28:
So... (a question that came to mind while commenting in another issue) ... which kernel source do I download if I want to compile a kernel for this device?

hexdump0815, 2022-12-28:
@bsammon - answered here: https://github.com/hexdump0815/imagebuilder/issues/104#issuecomment-1366862821

bsammon, 2022-12-28:
So, in this case, the relevant document would be https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_trogdor/readme.md ?
Looking at that document, it seems to be more of a general "how to build a kernel from git sources" document, and not specific information about which sources should be used for which hardware.
Or do I just want to use Linus's kernel tree?

...okay, poking around a bit more, maybe what I'm supposed to do is go to https://github.com/hexdump0815/imagebuilder/releases/tag/220814-01, and download the "source code" there?

hexdump0815, 2022-12-28:
@bsammon - its all described in the three documents i have linked in the other issue - please have a look at them ... i spent quite a bit of time to write them, so you should also invest that even less time it needs to read them :)

bsammon, 2022-12-28:
I've read them.  Either I'm confused, or they don't answer my questions (or both).  And it's probably compounded by me not expressing my questions clearly.

bsammon, 2022-12-28:
It seems like there's a lot of "You've never compiled a kernel before? Here's what you need to know" there, and I (as someone who's compiled a kernel before) am having problems finding the one or two lines that tell me what specifically to do to get the correct kernel source.  That doesn't mean it's not in there, just that I'm having trouble finding it.

bsammon, 2022-12-28:
Alternatively, I'm thinking, rather than looking for the answer in the documentation, I'll try reading the source to the scripts you've created, and maybe that'll tell me what source files get downloaded.

bsammon, 2022-12-28:
Digging and digging, I'm looking for information about kernel patches that you use, or a special git repository with trogdor-specific kernel support, and all I've found is that the version number is 5.19.1, and https://github.com/hexdump0815/kernel-extra-patches , which seems to be (optional-i.e. I can build a bootable, usable kernel without them) efficiency patches.

I'm starting to find myself coming to the conclusion that vanilla 5.19.1 kernel sourcecode is all that I need, with no special patches, just the right .config file.
Or am I missing something?

hexdump0815, 2022-12-28:
@bsammon - i just added some extra info to the kernel deb package issue - lets continue the kernel build discussion there

JamesLastBolero, 2023-03-16:
Image 230218-03 - debian bookworm runs fine on Homestar Rev3.
Working are Graphics, attached keyboard and touchpad, wifi.
Not working is sound (as noted).
Did an install to internal emmc and wiped chromeos (yeah), boots fine.
Did an update using your kernel config to linux 6.2.6, installed depthcharge package for automatic kernel and initrd upgrades, worked fine after deleting plymouth (image with graphical boo was to big to fit into the kernel partition).
Any suggestions where to look to get sound working ?
Thank you
James

bsammon, 2023-03-16:
> Did an install to internal emmc and wiped chromeos (yeah), boots fine.

Did you have/make a script/installer that does this, or did you do it manually?  Is there any documentation that you found about this?  I'm considering doing this myself.

> Any suggestions where to look to get sound working ?

I found https://github.com/hexdump0815/imagebuilder/blob/main/doc/alsa-ucm-pulseaudio-hacking.txt a few days ago.. haven't gotten around to trying it (or even understanding it) yet.
I'd be interested in hearing about what you've tried, may save me some time.
Also, [Cadmium](https://github.com/Maccraft123/Cadmium) claims to have sound working on the Duet 5, but it wouldn't boot (or it booted and didn't display anything) the time I tried it.  I gave up on Cadmium pretty quickly (I'm meaning to get back to it sometime soon)

SiFan229, 2023-08-20:
Excuse me, I found that the touch screen seems to be unusable during use

hexdump0815, 2023-08-20:
@SiFan229 - i must admit that i never really used the touchscreen and usually use such convertible devices usually just in laptop mode - maybe i should have a closer look at it ... maybe it is a similar problem as described here - https://github.com/hexdump0815/imagebuilder/issues/44 ?

raremotas, 2023-09-30:
I was able to verify bluetooth works.

Am looking to test things for the touch screen and audio support! I will report back anything else I can figure out.

bsammon, 2023-11-28:
I've found https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/lts/config.cbq-6.1.51-stb-cbq
Seems like a reasonable candidate.

hexdump0815, 2023-11-28:
@bsammon - in case you prefer a precompiled kernel, then you can have a look at https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases/tag/6.1.51-stb-cbq and https://github.com/hexdump0815/imagebuilder/blob/main/doc/installing-a-newer-kernel.md ... but compiling a kernel yourself is the better option anyway - for that https://github.com/hexdump0815/imagebuilder/blob/main/doc/building-own-kernels.md might be useful ...

bsammon, 2023-11-28:
Yeah, I may want that precompiled kernel for comparison's sake if the one I build doesn't work.

Since you didn't comment on my choice of config file, I'm assuming you have no objections or reservations.

bsammon, 2023-11-28:
Also, is https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/blob/lts/readme.cbq the latest version of the process you use to build the FIT images?

hexdump0815, 2023-11-29:
yes - for v6.1 it is and there is another branch with some initial (not really well tested yet) v6.6 experiments

hexdump0815, 2024-01-28:
@JamesLastBolero @bsammon @raremotas - please have a look at https://github.com/hexdump0815/imagebuilder/issues/44 as well, which now also contains some information useful for homestar and sound maybe worth trying.

LukIsHere, 2024-06-14:
[image]
homestar seams to only work with one external monitor instead of 2
[image]
not sure why

LukIsHere, 2024-06-17:
## [USI chromebook stylus](https://www.lenovo.com/us/en/p/accessories-and-software/stylus-pens-and-supplies/pens/4x80z49662) works fine
_Sidenote. it requires some obscure AAAA (<-it is not a typo) batery so i would recommend buying version 2.0_
[version 2.0](https://pl.aliexpress.com/item/1005005056660027.html) (rechargeable one) not tested **yet** (should work too)
_Note. assuming you use only built in screen_
when connecting second screen it stretches across screen (hard to explain, it is an issue with most non wacom tablets afaik) but i can't imagine anyone using chromebook as drawing tablet with additional screen connected so it's fine

marcusramberg, 2024-06-22:
I've tried this on my duet 5 booting from usb, and it mostly seems to work nicely, except for wifi, I can't find the interface at all. Do I need to modprobe some extra module, or could it be that the network card has changed in my revision of the hardware? 
@JamesLastBolero you said you've installed on your emmc, were you able to disable the firmware write protection, I'm curious if it works using a suzyqable.

hexdump0815, 2024-06-22:
@marcusramberg - i have no idea why wifi is not working for you - maybe @LukIsHere has an idea as it seems to be working there i guess ... maybe https://github.com/hexdump0815/imagebuilder/issues/44 has some useful info as well? i do not have a duet 5 at hand right now to check myself ...

for emmc installation see https://github.com/hexdump0815/linux-mainline-on-arm-chromebooks and https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md

LukIsHere, 2024-06-22:
@marcusramberg 

i [don't think there is any write protection](https://www.reddit.com/r/chromeos/comments/qywwu4/anyone_here_know_where_the_readwrite_protection/) for gbb flags in this model
but just to be sure i unplugged the actual battery and set them (it likely isn't required)
[image]
i managed to install onto emmc without and issue

also wifi is working out of the box on the latest debian image
are you using ubuntu image by any chance?

marcusramberg, 2024-06-22:
> @marcusramberg
> 
> i [don't think there is any write protection](https://www.reddit.com/r/chromeos/comments/qywwu4/anyone_here_know_where_the_readwrite_protection/) for gbb flags in this model but just to be sure i unplugged the actual battery and set them (it likely isn't required) ![image](https://private-user-images.githubusercontent.com/92756992/341998984-22c14123-ceb3-4aaa-9c58-c316a59639b6.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkwOTQyMjQsIm5iZiI6MTcxOTA5MzkyNCwicGF0aCI6Ii85Mjc1Njk5Mi8zNDE5OTg5ODQtMjJjMTQxMjMtY2ViMy00YWFhLTljNTgtYzMxNmE1OTYzOWI2LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyMlQyMjA1MjRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zNTcxNmEzOWFjYWQzMjY5NzRjOTQ2OTFiMGU0NzU0Y2Y3MDNkY2EwM2JlOWJhOTBiNDQ4Njg3Y2E4YTNkMjgwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XBuk1mPNhJbgBvB5e45T_74p9z3CdumpxZf7x_FGuPg) i managed to install onto emmc without any issue
> 
> also wifi is working out of the box on the latest debian image are you using ubuntu image by any chance?

Thanks, that makes sense. I am indeed using the ubuntu image. Will give the Debian one a go then :)

marcusramberg, 2024-06-22:
Can confirm that the wifi on the debian bookwork image worked out of the box :) Will probably do the battery hack just in case and flash to the emmc then. Thanks again.

LukIsHere, 2024-06-23:
@marcusramberg 
there is no write protection stoping you from installing to emmc
there sometimes is one when setting [gbb flags](https://wiki.postmarketos.org/wiki/Category:ChromeOS#GBB_flags) (but doesn't seams to be the case for duet 5)

@hexdump0815 the wifi problem seams to be ubuntu image issue

marcusramberg, 2024-06-23:
> @marcusramberg there is no write protection stoping you from installing to emmc there sometimes is one when setting [gbb flags](https://wiki.postmarketos.org/wiki/Category:ChromeOS#GBB_flags) (but doesn't seams to be the case for duet 5)
> 
> @hexdump0815 the wifi problem seams to be ubuntu image issue

Yepp, I meant to set the gbb flags, as I got the impression you can risk ending in a permanent brick situation without setting them. I just tried and write protection is enabled when the battery is connected, so I guess I'd have to disconnect the battery like you did. Just have to find my pry tool to get the display out 😅

LukIsHere, 2024-06-23:
@marcusramberg 
just in case here is a hardware mainteanance mamual so u don't break your device doing it
https://download.lenovo.com/consumer/mobiles_pub/chromebook_duet_5_hmm.pdf

marcusramberg, 2024-06-23:
> @marcusramberg 
> just in case here is a hardware mainteanance mamual so u don't break your device doing it
> https://download.lenovo.com/consumer/mobiles_pub/chromebook_duet_5_hmm.pdf

Thanks. Mission already successful after following https://youtu.be/WZ6yCSes5HI?si=_OFUgFBLrZHNN5Wg 😀

marcusramberg, 2024-06-24:
I've flashed the internal drive now (using the rsync procedure as my usb stick already had some config, and with the newer kernel touch works good as well. I guess sound is the only thing I'd like to get going if possible. Tried to look into some of the suggestions in the other threads, but it seems like the internal sound devices aren't detected at all right now? At least for me, I can only see the dummy output. Can confirm external usb sound devices are detected fine tho, and I guess bluetooth would work as well.

LukIsHere, 2024-06-24:
@marcusramberg there are some solutions higher up in this issue
the quickest is to go to [volume controll app](https://packages.debian.org/sid/pavucontrol)
[image]
and in the last section switch to advanced or sth
then switching the output to nr.2

marcusramberg, 2024-06-24:
Perfect, thanks. This is such a nice little linux tablet :)

LukIsHere, 2024-07-23:
@marcusramberg btw 
# if u remove the glue that held ur battery in place
ur device is on drop away from being a **brick**
[image]

marcusramberg, 2024-07-24:
@LukIsHere hmm, no I didn't remove the battery, just unplugged it, and reattached the fastener after plugging it back in. Thanks for the heads-up tho :)

LukIsHere, 2024-07-24:
no problem i did it for whatever reason and now i have to get a new motherboard

it's better to use suzyqcable for this stuff instead of opening that device but didn't have any at that time

hexdump0815, 2024-07-24:
maybe ask in one of those phone repair shops - they sometimes can do wonders around hardware ... or watch out for a used duet 5 with a broken screen - they usually go for much lower - most of my tablets are of this kind for that reason :)