velvet-os/imagebuilder #71: chromebook_kukui: status: kakadu (asus cm3 tablet)

hexdump0815, 2022-08-17:
## notes on kakadu running 5.19.1:

here is list of working and non working features on the asus cm3 tablet (kakadu).

### working
- wifi
- bluetooth
- builtin display
- energy: battery status, dpms, suspend/resume
- usb c for basic usb functionality
- gpu opengl: via mesa panfrost (opengl 3.1)
- sound: only speaker tested so far

### untested
- hw video encode/decode: most probably not working yet

### broken
- sound: not sure about the audio jack and the mic options yet - simply not yet tested
- external display: not working yet
- webcams: not working yet, unclear if they can be made working as they are just simple cameras and no usb video class cameras like on other chromebooks

### installation
coming soon

### problems
none so far

lentas1, 2024-04-04:
how to install

SnowFortIsTheBestProb, 2025-02-21:
can you give the file so i can try it myself (i own kakadu)

thenameisluk, 2025-02-21:
@lentas1 on the [main readme](https://github.com/thenameisluk/imagebuilder) there is big blue text that says [Getting started](https://github.com/thenameisluk/imagebuilder/blob/main/doc/first-steps.md)

you click it and go to the installation/booting section and chose [chromebooks](https://github.com/thenameisluk/imagebuilder/blob/main/doc/chromebooks/readme.md)

you scroll down and find all available installation methods

most common one [basic installation](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/installation/basic-installation.md)

thenameisluk, 2025-02-21:
@lentas1 
and since i see you haven't bother to look at docs at all

i urge you before you blindly follow the installation to read pages you are going though

otherwise you might skip some important information like [setting gbb flags](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/setting_gbb_flags.md) 

which are improtant to be set before proceeding with an installation (or your system might stop booting and you will have to wipe your installation to restore it)