velvet-os/imagebuilder #72: Odroid U3: NetworkManager error message flood in syslog

sirzur, 2022-08-19:
I don't know whether this is related to the kernel or the image, but the Odroid U3 syslog is flooded with the following messages, even when the device is not doing any net related activities.  It happens on both the 5.18.1 and 5.19.1 kernel.  The messages repeat every 4-5 seconds and now the syslog is about 18Mb

Aug 19 05:22:52 odroid NetworkManager[521]: <warn>  [1660850572.9341] ipv6ll[5380326cf862e163,ifindex=3]: changed: no IPv6 link local address to retry after Duplicate Address Detection failures (back off)
Aug 19 05:23:02 odroid NetworkManager[521]: <warn>  [1660850582.9388] platform-linux: do-add-ip6-address[3: fe80::3b9a:3586:5b63:de72]: failure 95 (Operation not supported)
Aug 19 05:23:04 odroid NetworkManager[521]: <warn>  [1660850584.9419] platform-linux: do-add-ip6-address[3: fe80::fa12:5956:320f:7ab1]: failure 95 (Operation not supported)
Aug 19 05:23:06 odroid NetworkManager[521]: <warn>  [1660850586.9425] platform-linux: do-add-ip6-address[3: fe80::31b1:218a:527b:c75b]: failure 95 (Operation not supported)
Aug 19 05:23:08 odroid NetworkManager[521]: <warn>  [1660850588.9453] platform-linux: do-add-ip6-address[3: fe80::66c1:3c66:cc23:ce5d]: failure 95 (Operation not supported)
Aug 19 05:23:10 odroid NetworkManager[521]: <warn>  [1660850590.9472] platform-linux: do-add-ip6-address[3: fe80::d407:2124:3b6d:3d5a]: failure 95 (Operation not supported)
Aug 19 05:23:12 odroid NetworkManager[521]: <warn>  [1660850592.9498] platform-linux: do-add-ip6-address[3: fe80::9559:d5a2:57e1:c2f9]: failure 95 (Operation not supported)

I noted that ipv6 is disabled in the /boot/extlinux/extlinux.conf file. 

I have also tried the following to see whether it helps.  On desktop, navigate to NetworkManager and edit the existing connection to switch ip6 from automatic to disable and saved.  The setting does not stick between reboots and in any case does not make any difference

Used the following commands in terminal to try to disable ipv6
sysctl -w net.ipv6.conf.all.disable_ipv6=1
sysctl -w net.ipv6.conf.default.disable_ipv6=1
sysctl -w net.ipv6.conf.lo.disable_ipv6=1

But messages back says
sysctl: cannot stat /proc/sys/net/ipv6/conf/all/disable_ipv6: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv6/conf/default/disable_ipv6: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv6/conf/lo/disable_ipv6: No such file or directory

Used nmcli connection modify 681b428f-beaf-8932-dce4-687ed5bae28e ipv6.method "disabled"
But with message
Error: Failed to modify connection 'Ifupdown (eth0)': failed to update connection: settings plugin does not support modifying connections

So my question is whether this is related to the kernel /image and has anyone encountered this

sirzur, 2022-08-19:
I decided that disabling ipV6 was the same as hiding whatever was causing the warnings, so I decided that I might as well hide the messages.  I added a section to /etc/NetworkManager/NetworkManager.conf

[logging]
level=ERR

And now the sylog is not flooded with these messages and I can at least see what other issues there are - I was trying to see why a service stopped by itself suddenly from the syslog messages when I first encountered the messages

hexdump0815, 2022-08-20:
@sirzur - ipv6 is not disabled to hide errors, i just disable it by default as it geve me trouble in some situations ... but due to it being disabled by default this might hide some potential problems ... please let me know in case you find out anything about those warnings. is this bullseye or jammy?

sirzur, 2022-08-20:
Funny though, despite it being disabled by default, there is still all those IPv6 messages flooding the syslog.  I actually wonder what Hardkernel did to fix those errors.  Certainly their later kernels with XU4 does not have IPv6 disabled and the errors are not there.  But that might be because of different CPU.  And it is in jammy that the errors are seen.  I will sporadically see whether I can find out more about the errors and if I do find out anything, I will keep you informed.  Thanks

p0358, 2023-03-09:
It might be the fact that it's disabled causing these errors, not the other way around