velvet-os/imagebuilder #74: Is it possible to install to emmc?

cdd100, 2022-08-29:
Not a big problem but I was wondering if I can install to emmc. Thanks!

hexdump0815, 2022-08-30:
on which system?

cdd100, 2022-08-30:
Lenovo c330. I wrote to a usb and it worked so I then tried using etcher to write to the emmc and it loaded from emmc once now it will load to emmc if I plug in my USB and it won't boot from usb.

hexdump0815, 2022-08-30:
yes, running from emmcis working very well on chromebooks - maybe have a look at https://github.com/hexdump0815/imagebuilder/blob/main/info/generic/install-to-emmc-with-luks-full-disk-encryption.txt - this is my usual workflow for installing on emmc which even includes a fully encrypted rootfs - those notes are maybe a bit hard to read as it describes the process for many different systems in parallel, but with some slow and careful reading it should be possible to follow ...

good luck and best wishes - hexdump

cdd100, 2022-08-30:
Thanks! I will give it a go and update you if it works.

cdd100, 2022-08-30:
Is it possible to flash the image to the emmc with etcher?

cdd100, 2022-08-30:
If found an arm version of etcher that loaded

hexdump0815, 2022-08-31:
there is no need for etcher if you are using it on the chromebook - just a "dd if=image of=/dev/emmcdevice bs=1024k status=progress" should do it on linux

cdd100, 2022-08-31:
I'm a beginner and don't understand. Could you explain it in more context? Thanks.

hexdump0815, 2022-08-31:
@cdd100 - maybe google a bit of information about the "dd" (disk dump) unix command - the command basically just dumps what you give it with "if" (in file) to what you give it as "of" (out file) - be very careful using it on any system where you have valuable data on any device as it blindly writes the data to the out file - this should be no problem in your chromebook situation: you want to overwrite the emmc in the chromebook anyway with linux and you run from the image on an sd-card or usk-disk which you can reimage in case you accidently overwrite it ... "image" in the above example image would be the diskimage for your system after running "gzip image.gz" to uncompress it - "/dev/emmcdevice" should be something like "/dev/mmcblk0" on the chromebook ...

but maybe also try to understand the steps described in the install-on-emmc text mentioned above (maybe with the help of some web research) as it would be the better option to install on emmc instead of just dumping the image onto it (which should work as well i think)

good luck and best wishes - hexdump

cdd100, 2022-08-31:
Is it possible to use etcher?

closed 2022-11-23 by cdd100