velvet-os/imagebuilder #77: odroid x2. install to emmc

redeemarr, 2022-09-01:
Hi!
Installed to SD card debian/ubuntu from https://github.com/hexdump0815/imagebuilder/releases/tag/220824-01 works nicely.
Then tried to just ```dd``` the image to emmc (stock one, from hardkernel), but that doesn't work, odroid lights up both blue LEDs and nothing happens.
I've learned that emmc has hardware boot partitions (which were probably used for previously installed android), so I've tried to disable them with
```
mmc bootpart enable 0 0 /dev/mmcblk1boot0
mmc bootpart enable 0 0 /dev/mmcblk1boot1
```
Still nothing, even no blue leds. Seems like odroid needs those boot partitions. But what and how should I stuff them?
Is it even possible to install ```img``` to emmc?

hexdump0815, 2022-09-01:
@redeemarr - i think this should explain how to do it: https://github.com/hexdump0815/imagebuilder/issues/1#issuecomment-665645956 - maybe best is to read the full issue and the linked pages in it

good luck and best wishes - hexdump

redeemarr, 2022-09-02:
@hexdump0815 wow! much thanks! i just ordered that emmc-sd adapter, will try later. exy-boot.dd will work for any image of yours?
dedicated thanks for your work! odroid got it's second life :)

hexdump0815, 2022-09-02:
@redeemarr - its a long time ago i built those u-boot images, but according to my old notes there is only one for all of the exy odroids and that one works for the x2 for the sd card boot i guess it should work for emmc as well - i think i never tested it on x2 myself, but i'm sure it should simply work

this bringing a second life into those old sbc's (and other devices) is one of the ideas of this project - most of them are simply still too useable to just throw them into the bin  ...

redeemarr, 2022-09-07:
well, finally got emmc-sd adapter, but i realized that it cannot be used to access emmc boot partitions since i got no native sd-slot in any my device, only usb card reader :)
so i made an experiment in few steps:
- inserted both emmc and bootable sd card in odroid
- enabled boot from sd and booted from sd
- found that there are two mmcblk devices - one for sd (mmcblk0) and another for mmc (mmcblk1)
- removed RO protection and dd'ed exy-boot.dd to corresponding boot emmc partition (mmcblk1boot0)
still no luck. what might be a reason that it doesn't boot?

hexdump0815, 2022-09-09:
@redeemarr - hm, it should work - some things which come to my mind:

- did you use the dd command as described here: https://github.com/hexdump0815/u-boot-misc/blob/master/readme.exy#L51 or here: https://github.com/hexdump0815/imagebuilder/issues/1#issuecomment-665645956 ?
- maybe there is a problem if there is still some data on the first block - "if=/dev/zero of=/dev/mmcblk0boot0 bs=512 count=1 iflag=dsync oflag=dsync status=progress" should clean the first block
- maybe the "skip=1" should not be used following point 4. from https://source.denx.de/u-boot/u-boot/-/blob/master/doc/README.odroid - i.e. no offset of one block required for the boot block on emmc - maybe the x2 does not require an offset and the u3 requires one? i think i used an offset on the u3 when i tested it and it seems to have worked for https://github.com/hexdump0815/imagebuilder/issues/1#issuecomment-665645956 as well on an u3

please let me know if you get it working somehow - good luck and best wishes - hexdump

redeemarr, 2022-09-12:
tried this way:
```
echo 0 > /sys/block/mmcblk1boot0/force_ro
dd if=/dev/zero of=/dev/mmcblk1boot0 bs=512 count=1 iflag=dsync oflag=dsync status=progress
dd if=exy-boot.dd of=/dev/mmcblk1boot0 bs=512 iflag=dsync oflag=dsync status=progress
sync # desperation
shutdown now
```
also tried to add skip=1 just in case - still no boot.
but i've noticed weird thing - it does boot from SD even when jumper is set to boot from mmc :)
if i remove an SD it doesn't boot (no blue blinks at all)

hexdump0815, 2022-09-13:
@redeemarr - i'm a bit out of ideas right now ... i'll let you know when something new comes to my mind and please keep me up to date here in case you make any progress regarding this topic

best wishes - hexdump

redeemarr, 2022-09-13:
same thing. i'm closing this ticket for now. thank you for help!

closed 2022-09-13 by redeemarr