velvet-os/imagebuilder #79: chromebook_nyan: status

hexdump0815, 2022-09-05:
## notes on nyan running 5.19.1:

here is list of working and non working features on the nyan chromebooks.

### working
- wifi
- bluetooth
- builtin display
- usb
- builtin speaker and mic
- webcam
- energy: dpms

### untested
- hw video encode/decode (most probably not working)
- external display (might work)

### broken
- energy: suspend/resume (s2idle suspend works partially, but not reliable)

### installation
coming soon

### problems
- gpu requires self built newer version of the xorg server and most proably older mesa version due to regressions in newer mesa versions

RaylynnKnight, 2022-10-10:
Thank you this appears to be functioning properly on an HP Chromebook 14 G3 (nyan-blaze).

eutampieri, 2024-06-21:
Hey! Thanks for the great work. I'm writing to report that, on nyan blaze (2GB RAM),
- HDMI works
- Audio is not working (I'd like to make it work, but I can use some pointers)
- WiFi is intermittent (I think it worked fine before running `sudo apt dist-upgrade`)

`uname -a`
```
Linux eugenio-chromebook 6.1.51-stb-cbt+ #1 SMP Tue Sep  5 16:22:44 CEST 2023 armv7l GNU/Linux
```

`lsb_release -a`
```
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

eutampieri, 2024-06-22:
[wifi_not_working.txt](https://github.com/user-attachments/files/15935431/wifi_not_working.txt)
[wifi_working.txt](https://github.com/user-attachments/files/15935432/wifi_working.txt)
Here are some dmesg logs with wifi working and not working. In ChromeOS, it always work.

hexdump0815, 2024-06-22:
i'm surprised a bit - i have a 2gb nyan here as well and audio is working (one needs to switch the output by hand properly iirc) and also wifi is working - maybe the updates overwrite the wifi firmware at some point - just try to put those files back in place and see if that helps: https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_nyan/extra-files/lib/firmware/mrvl

good luck and best wishes - hexdump

legendaryfishwastaken, 2024-06-23:
How do I install this?

eutampieri, 2024-06-24:
@legendaryfishwastaken what are you referring to? If you follow the guides on the repos, you'll download a rootfs image to extract to an SD card.

eutampieri, 2024-06-24:
```
ls /lib/firmware/mrvl/
pcie8897_uapsta.bin        sd8787_uapsta_cros.bin     sdsd8977_combo_v2.bin
pcie8997_wlan_v4.bin       sd8797_uapsta.bin          sdsd8997_combo_v4.bin
pcieuart8997_combo_v4.bin  sd8797_uapsta_cros.bin     usb8766_uapsta.bin
pcieusb8997_combo_v4.bin   sd8801_uapsta.bin          usb8797_uapsta.bin
prestera                   sd8887_uapsta.bin          usb8801_uapsta.bin
sd8688.bin                 sd8897_uapsta.bin          usb8897_uapsta.bin
sd8688_helper.bin          sd8897_uapsta_cros.bin     usbusb8997_combo_v4.bin
sd8787_uapsta.bin          sd8xxx_uapsta_cros.readme
```
Is it normal that I have so many files?

hexdump0815, 2024-06-24:
@eutampieri - that might be right, but only the ones copied in from the extra-files area will be used on the nyan ... after copying them in it might be a good idea to "apt-mark hold firmware-libertas" to avoid the firmware files getting overwritten again (maybe the copied over firmware fileswere only required for the legacy kernel initially used - i do not remember anymore completely) ... but looking at this package it seems the firmware files installed have a different name, so no risk for the added files to get overwritten ... so lets forget about all that - it seems to not be the problem on your system - which kernel version is this?

i looked at your dmesg output and "mmc2: error -110 whilst initialising SDIO card mmc2: error -110 whilst initialising SDIO card" does not look good - what do you get with "dmesg | grep -i firmware"? ... maybe it is not even firmware loading related as the above error comes (accoring to the time stamps) before the firmware loading happens in your working dmesg output ... can you please put at full non working wifi dmesg up somewhere?

Mia-Rain, 2024-08-28:
Any update on the gpu situation for this device? I recently purchased a c810 (nyan-big) and am trying to get 3D acceleration working.  I had it working before, but its been a number of years now, wondering if theres been any progress? 

From what I understand the proprietary driver blobs are known working and there is a lot of manual compiling for the nouveau driver

I also saw that there is a new nvk driver that might be a good idea to at least tinker with.

hexdump0815, 2024-08-28:
there is not much happening in this area - the legacy driver is a completely dead end, i had gpu support working partially with nouveau and mesa at some point in time (and i think i noted it down somewhere in the readme what was working) but it was not useable for anything a bit more demanding like wayland gnome etc. and it broke with a newer kernel and mesa ... nvk might be interesting, but do not expect it to just work - https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/24326 should give an idea about what to expect (search for k1 there) - so this is only something if you have the time and/or skills to really get into it ... in case you get anything working, please leave a note about it here ... good luck

Mia-Rain, 2024-08-29:
Well I had it working, but now glamor is causing an instant segfault

hexdump0815, 2024-08-30:
yes, one can get somewhere but not that far usually - to make that useable, quite a bit of work would be required

Mia-Rain, 2024-09-10:
It took several days, but after building a system from the ground up (using kiss) with the xorg and mesa changes I've got it working mostly perfect.

I'm still having issues with compiling things but that's unrelated, xorg is working perfectly with acceleration and 0 graphical glitches

it also seems to be using the tegra driver rather than the nouveau driver but idk if there's a difference (that's what glxinfo says)

needs some additional testing as all i've got compiled rn is icewm and basics but glxgears runs great, system is much faster than when using fbdev and aside from screen tearing since I don't have a compositor there's no graphical bugs

something like a gentoo system with a custom repo to override mesa and xorg would probably be the most ideal system

hexdump0815, 2024-09-10:
@Mia-Rain - that sounds very good - it would be nice if you could write down the steps you did to get it working here or somewhere and link it here

Mia-Rain, 2024-09-10:
The basics are building a fresh os image using the boot partition from one of your released builds and rewriting the rootfs with a stage 3 gentoo tarball, adding back in the modules and firmware from said release build, I've had trouble getting a kernel to compile and work and reusing yours simplified the process

I'll publish a custom repo with setup info later

Mia-Rain, 2024-09-15:
I've got a system built but i think it's possible that patches aren't needed, the tegra gallium driver isn't packaged by anything but it seems to work perfectly

I'm going to try building latest mesa and see if it works

Mia-Rain, 2024-09-16:
I managed to get it build and work on mesa 24 without any patches, although i only get 17 fps with glxgears, glxinfo shows it's using the tegra accelerated driver

it's possible that the latest mesa has moved the tegra driver to kwsrast as i saw in a pull request, as the cpu usage spikes when running glxgears

Befaro2063, 2024-09-16:
Can you make an Image?

Befaro2063, 2024-09-16:
> I managed to get it build and work on mesa 24 without any patches, although i only get 17 fps with glxgears, glxinfo shows it's using the tegra accelerated driver
>
> it's possible that the latest mesa [...]

can you provide us with an image or instructions? so we can test it?

hexdump0815, 2024-09-16:
it sounds a bit like they switched the tegra driver to use software rendering then, so i guess (if this is really the case) it might be even worse than mesas default sw renderer llvmpipe in the sense of less performance and more cpu usage ... but lets see, maybe its not that bad

hexdump0815, 2024-09-16:
just as a reference - it looks like this is an old and long discussion about what to do with tegra in mesa: https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/2960 ... looks like its still there in the current tree though ...

Mia-Rain, 2024-09-17:
I have everything working now, but i can't manage to get sound working.
I'm using pulse audio and i added the ucm2 files and /etc/asound.conf

pactl says the sink is GoogleNyanBig but still no output

hexdump0815, 2024-09-17:
that sounds good - do you have real gpu accel working now, i.e. relatively low cpu usage during opengl activity? ... for sound it might be required to enable some output by hand - i do not remember it in detail right now but might be that there was something like this ... i have not time to look into this myself right now, but i'm sure we will get this working as well as sound definitely was working on nyan to some degree

Mia-Rain, 2024-09-17:
It seems to be actually accelerated, glxgears uses abt 13% on the cou but glxinfo says it's accelerated and it's fast enough, fbdev might just be better tho

hexdump0815, 2024-09-18:
just for reference: this is the glmark2 output of the old hacked xorg and mesa where gpu accel works to some degree (with some bugs and not really useable with wayland at all):
```
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      NVIDIA
    GL_RENDERER:    tegra
    GL_VERSION:     4.3 (Compatibility Profile) Mesa 21.0.3 (git-091978e0ec)
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   800x600 windowed
=======================================================
[build] use-vbo=false: FPS: 358 FrameTime: 2.794 ms
[build] use-vbo=true: FPS: 345 FrameTime: 2.903 ms
[texture] texture-filter=nearest: FPS: 328 FrameTime: 3.057 ms
[texture] texture-filter=linear: FPS: 380 FrameTime: 2.638 ms
[texture] texture-filter=mipmap: FPS: 382 FrameTime: 2.619 ms
[shading] shading=gouraud: FPS: 386 FrameTime: 2.593 ms
[shading] shading=blinn-phong-inf: FPS: 333 FrameTime: 3.011 ms
[shading] shading=phong: FPS: 341 FrameTime: 2.937 ms
[shading] shading=cel: FPS: 380 FrameTime: 2.635 ms
[bump] bump-render=high-poly: FPS: 307 FrameTime: 3.265 ms
[bump] bump-render=normals: FPS: 358 FrameTime: 2.798 ms
[bump] bump-render=height: FPS: 330 FrameTime: 3.030 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 294 FrameTime: 3.411 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 216 FrameTime: 4.637 ms
[pulsar] light=false:quads=5:texture=false: FPS: 324 FrameTime: 3.091 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 222 FrameTime: 4.523 ms
[desktop] effect=shadow:windows=4: FPS: 277 FrameTime: 3.617 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 145 FrameTime: 6.934 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 158 FrameTime: 6.360 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 66 FrameTime: 15.185 ms
[ideas] speed=duration: FPS: 398 FrameTime: 2.515 ms
[jellyfish] <default>: FPS: 256 FrameTime: 3.918 ms
[terrain] <default>: FPS: 41 FrameTime: 24.893 ms
[shadow] <default>: FPS: 257 FrameTime: 3.903 ms
[refract] <default>: FPS: 65 FrameTime: 15.458 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 400 FrameTime: 2.502 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 391 FrameTime: 2.559 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 423 FrameTime: 2.369 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 326 FrameTime: 3.068 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 311 FrameTime: 3.223 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 306 FrameTime: 3.270 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 315 FrameTime: 3.178 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 316 FrameTime: 3.174 ms
=======================================================
                                  glmark2 Score: 294 
=======================================================
```
so in case gpu accel is working you should get somewhat similar values i guess

Mia-Rain, 2024-09-19:
here is my glmark2 output
```
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      NVIDIA
    GL_RENDERER:    tegra
    GL_VERSION:     4.3 (Compatibility Profile) Mesa 21.0.3
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   800x600 windowed
=======================================================
[build] use-vbo=false: FPS: 132 FrameTime: 7.613 ms
[build] use-vbo=true: FPS: 148 FrameTime: 6.760 ms
[texture] texture-filter=nearest: FPS: 141 FrameTime: 7.095 ms
[texture] texture-filter=linear: FPS: 139 FrameTime: 7.232 ms
[texture] texture-filter=mipmap: FPS: 140 FrameTime: 7.185 ms
[shading] shading=gouraud: FPS: 134 FrameTime: 7.511 ms
[shading] shading=blinn-phong-inf: FPS: 134 FrameTime: 7.464 ms
[shading] shading=phong: FPS: 128 FrameTime: 7.820 ms
[shading] shading=cel: FPS: 125 FrameTime: 8.039 ms
[bump] bump-render=high-poly: FPS: 88 FrameTime: 11.366 ms
[bump] bump-render=normals: FPS: 151 FrameTime: 6.630 ms
[bump] bump-render=height: FPS: 144 FrameTime: 6.985 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 85 FrameTime: 11.776 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 52 FrameTime: 19.367 ms
[pulsar] light=false:quads=5:texture=false: FPS: 127 FrameTime: 7.898 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 52 FrameTime: 19.402 ms
[desktop] effect=shadow:windows=4: FPS: 91 FrameTime: 11.025 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 98 FrameTime: 10.208 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 112 FrameTime: 8.942 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 93 FrameTime: 10.804 ms
[ideas] speed=duration: FPS: 120 FrameTime: 8.375 ms
[jellyfish] <default>: FPS: 73 FrameTime: 13.866 ms
[terrain] <default>: FPS: 11 FrameTime: 96.648 ms
[shadow] <default>: FPS: 69 FrameTime: 14.686 ms
[refract] <default>: FPS: 19 FrameTime: 53.662 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 146 FrameTime: 6.858 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 137 FrameTime: 7.334 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 147 FrameTime: 6.839 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 145 FrameTime: 6.922 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 135 FrameTime: 7.438 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 147 FrameTime: 6.829 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 146 FrameTime: 6.864 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 133 FrameTime: 7.532 ms
=======================================================
                                  glmark2 Score: 112 
=======================================================
```

this looks far different... My xorg-server is at the 7e142cb2a848acb6af986fa91d254d4c23963b24 commit and I'm using the latest libdrm

really no idea why its so different

Mia-Rain, 2024-09-19:
here is also my xorg log, maybe it will be of some use
```
[    10.349] 
This is a pre-release version of the X server from The X.Org Foundation.
It is not supported in any way.
Bugs may be filed in the bugzilla at http://bugs.freedesktop.org/.
Select the "xorg" product for bugs you find in this release.
Before reporting bugs in pre-release versions please check the
latest version in the X.Org Foundation git repository.
See http://wiki.x.org/wiki/GitPage for git access instructions.
[    10.349] 
X.Org X Server 1.20.99.1
X Protocol Version 11, Revision 0
[    10.349] Current Operating System: Linux localhost 6.1.51-stb-cbt+ #1 SMP Tue Sep  5 16:22:44 CEST 2023 armv7l
[    10.349] Kernel command line: console=tty1 root=LABEL=rootpart ro rootwait net.ifnames=0 ipv6.disable=1 fsck.repair=yes noresume no_console_suspend apparmor=0 systemd.unified_cgroup_hierarchy=0
[    10.349]  
[    10.349] Current version of pixman: 0.43.4
[    10.349] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    10.349] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    10.349] (==) Log file: "/var/log/Xorg.0.log", Time: Thu Sep 19 12:31:00 2024
[    10.351] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    10.352] (==) No Layout section.  Using the first Screen section.
[    10.352] (==) No screen section available. Using defaults.
[    10.352] (**) |-->Screen "Default Screen Section" (0)
[    10.352] (**) |   |-->Monitor "<default monitor>"
[    10.355] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    10.355] (==) Automatically adding devices
[    10.355] (==) Automatically enabling devices
[    10.355] (==) Automatically adding GPU devices
[    10.355] (==) Automatically binding GPU devices
[    10.355] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    10.356] (WW) The directory "/usr/share/fonts/TTF/" does not exist.
[    10.356] 	Entry deleted from font path.
[    10.356] (WW) The directory "/usr/share/fonts/OTF/" does not exist.
[    10.356] 	Entry deleted from font path.
[    10.356] (WW) The directory "/usr/share/fonts/Type1/" does not exist.
[    10.356] 	Entry deleted from font path.
[    10.356] (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/100dpi/".
[    10.356] 	Entry deleted from font path.
[    10.356] 	(Run 'mkfontdir' on "/usr/share/fonts/100dpi/").
[    10.359] (==) FontPath set to:
	/usr/share/fonts/misc/,
	/usr/share/fonts/75dpi/
[    10.359] (==) ModulePath set to "/usr/lib/xorg/modules"
[    10.359] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    10.359] (II) Loader magic: 0x68dff0
[    10.359] (II) Module ABI versions:
[    10.359] 	X.Org ANSI C Emulation: 0.4
[    10.359] 	X.Org Video Driver: 25.2
[    10.359] 	X.Org XInput driver : 24.3
[    10.359] 	X.Org Server Extension : 10.0
[    10.360] (++) using VT number 7

[    10.360] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    10.363] (II) xfree86: Adding drm device (/dev/dri/card0)
[    10.363] (II) Platform probe for /sys/devices/soc0/50000000.host1x/drm/drm/card0
[    10.380] (II) xfree86: Adding drm device (/dev/dri/card1)
[    10.381] (II) Platform probe for /sys/devices/soc0/57000000.gpu/drm/card1
[    10.381] (II) no primary bus or device found
[    10.381] 	falling back to /sys/devices/soc0/50000000.host1x/drm/drm/card0
[    10.384] (II) LoadModule: "glx"
[    10.385] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    10.435] (II) Module glx: vendor="X.Org Foundation"
[    10.435] 	compiled for 1.20.99.1, module version = 1.0.0
[    10.435] 	ABI class: X.Org Server Extension, version 10.0
[... 226 lines cut ...]
[    16.744] (II) modeset(0): Disabling kernel dirty updates, not required.
[    24.758] (II) modeset(0): EDID vendor "AUO", prod id 4397
[    24.758] (II) modeset(0): Printing DDC gathered Modelines:
[    24.758] (II) modeset(0): Modeline "1920x1080"x0.0  142.00  1920 2028 2076 2100  1080 1090 1100 1126 -hsync -vsync (67.6 kHz eP)
[    24.807] (II) modeset(0): EDID vendor "AUO", prod id 4397
[    24.807] (II) modeset(0): Printing DDC gathered Modelines:
[    24.807] (II) modeset(0): Modeline "1920x1080"x0.0  142.00  1920 2028 2076 2100  1080 1090 1100 1126 -hsync -vsync (67.6 kHz eP)
[    24.824] (II) modeset(0): EDID vendor "AUO", prod id 4397
[    24.824] (II) modeset(0): Printing DDC gathered Modelines:
[    24.824] (II) modeset(0): Modeline "1920x1080"x0.0  142.00  1920 2028 2076 2100  1080 1090 1100 1126 -hsync -vsync (67.6 kHz eP)
[    55.694] (EE) event2  - Elan Touchpad: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.26.1/touchpad-jumping-cursors.html for details
[    66.010] (EE) event2  - Elan Touchpad: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.26.1/touchpad-jumping-cursors.html for details
[   150.584] (EE) event2  - Elan Touchpad: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.26.1/touchpad-jumping-cursors.html for details
[... cut ...]
[   492.407] (EE) event2  - Elan Touchpad: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.26.1/touchpad-jumping-cursors.html for details
[   492.407] (EE) event2  - Elan Touchpad: WARNING: log rate limit exceeded (5 msgs per 24h). Discarding future messages.
[   646.748] (II) modeset(0): EDID vendor "AUO", prod id 4397
[   646.748] (II) modeset(0): Printing DDC gathered Modelines:
[   646.748] (II) modeset(0): Modeline "1920x1080"x0.0  142.00  1920 2028 2076 2100  1080 1090 1100 1126 -hsync -vsync (67.6 kHz eP)
```

hexdump0815, 2024-09-19:
ok - then its about the same setup i also used in the past and i think its the only setup to get at least partial gpu support ... i was hoping that maybe something got fixed in newer versions, but my impression over the last months/years is that nobody really cares much about tegra anymore - even the newer ones like the jetson nano start to bit-rot already

@Mia-Rain - what nyan version do you have: 2gb or 4gb? i'm asking as i back then ran into the situation that the above mentioned gpu accel setup worked better on the 2gb version and one of the glmark2 tests iirc even resulted in a crash on the 4gb version (but worked fine on the 2gb version) ... it would be nice if at least this problem would have gone meanwhile ...

Mia-Rain, 2024-09-21:
i'm using the 4gb nyan_big variant

from what i read the new tegra socs are based on the RTX architecture which i imagine must be harder to reverse engineer, best hope is that the open source kernel drivers Nvidia has released eventually come to support them

Mia-Rain, 2024-10-02:
I'm still working on a repo for gentoo ebuilds, various other things and some scripts. My current progress is here

[https://github.com/Mia-Rain/nyan-repo-gentoo](https://github.com/Mia-Rain/nyan-repo-gentoo)

This repo has everything to get it working with some knowledge.

Scripts will come soon to automate the process, you should wait for those unless you know how gentoo and custom ebuild repos work as the system will very quickly become very hard to maintain due to package conflicts.

I have some plans to develop systems for the postmarketos build and maybe adelie linux as it has optimizations for very light hardware.

hexdump0815, 2026-05-02:
@andremigenoi-bot - have a look at https://velvet-os.github.io/ for some documentation on how to install it ... i hope the "bot" in your name does not mean that you are some kind of agent as i prefer to have conversations with real humans in the github issues and discussions here ...

hexdump0815, 2026-08-16:
@techmaster2013 - is kitty a nyan system?

techmaster2013, 2026-08-16:
> @techmaster2013 - is kitty a nyan system?

@hexdump0815 yes, but it refused to boot numerous images

hexdump0815, 2026-08-17:
@techmaster2013 - oh - i was not even aware that a nyan chromebase existed - there is for sure no mainline support for it neither in u-boot nor the linux kernel ... the legacy chromeos kernel sources have a dtb for it: https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/release-R91-13904.B-chromeos-3.10/arch/arm/boot/dts/ - so the only thing you could do would be to build a legacy kernel and boot it directly as kaprt image and not via u-boot ... but that will not work with any modern linux with systemd due to the too old kernel version - something non systemd like devuan might work maybe ... or if you have the skills, you might try to mainline the device, but its so old and i guess not really worth it ...