velvet-os/imagebuilder #83: btrfs help

cdd100, 2022-09-09:
how to resize rootpart? everything I have tried makes it unbootable.

sirzur, 2022-09-10:
When I first encountered this, I popped the microsd card/card reader on a Ubuntu system and successfully used Diskpart to resize the btrfs partition.  One thing that I noted when I was having the same problem.  the microsd card reader was actually causing the corruption.  Bought a more expensive one, used that and the resizing corruptions went away.  Are you resizing using the actual device or using a microsd card reader?

hexdump0815, 2022-09-10:
@cdd100 - in the images there should be a script to extend the rootfs in /scripts - running it should extend the rootfs to the remaining free space on the disk

closed 2022-11-23 by cdd100