velvet-os/imagebuilder #85: Off topic but related: ARM UEFI firmware

EMLommers, 2022-09-14:
Do you know or have ARM uefi firmware for 8173 or Chromebook R13 firmware with UEFi capability?
For example to use fedora arm ISO's .
Mr. Chromebox is only supplying Intel firmwares.
It's off topic, but guess I am not the only one who would like this possiblity.
Thanks anyway.

hexdump0815, 2022-09-15:
@EMLommers - as far as i know there is nothing like that for the mt8173 and also not for any other arm chromebook - there is some experimental u-boot for rk3399 chromebooks and u-boot can be to some degree be made to boot efi, but i would say all this is quite far awaysfrom something easy to use and also for something working in a general way for arm chromebooks

a-priestley, 2022-10-03:
I wondered if you might shed some light on u-boot? When you power on the chromebook in developer mode and you have the option to either boot internal, or sd card / usb.. is that u-boot? Why is there only a choice between the two, and not having the ability to boot a separate partition on the internal emmc?

hexdump0815, 2022-10-03:
@a-priestley - the native boot firmware of chromebooks is some kind of coreboot with a special payload called "depthcharge" - so no u-boot ... the very first arm chromebooks (snow etc.) had some hacked version of u-boot instead of coreboot and i'm not really sure if the other arm boot firmwares are actually coreboot based - need to check

a-priestley, 2022-10-03:
Okay interesting, so when you do `lsblk`, and you get back the write-protected `mmcblk0boot0` and `mmcblk0boot1`, that is where depthcharge is? Two of them perhaps because one is for internal, the other for external booting

hexdump0815, 2022-10-03:
@a-priestley - on chromebooks the bootloader is in its own spi flash and not on the emmc etc. - this has the advantage that you usually cannot brick them completely whatever you write to emmc ... if you reflash the bootloader in the spi flash wrong, you can of course again brick them in the worst case ... for the newer chromebooks with ccd (closed case debugging) you can even rewrite the spi flash without much risk as long as you have a suzyqable (or something similar self built) at hand - in that sense the newer chromebooks are quite tinker friendly :) ... maybe a good read would be: https://github.com/hexdump0815/linux-mainline-on-arm-chromebooks and https://github.com/hexdump0815/linux-mainline-on-intel-chromebooks

a-priestley, 2022-10-03:
Ahh yes. I've already made good use of that first link for the GBB flags. Thanks for that!!
Mainline is definitely what I'm looking for on my chromebook. I think my ultimate goal for it is Arch Linux Arm, with mainline kernel *and* GPU acceleration, but as you already stated in another issue, I could be waiting a while for it to become a reality.