velvet-os/imagebuilder #86: Not issue, but of interest - XU4 Kernels

sirzur, 2022-09-15:
I have been keeping a close eye on the XU4 5.10 kernel development thread on the odroid forum, as I have a XU4 and also HC1.
https://forum.odroid.com/viewtopic.php?f=184&t=41300

I noted that they seemed to have made progress on the 5.19 Kernel, especially chewitt who has rebased to his samsung-5.19.y branch
https://github.com/chewitt/linux/commits/samsung-5.19.y

I have tried compiling a kernel using the samsung-5.19.y branch as source and applying all your patches for the 5.19 kernel (not 5.4 kernel from odroid), but have failed at "make modules_install" step, I suspect because I have mashed up the patches and have not compiled the kernel successfully.  When you next revisit the images for mc1/XU4, I would be interested in your take of the necessary patches for the 5.19 kernel with the improvements that has been made in the last couple of months from the odroid thread

hexdump0815, 2022-09-15:
@sirzur - maybe try to build just the chewitt tree without patches? i guess he should already have anything in which should make sense except maybe mglru, which can be added if this first experiment works ... mc1/xu4 is lower in my priority list currently and i plan to look at it around v6.0 or v6.1 again maybe

sirzur, 2022-09-15:
Yeah, good suggestion.  I will give that a go and see what happens