velvet-os/imagebuilder #89: Very messed up colors on Wayland

aneeshlingala, 2022-10-16:
Hello hexdump0815, I am having very weird colors on Wayland, it is very hard to see anything and still persists after logging out, I notice this is only happening on Ubuntu but not Debian, I am running on kappa. How do I solve this issue?

Thanks,
Aneesh

hexdump0815, 2022-10-16:
@aneeshlingala  - which ubuntu version are you using: an older focal one or the newer jammy one? in case of the later you might try to switch to the ubuntu mesa version - see: https://github.com/hexdump0815/imagebuilder/blob/main/doc/from-debian-stable-to-testing-etc.md#preparations

aneeshlingala, 2022-10-16:
I am using jammy.

aneeshlingala, 2022-10-16:
Changing mesa did not work.

aneeshlingala, 2022-10-16:
I have gone back to Debian for now, I may come back to Ubuntu once this
issue is resolved.

Thanks,
Aneesh

hexdump0815, 2022-10-21:
it could that ubunto and debian have different patches in their mesa build or elsewhere in the system resulting in such problems

aneeshlingala, 2022-10-22:
Yeah, could be.

zhxt, 2022-10-28:
@aneeshlingala Hi, have you tried on a external display via HDMI? 
I have seen a mtk device with color issue on eDP display with wayland.

aneeshlingala, 2022-10-28:
@zhxt Sadly can't test that as external display is broken. :(

aneeshlingala, 2023-08-03:
With latest Ubuntu jammy, only some Wayland sessions have weird colors

hexdump0815, 2023-08-03:
i guess this is mesa panfrost improving over time ...